MarlinFirmware/Marlin #3723: Temperature reading

Scratchbuild52, 2016-05-10:
Hi there,

Printer: Prusa I3
 board: Adruino Mega + Ramps1.4

Once I installed the RC6 version, the temperature behaves peculiar. The temperature seems to climb without problems to very near the target temperature and then suddenly start jumping around with 3 or 4 degrees i.e. 241, 243, 239, 242, 245, 241 ...

It then takes a frustrating long time to reach the actual temperature. I did an autotune and did not help with the problem.

Regards
Nic

jbrazio, 2016-05-10:
Could you please attach to this thread by drag and dropping your configuration file ?
It would also be helpful a serial log.
Thanks.

sarf2k4, 2016-05-11:
probably it's the stock pid in the eeprom that has been used instead the one in the auto tune, also make sure you enabled pid in config.h

hope this helps

Eriobis, 2016-05-17:
I had the same problem while using pid on the bed and on the hotend. Your power supply is getting a lot of noise and may cause reading problem. I changed it back to on/off method for the bed. Now everything is fine.

Scratchbuild52, 2016-05-17:
Hi there,

Sorry for getting back so late to you.

The following person had the same problem #3432. The temperature is very eratic. I have a perfect example i.e. example.log which I attached. In this particular log the bed temp assumably overshot and gave me a temp runaway halt error, luckily it was in the beginning of the print. The temperature looks normal at that stage on the screen but it still gave me the error. It seems like this happens when I do not let the bed cool down enough before I print again. I am also attaching my configuration file as you asked.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/268775/Configuration.txt)

[example.txt](https://github.com/MarlinFirmware/Marlin/files/268771/example.txt)

Regards

Nic

Blue-Marlin, 2016-05-17:
Your bed has enormous problems to reach and hold 105°.
Likely your power supply is not strong enough. The voltage seems to drop when the bead is heating full.

Your Configuration_adv.h is not configured adequate for your bed.
```
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius
```

```
> 18:16:53.294 : T:32.1 /0.0 B:97.6 /105.0 B@:127 @:0 W:? // The bed temperature is suddenly rising by ~10° when the bed heater shuts off.
> 18:16:54.311 : T:31.9 /0.0 B:106.1 /105.0 B@:0 @:0 W:9

> 19:08:02.628 : T:31.2 /0.0 B:95.1 /105.0 B@:127 @:0 W:? // dito
> 19:08:03.630 : T:31.2 /0.0 B:103.6 /105.0 B@:0 @:0 W:9

> 19:10:58.992 : ok T:238.4 /240.0 B:103.1 /105.0 B@:127 @:46

> 19:11:02.225 : ok T:239.2 /240.0 B:103.0 /105.0 B@:127 @:42  // Bed is 2° below 

> 19:11:05.283 : ok T:239.4 /240.0 B:102.5 /105.0 B@:127 @:47 // Heater is full on

> 19:11:08.281 : ok T:239.9 /240.0 B:102.6 /105.0 B@:127 @:47

> 19:11:11.338 : ok T:240.3 /240.0 B:102.1 /105.0 B@:127 @:46 // but the temperature is still falling

> 19:11:14.414 : ok T:240.6 /240.0 B:102.6 /105.0 B@:127 @:43

> 19:11:17.417 : ok T:235.8 /240.0 B:102.6 /105.0 B@:127 @:127

> 19:11:20.449 : ok T:242.2 /240.0 B:102.0 /105.0 B@:127 @:18 // at least not rising significant.

> 19:11:22.613 : Error:Thermal Runaway, system stopped! Heater_ID: bed // 20 seconds later the error triggers because the temperature is still to low.
> 19:11:22.613 : Error:Printer halted. kill() called!                  // That's exactly the default configuration, what your bed can never fulfil.
```

Your bed is perfectly self limiting. Likely it's save for you to deactivate `THERMAL_PROTECTION_BED` until you solved the heater/power problem.

You have a RAMPS! Try cooling the big fuses.

sarf2k4, 2016-05-18:
this also might lead to high resistance value on the pcb.

Have you tried insulate the bottom of the pcb?

Scratchbuild52, 2016-05-19:
Hi there,

First of all, I had done the following already before RC6 installation 
                            1) insulation on the bottom of my heated bed - after I installed the insulation, I tested it up to 112 degrees with no problem.
                            2) Right from the beginning I already have a 40mm fan cooling the RAMPS 1.4
                           
Testing:
                           1) I have tested the voltage drop and it is never higher than 400mv - 500mV.
                            2) If I start the bed from cold, I do not have a problem and the temperature rises gradually till the  requested temperature and held it comfortably.
                            3) When I pre-heat the bed or do quick prints successively, I get the behaviour you see in the log file.
                           4) If you look at the log file, you will see that the fluctuations of the temperatures happens in a matter of seconds and if that was equipment it would not jump around like that but would gradually pick up or loose heat as the bed cools down. Just cutting the power, my bed takes about ten to fifteen seconds to drop a few degrees.
                           5) What I also realised is that while the bed temperature is jumping around the bed temperature actually rises normally in the background, with a linear curve, maybe a bit curved but that is expected.
                           6) When I first reported this problem, it was the extruder having the problem, I did a autotune and the behaviour basically stopped on the extruder but then it started on the bed. I have done a few autotunes but with no success.

This all started when I installed RC6, why did it not happen when I had the previous release on my RAMPS1.4. if it was my equipment that was not up to the job?

Regards

Nic

ruggb, 2016-05-22:
I also had a temp problem. It was due to the thermistor leads picking up noise from the fan motor leads. Drove me crazy till I put a scope on it. I separated the leads and it is beautiful now.

thinkyhead, 2016-05-22:
> thermistor leads picking up noise

Always a possibility. The only thing I can think of, as far as different versions of the software being affected differently, is that there are minor changes in signal frequency from version to version, so maybe when it's right on the (rising or falling) edge, noise might be more prone to distort the signal. Total speculation… All I can do on this end is to keep working on optimization and speed up the stepper-signaling regions of the code, make them as performant and robust as possible, etc.

Scratchbuild52, 2016-05-22:
Thanks, I look forward to the next release :-)

closed 2016-05-24 by jbrazio