MarlinFirmware/Marlin #3724: building using platformio - generate_version_header script does not add all needed entries on Marlin 1.1.0-RC6

nicoruy, 2016-05-10:
I'm running "platformio run -e printrboard" to build Marlin for my Printrbot.
The _Version.h file is generated, but some entries are missing.
SanityCheck makes the build to fail.

The missing entries are:
_PROTOCOL_VERSION
MACHINE_NAME
SOURCE_CODE_URL
DEFAULT_MACHINE_UUID
WEBSITE_URL_

I manually added those entries to the _Version.h file and the build succeeded

jbrazio, 2016-05-10:
Platformio is currently unmaintained and we're considering the option to remove it's support from Marlin.

This is the only method you have to upload Marlin to your board or you use it for any other reason ?

nicoruy, 2016-05-10:
I'm tried building the firmware using the arduino IDE and teensyduino.
I need to use a separate tool to upload the firmware to the printer (Atmel Flip) due to the file size.
After uploading the firmware to the printer the printer is not even recognized by USB.

Also manual changes are needed on the hardware config files on the arduino folder to be able to select the printerboard.   
Most posts I found suggest building Marlin using an old arduino version (022 or 1.0.5 rc) and most of them have a pre-configured version to download.
That's not an option with the latest versions of Marlin that require a newer Arduino version.
As you see the whole process of building and uploading a working version of the firmware with arduino+teensyduino+flip is extreemly difficult and error prone.

I tried using platformio as an alternative and (after adding the proper entries to the _version.h file) the whole build process worked without issues.
I still need to try the resulting hex file on my printer but from a build perspective the whole process is a lot better than the alternative. 

What is the reason to remove support for platformio? Is not widely used or supported?

Roxy-3DPrintBoard, 2016-05-10:
@nicoruy    Do you know the details of what it takes to make the current Arduino compile and upload to a PrintRboard?     It would be very helpful if you could give a step by step set of direction to make that happen!

nicoruy, 2016-05-10:
**Building using arduino**
I was not able to build a running version of the firmware with the arduino IDE yet. 
The printer is not recognized after flashing but I will try to explain the basic process and put some links to a more detailed information.
I still need to test the one I built with platformIO.

The board on the printrboard is derived from the teensyboard. 
That's why is not supported by the arduino IDE out of the box.
In order to support it teensyduino needs to be installed on top of arduino.
http://www.pjrc.com/teensy/td_download.html

Unfortunately there is no reference to the printerboard after Teensyduino is installed, and it seems that the bootloader is different from the one used on the Teensy boards.
That's why boards.txt and programmers.txt needs to be manually edited on the arduino folder.
More information can be found on the following articles:
http://blog.lincomatic.com/?p=1508
http://www.printrbottalk.com/forum/viewtopic.php?f=22&t=6092
(as you will see those posts are using and recommending an older version of arduino)

After doing that I can select the printerboard in arduino and build an hex file (by pressing verify)
Then I need to find the temporary folder of the compilation and copy that file to another location (the temp folder will be deleted after closing the arduino IDE)

The other issue with the arduino is that it's not possible to upload the firmware to the board directly from the IDE.
From what I understand this happens due to a limitation on the bootloader and the firmware file size.
On the following link you can see the different bootloaders and its limitations.
http://reprap.org/wiki/Printrboard#Bootloaders 
And you can see the process of uploading the firmware using Atmel Flip
http://www.printrbottalk.com/wiki/index.php?title=Firmware_Update#FLASHING_A_COMPILED_.28HEX.29_VERSION
Flip can be downloaded from the following location:
http://www.atmel.com/tools/FLIP.aspx

After following that process I was able to compile the firmware and flash it to my printrboard.
Unfortunately the board is not even recognized so there might be something I'm doing wrong,
I tried flashing a precompiled version of the firmware downloaded from the following page (details and links on the description):
http://www.thingiverse.com/thing:559657
Also you will see on that page that the author is suggesting using arduino 022 and he mentioned that no other version worked for him.
The precopiled version from that page worked for me, so I don't think there is an issu with the flashing process.
Unfortunately the build process has many dependencies, is complicated and error prone.

---
**Building using platformIO**
The process of building the firmware using platformIO is a lot easier.
After installing platformIO (or the new PlatformIO IDE), I just need to go to the PlatformIOAddons folder and run
_platformio run -e printrboard_

(I need to fix the _Version.h as described on the issue and run it a second time to build it)
As a result I have a firmware.hex file under the .pioenvs/printrboard subfolder.

I can use flip to upload the firmware or
_platformio run -e printrboard --target upload_
(I didn't tried this yet)

Information on how to do this was fouond here: 
http://www.printrbottalk.com/forum/viewtopic.php?f=22&t=10160#p65585

The additional advantage is that the latest arduino is used, and also only used libraries are linked making the whole firmware smaller

Roxy-3DPrintBoard, 2016-05-10:
Thank You for the detail....   I'm reading everything now!

emartinez167, 2016-05-10:
Roxy,

Wasn't this the problem Boo had?

Regards,

Ernesto

paulusjacobus, 2016-05-10:
I actually was able to build and compile Marlin for a Printrbot board (Rev
F) with the Arduino IDE 1.6.8. I installed the Teensy2++ extensions and
select these in the IDE before compile (there is no printrbot in the list,
you to select teensy2++ and the usbcon). Then I take the hex file from the
app/data location (can't lookup the exact location right now) and load that
into the ATMEL Flip flash tool.
Took me a day to get it going but once figured it out it works like a
charm. Just ensure you jumper the board which changes the usb comms port
into a usb flashport and hit the reset button on the printrbot board.

nicoruy, 2016-05-11:
@paulusjacobus I was putting the jumper on the correct position (I just forgot to mention on my coment above) and I was able to upload a precompiled firmware without issues using flip. The printer was recognized and it worked.

Today I tried uploading the firmware built with platformIO.
I tried by using  _platformio run -e printrboard --target upload_ and by using the FLIP tool.
I got the same result. The printer is not even recognized (the same that happened to me building with the arduino IDE and Teensy)

It seems I'm having problems building the firmware. 
I tried now both, using Teensy2++ with arduino (it was actually my first test) and platformIO.
I prefer platformIO because it's easier, it downloads all dependencies, and the IDE based on atom is much better than the arduino IDE.
Unfortunately I'm having the same problems with both platformIO and arduino.

thinkyhead, 2016-05-11:
Maybe it's possible to adapt https://github.com/sudar/Arduino-Makefile to Marlin and cover more bases.

Roxy-3DPrintBoard, 2016-05-11:
> Wasn't this the problem Boo had?

I don't know for sure.  But they are suspiciously similar.

nicoruy, 2016-05-12:
I found the problem why my printer was not recognized.
The problem was not on the build or upload process.
I was recompiling the firmware because I'm switching from the standard extruder to an e3d v6 which has a different termistor. Also it was a good opportunity to switch from the factory firmware to the latest one.
I was testing the new firmware without the extruder and termistor connected. It seems that the firmware does not start on a correct state in that scenario.
With the extruder and termistor connected my printer was recognized. 
The firmware I'm testing was built using platformIO. 
The only change I needed to make was to include the missing #include directives as described on the initial report. I think we can discard other issues.

jbrazio, 2016-05-12:
I'm working on PR #3715 which is deeply tied up with this issue.
I have build a list of "supported" boards which means boards that are listed in `boards.h`, for each board I found which are the micro-controller restrictions, required Arduino core and some miscellaneous notes.

You may see the information here: https://goo.gl/gFpujH
Remember this is a **work in progress**.

A question to the elder ones, anyone knows wtf is the board 99 ?

thinkyhead, 2016-05-12:
@jbrazio Board 99 can go away now.

closed 2016-05-16 by jbrazio