MarlinFirmware/Marlin #3726: WE NEED YOUR HELP !

jbrazio, 2016-05-10:
Dear Marlin users and developers, we the Marlin Development team, require your assistance filling the following Google Form: http://goo.gl/forms/gRsLbQmti6

It's a "one minute" survey were we ask you which board and Marlin version you use.
You may submit multiple times for more than one printer.

boelle, 2016-05-10:
maybe add what IDE they use?

emartinez167, 2016-05-10:
I have posted it on the G+ Rigidbot forum.

Regards,

Ernesto

jbrazio, 2016-05-10:
@boelle good suggestion, added.

emartinez167, 2016-05-10:
Joao,

Some Rigidbot users are not quite sure what to respond regarding the version of Firmware they are on, as it is the original supplied by Rigidbot (we suspect it is an old fork from 1.0). Is there a way for them to provide that information (or add an option stating Rigidbot original FW)?

Regards,

Ernesto

jbrazio, 2016-05-10:
@emartinez167 I added the option "Other/Builtin" and "Other/Unknown".
Interesting to see those people have never upgraded their firmware, is it for any particular reason ?

Blue-Marlin, 2016-05-11:
Where can we see the results?

emartinez167, 2016-05-11:
I believe mostly because of one or more of the following reasons:
a) Code in that version tied to the hardware for that machine
b) a lot of users unfamiliar/uncomfortable with coding Arduino
c) If it ain't broken, don't fix it attitude
d) Lack of knowledge about the existence of a mainstream Marlin
e) Originally the Rigidbot was not officially supported by mainstream Marlin. I submitted the original PR and @thinkyhead completed it for me. It was a long time between the RB coming to the market and the PR. Ini the interim, some RB users with enough knowledge rigged up what is known in the community as RBC (RigidbotCommunity) version which was based on a newer version, but it stayed there and never caught up with the current release.

Regards,

Ernesto

jbrazio, 2016-05-11:
I will publish the results as soon as people start to stop voting, were is a preview.

[image]

[image]

[image]

RalphSutherland, 2016-05-11:
Two problems,  my board not listed, no way to select a custom board (4 power outs: EEFB)   (I chose ramps 1.4 as a default)   No way to allow for not using an unmodified Marlin firmware.  ( I chose the RD branch that I am currently working from, eg RC 6)  I got the IDE question OK :)

jbrazio, 2016-05-11:
But it's a RAMPS 1.4 or does it have any pin diff ?
I listed all boards on `boards.h`, there was the only way to have some consistency.

tkurbad, 2016-05-11:
Shouldn't it be 1.1.0-RCx instead of 1.1.1-RCx?

jbrazio, 2016-05-11:
Damn you're right :-/

JimBrown, 2016-05-12:
BOARD_MKS_13 is missing from the list...

thinkyhead, 2016-05-12:
It should also note that the "MKS" is the same as the "HICTOP" for those who have that branding.

boelle, 2016-05-12:
Where can we see the results?

jbrazio, 2016-05-12:
I will post them soon, just waiting for the survey to ran at least one week.

@Roxy-3DPrintBoard could you please share the link in the forums you have access to ?

We are having a bit less than 200 data points right now.

boelle, 2016-05-24:
are there any results yet?

maybe just view access to the data

jbrazio, 2016-05-24:
Will do !

Roxy-3DPrintBoard, 2016-05-24:
@JBrazio   I just saw the request to post it in different forums.   (I need
your email address!!!)

It is posted at both RepRap.org (in the Marlin section) and at
3D-PrintBoard in the RepRap section.

jbrazio, 2016-05-24:
@Roxy-3DPrintBoard I will post it on the chat we have.

VicentM, 2016-05-24:
Published in spanish google group (Clone Wars)

closed 2016-06-11 by jbrazio