MarlinFirmware/Marlin #3727: coreXY  problem

PeetKos, 2016-05-10:
After upgrade from    RC5 to  RC6        after each layer,        all  layer step   by XY   coordinate

Blue-Marlin, 2016-05-10:
Please rephrase. I don't understand, what the problem is about.

PeetKos, 2016-05-11:
after upgrade   from RC5  to RC6    
on my  handmade  coreXY   RAMPS 1.4   printer    with external  motor  driver   

 i have new problem : 

after printing  first layer  and   Z-step      next layer printed  with  XY  offset  (about -0.5 mm  on X and Y ),    on next  layer   i have   more shift ,   next more   and so one...      in result   cubic obgect will look like  prizm . 

May by   when  I enable   coreXY  in   config  - in fact enabled  coreXZ  or   hexagon algarytm  (i don't know)

sorry  for   my english.

PS  its not  a hardware problem!  because when I rollback  to RC5    -   all   work is  normal

thinkyhead, 2016-05-11:
Could be a bug. Please post your `Configuration.h` file.

PeetKos, 2016-05-12:
anybody have  working corexy printer with  RC6 ?

thinkyhead, 2016-05-12:
> head can move to crash by software comand

Try enabling these:
```cpp
#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops false  // If true, axis won't move to coordinates greater than the defined lengths below.
```

thinkyhead, 2016-05-14:
The config looks good to me.

Have you tested the `RCBugFix` branch to determine if the bug still exists in the latest code?

ozligia, 2016-05-16:
CoreXY is definitely working on RC6.

thinkyhead, 2016-05-16:
Hmm. Well, knowing that CoreXY does work in `RC6` is helpful. And it should be a little better in `RCBugFix` due to minor tweaks here and there. Of course, Z movement should not be affecting XY movement with CoreXY, since XY are the two axes that are tied together. Code-wise, I can see no reason why there might be extra stepping with Z.

So @PeetKos have you tried `RCBugFix`? Can I assume you are using a recent version of Arduino to upload with?

PeetKos, 2016-05-17:
no   I did't   try     RCBugFix

I'm using  last arduino  version for Linux

adamfilip, 2016-05-20:
running Bugfix RC6 with corexy. no issues so far

PeetKos, 2016-05-20:
Sorry,   I  can't  find   documentation  How to pach last bugfix

thinkyhead, 2016-05-20:
@PeetKos Are you just looking for information on how to download `RCBugFix`? If you use your configuration from RC6 it should still work. If you still see layer-shifting in `RCBugFix` then we have to look into it. So far in my scanning of the code I can see nothing strange that would affect XY when moving Z on `COREXY`.

ruggb, 2016-05-22:
I had a similar twisting issue in one of the previous commits when I increased the print speed over 50mm/sec. I determined it was due to Marlin not responding fast enough to the data stream and the result was some code was lost and the print twisted. Marlin was fixed but it seems to be back again at 60mm/sec in 8aa594c - not as bad as it was. I am still experimenting to see at what speed it really breaks.

thinkyhead, 2016-05-22:
@ruggb Missed steps is always disconcerting. I still have to sit down and push through some optimization of the singleton classes recently added, to reduce time spent in the stepper ISR especially. That may help, but it's a minor challenge to get it compiling happily.

ruggb, 2016-05-22:
@thinkyhead good to know u r on it. Just commented to let ppl know that reducing print speed rather than baud rate may do more to get around the problem at this point -- if it is the same thing i ran into.

ruggb, 2016-05-23:
FY(eye) - @ 100mm/sec prntng a honeycomb nfl @ 40% my computer s constantly rngng ERROR
sorry, now me must fx my kbd as me splled wne on t and "eye" doesn't work.8aa591c commt

PeetKos, 2016-05-23:
heh  i sill  can't find  time to  test bugfix   ...  but on RC5  with 115k boud rate  i  print   on 150 mms and 300mms  free   move       (yes my extruder not fast enought and quality bad)    but  no problem  with movment.  
anyway  IMHO max-speed    must be calculated    in  one  main rule:      not more than   10k   steps  per second  in     any   movment .

jbrazio, 2016-05-24:
@PeetKos has we have confirmation from other users that CORE XY is working in the latest RCBugfix branch and due to your lack of time to test the branch, I will close the ticket.

Please re-open it at any time if after testing your problem still persists.

closed 2016-05-24 by jbrazio

PeetKos, 2016-05-26:
I  test   BugFix  but have  same  problem   

on  picture     one  test ittem  printed by RC5        another    with BugFIX
I  try     Symplify3D   and  CURA    sliser   -      bug is  same    but   in   CURA  its  more   linear  (on picture)  When i  print  from  Symlify3D     i   have problem only     after printed  5-10 layer
      
[image]
[image]

ruggb, 2016-05-26:
u don't have any config data posted so I am guessing. U r running 50+ mm/sec speed.
Try running at 30mm/sec. 
That should at least make it straight vertically. If it does not, you have a machine problem. But since it works with RC5, probably not. It may also solve the jagged edge problem. If the jagged corners still remains try tweaking the jerk and accel settings. The jagged edges could be a temp issue. 

I use Repetier Host and have sounds for errors turned on and mine plays a tune to me all the while it is printing over 60mm/sec. That is definitely an issue with Marlin not being able to process the input data fast enough and loosing steps. Changing the baud rate will not help. If you get it slow enough it will make it worse.

This problem comes and goes with each commit. Since the firmware cycle time is a significant performance issue, the team should test it prior to releasing a build. IMHO.

BTW, my machine is a home brew CoreXY. I'm thinking since an X movement involves commands to both X&Y, this type machine may be pushing the firmware.

I would like to see a spec on the max print speed of the firmware. I'm not sure how it would be referenced to make it applicable to various processors or machines. But comparing RC5 to RC6 would be interesting as RC5 seems to work as high as I have run - about 120mm/sec.

Note: running a print with a lot of small XY movement - like honeycomb infill - is definitely where the measurement needs to be made. Long straight lines will not stress it much. Also the orientation of the print on the bed may cause a difference for a corexy and a cartesian.

PeetKos, 2016-05-27:
Ok   i will try it in 30 mmc    ...     on   rc5    i  use 1/4   microstep   on driver  to   rich   max speed  at 0.05 mm  step...          and  have   about   250mms  speed without  lost  steps ...   also   i have  some speed test up to  500mms   ...   but  its   not  stable

PeetKos, 2016-05-27:
test on 30mms       have same result   ...   ^(

ruggb, 2016-05-28:
if u rotate the model on the bed does it tip in the same direction or does a different side tip?
IE is the tip toward the same physical side of the printer or the same side of the model.
I'm thinking if the same side of printer it is a printer issue. if the same side of the model, it is a firmware issue.