MarlinFirmware/Marlin #3729: Crash (Hard Reset) when issuing G2/G3 command

manianac, 2016-05-11:
I am using a RUMBA board with a 300x300x300 build area.  I'm running RCBugFix commit 735f1f5bfa194d6fabecb88f8e8558a6ce269bb8.

I've found when running this code, I lose all comms with the board, and the graphical LCD shows the splash screen (full reset).

```
G28
G1 X130 Y130 Z5 F9000
G2 X130 Y130 I50 J50
```

Still experimenting to see if its only large moves, or any arc moves causing this

jbrazio, 2016-05-11:
Could you please attach a serial log ?

fbarcenas, 2016-05-11:
What slicer you use?

thinkyhead, 2016-05-11:
Also, what compiler (Arduino version) are you using?

manianac, 2016-05-12:
Here is the serial log:
```
Send: G28
Recv: X:0.00 Y:0.00 Z:0.00 E:23.79 Count X: 0 Y:0 Z:6539
Send: G1 Z5
Recv: ok
Send: G1 X130 Y130 F9000
Recv: ok
Send: M105
Recv: ok T:29.5 /0.0 B:29.5 /0.0 B@:0 @:0
Send: G2 X130 Y130 I50 J50
Recv: start
Recv: echo:Marlin 1.1.0-RCBugFix
Recv: echo: Last Updated: 2016-04-27 12:00 | Author: (Robert Kirk, custom config)
Recv: Compiled: May 10 2016
Recv: echo: Free Memory: 3027  PlannerBufferBytes: 1232
Recv: echo:V23 stored settings retrieved (453 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z4000.00 E900.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X250.00 Y250.00 Z2.00 E15.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X750 Y750 Z500 E500
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P750.00 R500.00 T750.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.30 E10.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: Mesh bed leveling:
Recv: echo:  M420 S1 X4 Y4
Recv: echo:  M421 X20.00 Y20.00 Z1.80
Recv: echo:  M421 X95.00 Y20.00 Z1.98
Recv: echo:  M421 X170.00 Y20.00 Z1.94
Recv: echo:  M421 X245.00 Y20.00 Z1.76
Recv: echo:  M421 X20.00 Y96.00 Z1.85
Recv: echo:  M421 X95.00 Y96.00 Z1.97
Recv: echo:  M421 X170.00 Y96.00 Z1.97
Recv: echo:  M421 X245.00 Y96.00 Z1.93
Recv: echo:  M421 X20.00 Y172.00 Z1.94
Recv: echo:  M421 X95.00 Y172.00 Z1.97
Recv: echo:  M421 X170.00 Y172.00 Z1.97
Recv: echo:  M421 X245.00 Y172.00 Z1.86
Recv: echo:  M421 X20.00 Y248.00 Z1.85
Recv: echo:  M421 X95.00 Y248.00 Z1.85
Recv: echo:  M421 X170.00 Y248.00 Z1.87
Recv: echo:  M421 X245.00 Y248.00 Z1.66
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H190 B60 F0
Recv: echo:  M145 S1 H220 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P35.65 I3.59 D88.49 C1.00 L20
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:  M207 S2.00 F900.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:  M208 S0.00 F660.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:  M209 S0
Recv: echo:Filament settings:
Recv: echo:  M200 D2.90
Recv: echo:SD card ok
Recv: echo:SD card ok
```

I am unable to send anything until I disconnect and reconnect (OctoPrint).
Compiler: Arduino IDE 1.6.8
Cartesian Robot, Makerfarm Prusa i3v.
Software endstops are enabled, I will attach my Configuration.h

A slicer did not generate this, I have a gcode I've run to test new config options or features.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/260498/Configuration.h.txt)

thinkyhead, 2016-05-12:
Do you get the same result with:
```gcode
G1 X130 Y130 F9000
G2 I50 J50
```
It's the same idea, and should make a complete circle with a radius just over 70mm, centered around the point 180, 180.

thinkyhead, 2016-05-12:
Let's debug, shall we? Add the following lines to the `plan_arc` function in `Marlin_main.cpp`:
```cpp
    // Update arc_target location
    SERIAL_ECHOPAIR("Segment ", segment);
    SERIAL_ECHOPAIR("at ", arc_target[X_AXIS]);
    SERIAL_ECHOPAIR(",", arc_target[Y_AXIS]);
    SERIAL_EOL;
```
…immediately after the lines…
```cpp
    arc_target[X_AXIS] = center_X + r_X;
    arc_target[Y_AXIS] = center_Y + r_Y;
    arc_target[Z_AXIS] += linear_per_segment;
    arc_target[E_AXIS] += extruder_per_segment;
```
This will tell us how far it gets before it blows up.

manianac, 2016-05-12:
70mm arc @ 180 did crash it, I will update the firmware tonight (long work hours).

manianac, 2016-05-12:
Ok, ran the updated firmware (segment wasn't a defined variable, ended up using "i" from the loop).

```
Send: G28
Recv: X:0.00 Y:0.00 Z:0.00 E:23.79 Count X: 0 Y:0 Z:6539
Send: G1 Z5
Recv: ok
Send: G1 X130 Y130 F9000
Recv: ok
Send: M105
Recv: ok T:29.5 /0.0 B:29.5 /0.0 B@:0 @:0
Send: G2 I50 J50
Recv: Segment 1 at 129.30,130.71
Recv: Segment 2 at 128.61,131.44
Recv: Segment 3 at 127.92,132.17
Recv: Segment 4 at 127.25,132.91
Recv: Segment 5 at 126.59,133.66
Recv: Segment 6 at 125.94,134.42
Recv: Segment 7 at 125.30,135.19
Recv: Segment 8 at 124.67,135.97
Recv: Segment 9 at 124.05,136.76
Recv: Segment 10 at 123.45,137.55
Recv: Segment 11 at 122.85,138.36
Recv: Segment 12 at 122.27,139.17
Recv: Segment 13 at 121.70,139.99
Recv: Segment 14 at 121.14,140.82
Recv: Segment 15 at 120.59,141.66
..
[lines removed for length of post]
..
Recv: Segment 363 at 204.97,113.84
Recv: Segment 364 at 204.03,113.50
Recv: Segment 365 at 203.08,113.16
Recv: Segment 366 at 202.14,112.84
Recv: Segment 367 at 201.18,112.54
Recv: Segment 368 at 200.23,112.24
Recv: start
Recv: echo:Marlin 1.1.0-RCBugFix
```

I do wonder though, am I the only one duplicating this issue?  If so, I'll focus on it on my own board, I just wasn't sure if anyone else was experiencing this.

manianac, 2016-05-13:
Alright, life would've been easier if the 2560 bootloader didn't zero out MCUSR, but I'm 90% positive its a watchdog reset.  Disabling watchdog temporarily solves the issue, and WATCHDOG_RESET_MANUAL causes a bootloop that otherwise would've pointed out the issue.

You may want to consider another location for wdt_reset other than Temperature.cpp

Blue-Marlin, 2016-05-13:
Uiiii.
That means you temperatures and heaters are not updated for that long.
Calculating the next arc sub move to plan, takes longer than it's execution in the stepper-ISR. For that there is always a free place in the planner buffer and
```
  // If the buffer is full: good! That means we are well ahead of the robot.
  // Rest here until there is room in the buffer.
  while (block_buffer_tail == next_buffer_head) idle();
```
idle() is not called, what should call the heater manager.

As a workaround try to increase `MM_PER_ARC_SEGMENT` and/or `N_ARC_CORRECTION`.

One of the possible real solutions could be to call at least `manage_heater()` better `idle()` every ~200ms in `plan_arc`. Something similar is probably needed for the splines (G5)

AnHardt, 2016-05-13:
Someting like
```
    float feed_rate = feedrate * feedrate_multiplier / 60 / 100.0;

+   millis_t previous_ms = millis();

    for (i = 1; i < segments; i++) { // Iterate (segments-1) times

+     millis_t now = millis();
+     if (now - previous_ms > 200) {
+       idle();
+       previous_ms = now;
+     }
      
      if (++count < N_ARC_CORRECTION) {
        // Apply vector rotation matrix to previous r_X / 1
```

AnHardt, 2016-05-13:
I can reproduce the problem with setting `MM_PER_ARC_SEGMENT` smaller and smaller. First it begins to stutter (~0.5). When further decreasing i get watchdog resets (<0.33).

AnHardt, 2016-05-13:
#3751 prevents from the watchdog reset but increases the stuttering a bit. Using `thermalManager.manage_heater();` instead if `idle()` would have a bit less stuttering but all the jobs in `manage_inactivity()` should be done from time to time.

thinkyhead, 2016-05-15:
How about something like this…
```cpp
uint8_t idle_counter = 0;
millis_t next_ping_ms = millis() + 200UL;

for (i = 1; i < segments; i++) { // Iterate (segments-1) times

  millis_t now = millis();
  if (ELAPSED(now, next_ping_ms)) {
    next_ping_ms = now + 200UL;
    if (idle_counter++ & 0x03)
      thermalManager.manage_heater(); // ...and reset watchdog timer
    else
      idle();
  }
. . .
```

jbrazio, 2016-05-24:
@AnHardt @thinkyhead what is the status of this ticket, should we consider it a bug that needs a patch or can it be closed ? Thanks.

AnHardt, 2016-05-24:
Fixed by #3751 and follow ups.

closed 2016-05-24 by jbrazio