MarlinFirmware/Marlin #3733: Mini Kossel Arduino RAMPS 1.4 Stutters During Print

Eddiiie, 2016-05-11:
Hello.

First, thank you so very much for all the hard work that has gone into creating Marlin and making into what it is today.  

I apologize ahead of time if this is not the appropriate place or if this is a known, reported issue, or something I have done incorrectly in my configuration.

I have no idea when this started but with newer (define newer, right?) versions of Marlin, my mini kossel will stop / studder  every couple of seconds and start back up again.  It causes blistering, really impacting my prints.    

Printing from SD card, no USB connected.

If I go back to the firmware that shipped with my printer, the issue disappears.   The differences between these two versions in configuration.h (and _adv) are enormous and at some point I had to start over from scratch.   Perhaps I am missing something.

Can someone else with a Mini Kossel, RAMPS, Arduino setup confirm or deny they see the problem?

I remember seeing a setting in configuration.h that said something about setting this too high takes too much CPU power, cycles, whatever.   Meh?

I also have a Prusa I3, same hardware.  The issue does not exist on that printer.  I think it is a Delta thing.

Thank you again for the time and effort on Marlin and looking into my issue.

All the best!

Eddiiie, 2016-05-11:
Well, I am about to rip out the controller setup and move to a Rumba.
If there is anything anyone wants me to test, check, confirm, please let me know asap
It may already be too late because I've added a Diamond Head on that printer.  I can put things back the way they were if necessary.

I also have a due..  Waiting for the due shield to arrive.

jbrazio, 2016-05-11:
@Eddiiie just to make sure, are you using `RCBugfix` when you experience that stutter ?
We do have an example config file for your printer in `example_configurations/delta/kossel_mini/`, you should have a quick look to see if something appears to be out of place.

thinkyhead, 2016-05-12:
We are in the process of doing some optimization to help reduce this kind of stuttering, which is especially problematic when using a Graphical LCD. In the meantime, the recommended solution is to reduce `DELTA_SEGMENTS_PER_SECOND` until the stuttering goes away.

Eddiiie, 2016-05-13:
DELTA_SEGMENTS_PER_SECOND is what I was looking for.
The Graphic display is exactly it and sorry I left that out.

The printer stops, updates the display, then print, repeat.

Ton of hits on Google, sorry for the duplicate.

closed 2016-05-13 by Eddiiie