MarlinFirmware/Marlin #3734: feature request Solenoid

hexxter, 2016-05-11:
I use a Solenoid to auto level. If you enable a solenoid for long time it is overheating. Is it possible to build in a pin what be enabled short before the z probe is used an disable after the measurement? In other words on/off on every step of probing (on each single point).

Thank you for your work Marlin is excellent!

Roxy-3DPrintBoard, 2016-05-11:
If your mechanical design is tolerant of doing this:  You can switch off the servo after the move.  You can set the timeout value which by default is 300 ms.

hexxter, 2016-05-11:
sorry i don't know what you mean. If i start the auto bed leveling then the firmware start measuring 9 points on my print bed. The problem is i can only activate the magnet (solenoid) before i start bed leveling but the complete procedure takes to long (–15/20sec) and i get a overheating of the solenoid.
i need a pin toggle on if single measurement step ( in my case 9 3x3).

BR Hexxter

Roxy-3DPrintBoard, 2016-05-11:
Check out these lines within the Configuration.h file:
~~~~
// Servo deactivation
//
// With this option servos are powered only during movement, then turned off to prevent jitter.
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
  // Delay (in microseconds) before turning the servo off. This depends on the servo speed.
  // 300ms is a good value but you can try less delay.
  // If the servo can't reach the requested position, increase it.
  #define SERVO_DEACTIVATION_DELAY 300
#endif
~~~~

On your G29 command you may choose to add the 'E' option which will allow retraction of the probe between points to limit how long the servo is energized.

Blue-Marlin, 2016-05-11:
@Roxy-3DPrintBoard 
He has no servo. He also has no BLtouch. He has a solenoid and just a solenoid. With the low duty cycle servo output he can do nothing. What he wants is a new kind of probe - a solenoid probe. Similar to a servo probe, but instead of changing the servos angle, switching a pin high and low. Together wit the 'E' parameter of G29/M48 he then gets 9 short activation cycles for a 3x3 matrix, instead of one long.

@hexxter 
Sorry until we sorted out the probe problems, at least until we are ready with Marlin 1.1.0 we don't want to integrate new kinds of probes into Marlin. For your own purpose, follow the code-path for a servo probe. Everywhere where a servo-angle is changed switch your solenoid-pin. Then use G29/M48 with the 'E' parameter.

Roxy-3DPrintBoard, 2016-05-11:
He said:

> I use a Solenoid to auto level. 

So I assumed he had it working to some degree.

> He has no servo. He also has no BLtouch. 

And because I was thinking he had it working to some degree with Auto Bed Leveling I kind of assumed he was using the normal servo stuff to control it.

Grogyan, 2016-05-11:
@hexxter can you provide more details about your solenoid setup, pictures etc?

I highly recommend you look into the [BLTouch](https://plus.google.com/communities/109126963511513081214) It uses a hall effect sensor to detect changes in the flux of the solenoid to trigger events

Blue-Marlin, 2016-05-11:
Solenoid on. G29. Solenoid off. 
Works but is overheating the solenoid.
That is kind of working.

Roxy-3DPrintBoard, 2016-05-11:
One idea might be to have the solenoid turned on in the run_z_probe() code, let the point be sampled, and then turn off the solenoid and wait some amount of time for it to cool.   That is kind of a brute force way to do things so if that approach is tried, it would be wise to call the functions that monitor and control temperature while you wait for the solenoid to cool down.

Grogyan, 2016-05-11:
If it is getting too hot, then a current limiting resistor i series with the solenoid is required

Blue-Marlin, 2016-05-11:
Right. It's a matter of balancing power and duty cycle by a resistor in series with the solenoid. 
For example: http://www.produktinfo.conrad.com/datenblaetter/500000-524999/506151-da-01-en-ZYLINDERMAGNET_ITS_LZ_1642_Z_12VDC.pdf has not much power at 5.5W but it can be always on. At 55W the power is much higher, but you cant switch it on for longer than 6 seconds

hexxter, 2016-05-12:
@Grogyan here pics: https://cloud.kratzer.it/index.php/s/onv2HptA54wBuGO
https://cloud.kratzer.it/index.php/s/cETLHk0QFpVeCPJ
i use an optical sensor and not an hall sensor. At the moment i have Marlin RC3 running with a dirty patch and it works but i think its was better to have it in the official release. For this Prob is not so many things to do only a configurable pin high short bevor a point is probed and off after the point is measured.

closed 2016-05-24 by jbrazio