MarlinFirmware/Marlin #3736: Marlin 1.1.0-RC4: "Unknown command" error

ericsammut, 2016-05-11:
I really hope someone can help me with this. Here is some background info: Marlin 1.1.0 - RC4 - configured for a delta printer. Running an Azteeg X3 Pro with a Viki 2 LCD controller.

I have been working on getting this firmware up and running and now I am at the point where everything is working except the SD printing. Unfortunately I really need to print via SD card since Marlin has a hard time processing delta movements through USB.

The Viki 2 is supposed to print from Micro SD but I keep getting an "unknown command" error when I try printing. The LCD works, it detects the sd card, but does not properly start the print. The first G28 command works and homes the printer but after that none of the steppers move and I start getting this error. In the end I just kill the print because nothing is actually happening despite the fact that the controller appears to be reading the sd card very quickly. I had a 1 hour print that indicated 30% complete after 5 minutes even though none of the steppers had moved since the initial G28 command at the beginning of the gcode.

I have been reading though all of the other posts on here about "unknown command" errors but it seems like the root of my problem is different. I tried checking the SD cables from the LCD to the board and they are fine. wiggling the cables does not affect this problem.

 Looking at the log on Repetier Host it looks like the gcode is not being parsed properly. The controller will appear to read half of a line of gcode and fail to understand it, and then output the  "unknown command" error. Please take a look at my attached log output. Immediately after the initial G28 homing command the log starts looking like this:

[sd printing error.txt](https://github.com/MarlinFirmware/Marlin/files/259922/sd.printing.error.txt)

Any ideas?

tiagom62, 2016-05-11:
Is the issue present when using RCBugFix?

ericsammut, 2016-05-11:
I'll give it a try and let you know if it works.

fbarcenas, 2016-05-11:
The problem is the VIKI2 and the wires, totally sure...
This display seem very well, but not work well i check along time and i can talk sure...

Blue-Marlin, 2016-05-11:
I' say. A known bug in RC4 and fixed before RC5.

ericsammut, 2016-05-11:
I managed to get RC Bug Fix working with the sd. Thanks so much!

closed 2016-05-11 by jbrazio