MarlinFirmware/Marlin #3742: Write a file in SD inside marlin.

CoderSquirrel, 2016-05-12:
I'm tring to save a file inside the SD in marlin, and i was searching about SdFat and Sd2Card. And then i saw this method insed marlin:

```
void CardReader::write_command(char *buf) {
  char* begin = buf;
  char* npos = 0;
  char* end = buf + strlen(buf) - 1;

  file.writeError = false;
  if ((npos = strchr(buf, 'N')) != NULL) {
    begin = strchr(npos, ' ') + 1;
    end = strchr(npos, '*') - 1;
  }
  end[1] = '\r';
  end[2] = '\n';
  end[3] = '\0';
  file.write(begin);
  if (file.writeError) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_ERR_WRITE_TO_FILE);
  }
}
```

I tried to adapt it to my use based at the things i saw about sdfat:
```
void CardReader::write_new_file(char *buf, char* filename) {
  SdFile myNewFile;
//   if (!myNewFile.open(root, filename, O_RDWR)) {
//    SERIAL_ECHO_START;
//    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
//    SERIAL_ECHOLN(filename);
//  }
  if (!myNewFile.open("teste.txt", O_CREAT | O_EXCL | O_WRITE)) {
    SERIAL_ECHO_START;
    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
    SERIAL_ECHOLN(filename);
  }
  char* begin = buf;
  char* npos = 0;
  char* end = buf + strlen(buf) - 1;

  myNewFile.writeError = false;
  if ((npos = strchr(buf, 'N')) != NULL) {
    begin = strchr(npos, ' ') + 1;
    end = strchr(npos, '*') - 1;
  }
  end[1] = '\r';
  end[2] = '\n';
  end[3] = '\0';
  myNewFile.write(begin);
  if (myNewFile.writeError) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_ERR_WRITE_TO_FILE);
  }
}

```

But i always get error.
if i use the filename, the first if, i got

> echo:Cannot enter subdir: teste
> Error:error writing to file

if not i get
> Error:error writing to file

Do guys guys have done it before? OR What is the right way?

Blue-Marlin, 2016-05-12:
> //   if (!myNewFile.open(root, filename, O_RDWR)) {
//    SERIAL_ECHO_START;
//    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
//    SERIAL_ECHOLN(filename);
//  }
  if (!myNewFile.open("teste.txt", O_CREAT | O_EXCL | O_WRITE)) {

Check the number of your parameters in `myNewFile.open()`

CoderSquirrel, 2016-05-12:
@Blue-Marlin 
the number of parameters are 
`bool SdBaseFile::open(const char* path, uint8_t oflag) { `
or are your saying about  `O_CREAT | O_EXCL | O_WRITE` ?

Blue-Marlin, 2016-05-12:
Dear @CoderSquirrel 
what you commented
> // if (!myNewFile.open(root, filename, O_RDWR)) {

has an other number of parameters.

CoderSquirrel, 2016-05-12:
@Blue-Marlin  i understand npw, thanks
i've tried now
```
void CardReader::write_new_file(char *buf, char* filename) {
  SdFile myNewFile;
  SdFile *parent = &root;
   if (!myNewFile.open(*parent, "teste", O_CREAT | O_EXCL | O_WRITE)) {
    SERIAL_ECHO_START;
    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
    SERIAL_ECHOLN("teste");
  }
.... 
}
```

and got the same : 

> echo:Cannot enter subdir: teste
> Error:error writing to file

Blue-Marlin, 2016-05-12:
Think about objects and pointers to objects.
Is `root` already initialised when you call `write_new_file()`

CoderSquirrel, 2016-05-12:
I initialize the card, open it and then i try to access it via my method, i tought that this way the root would already be setted.

CoderSquirrel, 2016-05-12:
I could make this work this way:
```
void CardReader::write_new_file(char *buf, char* filename) {
SdFile myNewFile;
curDir = &workDir;
    if (!myNewFile.open(curDir, "arquivo.txt", O_CREAT  | O_RDWR | O_EXCL | O_SYNC)) {
      SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
      }
     myNewFile.writeError = false;
myNewFile.print("1 2 3");
  myNewFile.close();
  if (myNewFile.writeError) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_ERR_WRITE_TO_FILE);
  }
}
```
as you said i needed to tell which directore i was, so i add `curDir = &workDir;` but i notice a problem, i only can access it once i clicked at SD Menu option,

closed 2016-05-24 by jbrazio