MarlinFirmware/Marlin #3743: Issues compiling RCBugFix on Linux

Alex9779, 2016-05-12:
I tried the new RCBugFix of last night on my box but I had issues.

At first I used a build my build host made running Debian Jessie with a build script based on what you do for TravisCI. This uses 1.6.4 of Arduino to compile.
It compiles without issues but the problem on my box is then I am able to click the LCD button but scrolling in the menus does not work. So I only can access the main menu and return to the status screen. Didn't test printing...

Then I tried exactly that commit to build on OS X with Arduino 1.6.8 which produced some warning but the firmware on the box worked fine without the issue not being able to scroll.

So I tried using another version of Arduino IDE on the build host to compile but every version starting from 1.6.6 is not able to complete with the following errors:
```
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorised'
```

I currently have no idea where the problem is located. Any ideas or hints?

Alex9779, 2016-05-12:
Ok an update... I tried to print with the firmware I compiled on my Mac and it seemed ok until I noticed that the hotend fans did not start at the setup threshold of 40° C. Noticed it too late, the heatsink already got hot and melted the lower X carriage...
Had two switch to my old carriage to print a replacement and so to a different config branched from RC6. There the fans run normal...

jbrazio, 2016-05-12:
> running Debian Jessie with a build script based on what you do for TravisCI.

Why can you simply do `arduino --verify Marlin/Marlin.ino` ?

Alex9779, 2016-05-12:
> Why can you simply do arduino --verify Marlin/Marlin.ino

I could and basically I am doing this. I am just getting the Arduino env and the needed libs every time, copy all together and then run the build.

To clarify the problem is that compiling with the same environment I used last week for RCBugFix does not work this week anymore. When I flash that firmware I am not able to navigate in the menus.
But this only happens if I compile it on Linux. The Mac version compiled and I was able to navigate the menu but then the other issue occurred, that the fans didn't turn on and my hotend melted the carriage...

thinkyhead, 2016-05-12:
> hotend fans did not start at the setup threshold of 40°

I guess we'll need to see your configuration.

Alex9779, 2016-05-12:
It is all on Github: https://github.com/Alex9779/Marlin/tree/BigBox/Dual/Alex9779/RCBugFix_Hybrid_Titan_Mirrored_MBL
That is the branch which latest commit is compiling on Linux but the navigation in the menus is not possible. Compiling the same commit on OS X with 1.6.8 lets me navigate the menus but does not start the hotend cooling fans.

Alex9779, 2016-05-13:
I did some further research and found that the problem started with the conversion to the singleton stuff.
I get the following warnings with my config on current RCBugFix:
`sketch/temperature.cpp: In member function 'void Temperature::checkExtruderAutoFans()':
sketch/temperature.cpp:335:131: warning: narrowing conversion of '-1' from 'int' to 'const uint8_t {aka const unsigned char}' inside { } [-Wnarrowing]
     const uint8_t fanPin[] = { EXTRUDER_0_AUTO_FAN_PIN, EXTRUDER_1_AUTO_FAN_PIN, EXTRUDER_2_AUTO_FAN_PIN, EXTRUDER_3_AUTO_FAN_PIN };
                                                                                                                                   ^
sketch/temperature.cpp:335:131: warning: narrowing conversion of '-1' from 'int' to 'const uint8_t {aka const unsigned char}' inside { } [-Wnarrowing]
sketch/temperature.cpp:346:32: warning: iteration 2u invokes undefined behavior [-Waggressive-loop-optimizations]
       if (current_temperature[f] > EXTRUDER_AUTO_FAN_TEMPERATURE)
                                ^
sketch/temperature.cpp:345:5: note: containing loop
     for (int f = 0; f <= 3; f++) {
     ^`

Seems to be related to 1a97442d19869a95bb8f942548361c4ef273bcb3...

I ran an RCBugFix successfully on my machine compiled from ad0f70f2f34b9cc7d0b2d4c76148d3708263c720, then I paused updating a bit until now...

thinkyhead, 2016-05-14:
P.S. You need to use [three back-ticks](https://guides.github.com/features/mastering-markdown/) for multi-line quotes.

Alex9779, 2016-05-15:
Compiles fine again, no warnings... Going to test on the machine later...

Alex9779, 2016-05-15:
Test seemed ok on the machine, then I tried again printing but again the hotend cooling fans do not start at the threshold of 40° C I set...

thinkyhead, 2016-05-16:
PR #3771 will ensure the fan is not cleared after it has been set. Merging soon.

jbrazio, 2016-05-16:
Before merging #3771 merge #3769 so we don't have to rebase anything.

Alex9779, 2016-05-16:
Ok compiled (took a while to get my process working again because of all those changes) and tested on the box, did not print yet but the fans are running, looks good.

closed 2016-05-16 by Alex9779