MarlinFirmware/Marlin #3755: Floating Constant?

nefiwashere, 2016-05-14:
need some help getting this error solved, was on an old version of marlin so i copied over all my setting to the new one

Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Conditionals.h:276:0,

                 from sketch\Configuration_adv.h:688,

                 from sketch\Configuration.h:1130,

                 from C:\Users\CLERIC\Desktop\3d print\Marlin-1.1.0-RC6 10''\Marlin\Marlin.ino:37:

Configuration.h:606: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER 49.2  // X offset: -left  +right  [of the nozzle]

                                        ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:24: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                        ^

sketch\Conditionals.h:354:53: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

     #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:278:37: note: in expansion of macro 'MIN_PROBE_X'

       #if LEFT_PROBE_BED_POSITION < MIN_PROBE_X

                                     ^

Configuration.h:606: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER 49.2  // X offset: -left  +right  [of the nozzle]

                                        ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:32: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                                ^

sketch\Conditionals.h:354:53: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

     #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:278:37: note: in expansion of macro 'MIN_PROBE_X'

       #if LEFT_PROBE_BED_POSITION < MIN_PROBE_X

                                     ^

Configuration.h:606: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER 49.2  // X offset: -left  +right  [of the nozzle]

                                        ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                        ^

sketch\Conditionals.h:355:53: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

     #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:280:40: note: in expansion of macro 'MAX_PROBE_X'

       #elif RIGHT_PROBE_BED_POSITION > MAX_PROBE_X

                                        ^

Configuration.h:606: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER 49.2  // X offset: -left  +right  [of the nozzle]

                                        ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:32: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                                ^

sketch\Conditionals.h:355:53: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

     #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:280:40: note: in expansion of macro 'MAX_PROBE_X'

       #elif RIGHT_PROBE_BED_POSITION > MAX_PROBE_X

                                        ^

Configuration.h:607: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -10.00  // Y offset: -front +behind [the nozzle]

                                         ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:24: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                        ^

sketch\Conditionals.h:356:53: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

     #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:282:40: note: in expansion of macro 'MIN_PROBE_Y'

       #elif FRONT_PROBE_BED_POSITION < MIN_PROBE_Y

                                        ^

Configuration.h:607: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -10.00  // Y offset: -front +behind [the nozzle]

                                         ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:32: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                                ^

sketch\Conditionals.h:356:53: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

     #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:282:40: note: in expansion of macro 'MIN_PROBE_Y'

       #elif FRONT_PROBE_BED_POSITION < MIN_PROBE_Y

                                        ^

Configuration.h:607: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -10.00  // Y offset: -front +behind [the nozzle]

                                         ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                        ^

sketch\Conditionals.h:357:53: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

     #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:284:39: note: in expansion of macro 'MAX_PROBE_Y'

       #elif BACK_PROBE_BED_POSITION > MAX_PROBE_Y

                                       ^

Configuration.h:607: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -10.00  // Y offset: -front +behind [the nozzle]

                                         ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:32: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                                ^

sketch\Conditionals.h:357:53: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

     #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                     ^

sketch\SanityCheck.h:284:39: note: in expansion of macro 'MAX_PROBE_Y'

       #elif BACK_PROBE_BED_POSITION > MAX_PROBE_Y

                                       ^

In file included from sketch\Marlin.h:44:0,

                 from sketch\thermistortables.h:26,

                 from sketch\Configuration.h:1131,

                 from C:\Users\CLERIC\Desktop\3d print\Marlin-1.1.0-RC6 10''\Marlin\Marlin.ino:37:

Configuration.h:1133: error: #endif without #if

 #endif //CONFIGURATION_H

  ^

In file included from C:\Users\CLERIC\Desktop\3d print\Marlin-1.1.0-RC6 10''\Marlin\Marlin.ino:37:0:

Configuration.h:1133: error: #endif without #if

 #endif //CONFIGURATION_H

  ^

exit status 1
floating constant in preprocessor expression

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

[image]

jbrazio, 2016-05-14:
Basically you cannot use floats.
You should change `#define X_PROBE_OFFSET_FROM_EXTRUDER 49.2` to `#define X_PROBE_OFFSET_FROM_EXTRUDER 49`.

Roxy-3DPrintBoard, 2016-05-14:
The X & Y distance from the nozzle do not need to be set with sub-milli-meter precision.   The Grid Based Auto Bed Leveling does take the distance from the nozzle into account when it sets the height of the nozzle.   But because the bed is not tilted very much, using sub-milli-meter precision on the offset will hardly change anything.   

(The C Pre-Processor does not allow the use of floats in its computations.  We can't do sanity checks on the values if they are floats.   You will notice Z_PROBE_OFFSET_FROM_EXTRUDER really is a float.  And as a result, we can't do sanity checks on it.  That one really does need sub-milli-meter precision.)

thinkyhead, 2016-05-14:
Note the comment in `Configuration.h` above these options:
```cpp
  // X and Y offsets must be integers.
```

closed 2016-05-14 by thinkyhead

nefiwashere, 2016-05-15:
now im getting this endif with out the if error

Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin.h:44:0,

                 from sketch\thermistortables.h:26,

                 from sketch\Configuration.h:1131,

                 from C:\Users\CLERIC\Desktop\3d print\Marlin-1.1.0-RC6 10''\Marlin\Marlin.ino:37:

Configuration.h:1133: error: #endif without #if

 #endif //CONFIGURATION_H

  ^

In file included from C:\Users\CLERIC\Desktop\3d print\Marlin-1.1.0-RC6 10''\Marlin\Marlin.ino:37:0:

Configuration.h:1133: error: #endif without #if

 #endif //CONFIGURATION_H

  ^

exit status 1
#endif without #if

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

nefiwashere, 2016-05-15:
[image]

thinkyhead, 2016-05-15:
It looks like you might have solved this already. But if not…

If you have a diff tool (or can download one), take an unmodified copy of `Configuration.h` and compare it to your modified copy side-by-side. Your changes will stand out, as will any missing lines.

nefiwashere, 2016-05-16:
i have thank you.    now im running to some new issues. im on an old firmware and trying to move to new updated version i have copied the best that i could settings over to the new firmware but still things dont work. there is so much new stuff in there that i dont even know what to do with most of it

jbrazio, 2016-05-16:
Depending how old your previous config is a lot of options has changed, I would advise to start with a new RC6 config and option by option get the values from the previous config file.

Going back to the diff program you may use [meld](http://meldmerge.org/) which is multi-platform and very simple to use.