MarlinFirmware/Marlin #3764: What is Function FR and the percentage in reprapdiscount full graphic smart controller ??

pubalan12, 2016-05-15:
[image]

Thank You

paulusjacobus, 2016-05-15:
Fr stands for flow rate which is used for the extruder. 105 means 1.05 times the normal flow rate. Used to correct the extruding when the filament diameter is slightly off.

pubalan12, 2016-05-15:
ok. Got it. Thanks paulusjacobus.

bdowling, 2016-05-15:
The FR% on the main screen is for feed rate.  It allows you to increase the speed of your printing if you need to up or down by a ratio applied to all stepper speeds (extruders are adjusted volumetrically). 

Flow rate is tweakable from the tune menu if needed.  Which would only increase or decrease extrusion irrespective to speed.

closed 2016-05-15 by jbrazio

pubalan12, 2016-05-17:
got it. Thanks bdowling