MarlinFirmware/Marlin #3765: change filament

Jinxycob, 2016-05-15:
hello 
 I just installed  the reprapdiscount full graphic smart controller  on a RAMPS 1.4 using the 1.1.0 RC6 and the option  to change filament is missing, thought it was supposed to be the last option in  TUNE? but cant see the option anywhere in the controller setting.

Blue-Marlin, 2016-05-15:
`FILAMENTCHANGEENABLE` Configuration_adv.h ?

closed 2016-05-15 by jbrazio

Jinxycob, 2016-05-15:
thanks for that, but I do have one problem with it , park the effector plate  ok, but the moment  it finished retracting it moves  slowly straight  into the bed leaving little time to exchange fila.
"have to hit the kill switch every time".

thinkyhead, 2016-05-16:
Sounds like another issue should be posted about this. Filament change shouldn't be causing that weird Z downward movement.

WZ9V, 2016-05-16:
I have to do some more testing but when I tried this it seemed like it disabled the X stepper which allowed me to accidentally shift the carriage and this restart in the wrong place.  I'll do some more testing to see if I can figure out what was happening at the time but I do know it took me a long tome to get the filament changed so it might have been a timeout of some sort.

thinkyhead, 2016-05-16:
Hmm… Yeah, if users are going to be fiddling with the extruder, it's probably a Bad Thing™ to disable the axis steppers during the process. Something to patch, along with the weird Z movement.

WZ9V, 2016-05-16:
I want to see if I can reproduce the problem first, if I can I'll open up an issue because yes the whole point of change filament is to mess with the extruder (changing or swapping filament perhaps) ;-)