MarlinFirmware/Marlin #3773: fila change #2

Jinxycob, 2016-05-16:
ello, 
continuous of this []( https://github.com/MarlinFirmware/Marlin/issues/3765) made a short vid [](https://www.youtube.com/watch?v=apTq_IyEDbA&feature=youtu.be)

thinkyhead, 2016-05-16:
Some of this hopefully will be improved (at least for Manual Filament Change) by #3662. It would be good to add some fixes for the bugs you're reporting to that same PR.

jbrazio, 2016-05-17:
@thinkyhead do you believe we can merge #3605 and #3662 ?
It would make sense doing so an let @Jinxycob validate if the same behaviour is still happening.

Jinxycob, 2016-05-17:
@jbrazio  thanks for those "A New Hope #3605" inserted that  so far , but something just aint right with positions, when I click REMOVE 'the nozzle heads straight to the  bed without stopping and INSERT  moves half way down  and straight into the  Z tower. 
        "on a plus note speed  of the axis  is good".
 but its made no difference to changing filament

clexpert, 2016-05-17:
@jbrazio I am sure that #3662 can be merged without any problems. But I am working on manual filament change complete rewriting for next release. So, it will be ready in several days, I think.

jbrazio, 2016-05-17:
For the next release @clexpert or for this RC cycle ?
Now that multi color printing is going mainstream we need to have something more reliable even if it's not the final version of it (we may ship it as *experimental* or *tech preview*).

thinkyhead, 2016-05-17:
@Jinxycob Are you using a `DELTA`? I've just now updated #3605 to handle Delta properly, because it wasn't taking it into consideration, and that could certainly have been causing Z (or other weird) movements.

Jinxycob, 2016-05-18:
@thinkyhead  yea its  a kossel mini , am a little lost to find  the " Travis test " just cant see it in my marlin files.

clexpert, 2016-05-18:
@jbrazio Yes, for this RC cycle, of course. In other words - ASAP :smile:

jbrazio, 2016-05-18:
> am a little lost to find the " Travis test " just cant see it in my marlin files.

I do not understand what you want to accomplish..

thinkyhead, 2016-05-19:
> am a little lost to find the "Travis test"

It's in the root of the file-tree, an invisible file named `.travis.yml`, if you need to mess with it for some reason.

So you might still test out #3605 now that it's been patched for Delta. I'd be curious to know your `FILAMENTCHANGE_ZADD` setting, as that determines how far to move Z after the first retract, as set by `FILAMENTCHANGE_FIRSTRETRACT`. I haven't touched #3662 yet, but perhaps it's good enough to merge (ahead of #3605) at this point… It probably needs more testing and feedback just to be sure.

Jinxycob, 2016-05-19:
got myself in a bit of tiss here, I start over I think, reason a thought I had to change  the Travis  was this    https://github.com/MarlinFirmware/Marlin/pull/3605/commits/df7d1748325078c69c3235c7a280327213116d83

the Zadd is set to 100 
    #define FILAMENTCHANGE_XPOS 40
    #define FILAMENTCHANGE_YPOS -60
    #define FILAMENTCHANGE_ZADD 100
those positions  work fine on the delta brings the nozzle to the front, BUT   I think  i still got the zadd wrong, because it may hit the end stops  if i change on taller prints  so I be changing that back 30

thinkyhead, 2016-05-20:
And you're simply issuing the `M600` command with no other arguments?

jbrazio, 2016-05-30:
This issue will be closed due to inactivity as it is planned to improve the process for 1.1.0 GA.
If required please re-open it.

closed 2016-05-30 by jbrazio