MarlinFirmware/Marlin #3774: A question about the planner

i-make-robots, 2016-05-16:
I'm trying to understand the planner / ISR relationship, and something doesn't click.  Let's say planner will add many blocks to an empty block buffer.

- Planner adds block 1, which is set to decel to zero at the end of the block.  
- Planner calls st_wake_up(), tho as far as I can tell timer is always running.
- ISR immediately starts working on block 1, sets block 1 to busy.  
- Planner adds block 2.  Trapezoids recalculation doesn't touch block 1 because it is busy.

Why don't I see the machine hiccup on the first block and then smoothly move through the rest of the blocks?

paulusjacobus, 2016-05-16:
@danroyer because the buffer in the planner which is a ring buffer that is first filled with calculated blocks and after being filled, the planner recalculates the blocks again but starting with the last block (reverse). It matches the exit and entry speeds to have a smooth transition between the blocks. When the buffer is being consumed, it is continuously being filled up and recalculated. Each block contains a g code move.

If the buffer would run empty because of a communication hicup and being refilled from empty we would see a hiccup between g codes but by then the timeout would have kicked in...and the printer would stop. The tick is to fill the ring buffer and calculate the entire buffer not just block by block. Smoothieware has a very good graphical explanation of the process on their website, just google smoothieware planner. Btw all planners seemed to be ported from grbl.😜

i-make-robots, 2016-05-16:
We seem to have a misunderstanding.  I will rephrase the hypothetical situation.  Assume all blocks being added are in a line, and the same feed rate.

Start with an empty ring buffer.  add one block.  The block starts at v=0, rises to v=nominal, and drops at the end to v=0.  Add a second block.  Planner would remove the decel and accel between blocks 1 and 2.  EXCEPT by the time block 2 is being added to the buffer the ISR is already working on block 1, so it is off limits to the trapezoid recalculation system (see https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/planner.cpp#L215).  Now add a block 3, still in a line, while ISR is working on block 1.  Accel between blocks 2 & 3 will be smooth because both are !busy.  So we get speed up, nominal 1, speed down to zero, speed up, nominal 2, nominal 3, speed down.  I don't see this behavior on my printer and I don't understand why.

The only way I can see that block 1 would not be off-limits is if block 2 is added while the ISR is off.

paulusjacobus, 2016-05-16:
@Dan Royer I believe what you described here is called jerk. There is
another routine that resolves the jerk between blocks. BTW there are
multiple ISRs at work.
Another explanation is that the ISR timer (freq 10KHz) is just too high to
get this situation.
@sebastianv650 you might be able to explain this in more detail since you
have reworked the ISR stepper code.

i-make-robots, 2016-05-16:
I have read the jerk compensation code, which is part of the planner, before the trapezoid recalculation.  If the stepper ISR were firing fast enough to exhaust block 1 before block 2 was added, then there'd be no point in trapezoid recalculation.  I would be surprised if the trapezoid code fires before the first ISR step.

i-make-robots, 2016-05-17:
PS @paulusjacobus if you tag me with my github name instead of my real name, i'll get a notice.  Github doesn't know who @Dan is.

Sebastianv650, 2016-05-17:
@i-make-robots you are absolutely right, the first move after comming from an empty buffer should always end with minimum speed.
Do you have written a test g-code file to see the hiccup? If not, I guess the reason why it can't be seen in normal printing is the start gcode: The first commands are always things like home axis, where an axis will come to a stop after a move in any way. So you can't see the stop here. In the mean time, the buffer is filled and no further hiccup will occur.

If somebody wants to test this, a simple gcode file with **only** two G1 movements in the same direction should do the trick.

i-make-robots, 2016-05-17:
My machine is homed before I start a print, and typically the first commands are "wait for bed to heat".  I guess the buffer fills while it waits?

I have only a variation of Marlin that I've used on the Makelangelo polargraph robot.  In my attempts to optimize it I've had buffer underruns which cause the mistake repeatedly.

Blue-Marlin, 2016-05-17:
Oh. Makelangelo's cinematic looks as if g-code lines have to be divided into lots of straight planner lines. You probably never see a single planner entry long enough to recognise..

Sebastianv650, 2016-05-17:
> My machine is homed before I start a print, and typically the first commands are "wait for bed to heat". I guess the buffer fills while it waits?

In Marlin yes. I can't speak for Makelangelo.

i-make-robots, 2016-05-17:
@Blue-Marlin you are correct, M has to split gcode into many planner blocks to compensate for the polargraph coordinate system.  Playing with the length of planner blocks revealed the hiccup.  I wondered what Marlin did about it, then discovered that Marlin does nothing.  Seemed... wrong.  Like, what happens after a G04, which waits for the planner buffer to empty?

Blue-Marlin, 2016-05-17:
Buffer under-runs because of the time for (calculating the cinematics + planing the move) > (steppers need to execute the move) are common. What you see there is a really empty planner buffer. 

SLOWDOWN is one of the attempts to deal with the problem. But if the segments are not only to fast, but to short this has little effect.
Things go really bad when you thy to plan moves shorter then 'dropsegments' steps (for example 1/80 * 5 = 0.0625mm). The then planer will throw away the move we calculated. Some slicers do produce such a shit.

closed 2016-05-24 by jbrazio