MarlinFirmware/Marlin #3777: Does  MBL work with arcs (M2/M3)?

AnHardt, 2016-05-17:
As far i can see `plan_arc()` calls `planner.buffer_line()` directly,  so `mesh_buffer_line()` (, what's normally called from `prepare_move_cartesian()`, called from `prepare_move()`) is omitted.

Let's see what it needs to replace https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L7545-L7573 with two calls to `prepare_move()`.

AnHardt, 2016-05-17:
https://github.com/AnHardt/Marlin/pull/34 is a potentioal fix. Needs more testing. (At least still draws circles on a Pi3 :-) )

epatel, 2016-05-17:
@AnHardt If the line segments are small enough I believe it's a smaller risk for problems. The mesh version of plan_buffer_line splits on mesh boundaries, the actual MBL compensation is added within plan_buffer_line.

AnHardt, 2016-05-17:
A ja, thanks. I was fooled by the delta leveling again, what is correcting the height before entering the planner - i supposed a symmetry.

So https://github.com/AnHardt/Marlin/pull/34 is only a (for now) unnecessary beautifier. :-)

closed 2016-05-23 by jbrazio