MarlinFirmware/Marlin #3778: Dual X Carriage compile issue

pbrgle1, 2016-05-17:
Hi.

The code won't compile with Dual X Carriage enabled in this version (RC6 and RCBugfix).

**Error message:**
Arduino: 1.6.5 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from Configuration_adv.h:689:0,
                 from Configuration.h:1129,
                 from Marlin.h:44,
                 from blinkm.cpp:28:
SanityCheck.h:362: error: #error Missing or invalid definitions for DUAL_X_CARRIAGE mode.
     #error Missing or invalid definitions for DUAL_X_CARRIAGE mode.
      ^
#error Missing or invalid definitions for DUAL_X_CARRIAGE mode.

Do you need some more information?

pbrgle1, 2016-05-17:
Here is my code for the dual x carriage. It only doesn't compile when dual x carrige is enabled.

// Enable this for dual x-carriage printers.
// A dual x-carriage design has the advantage that the inactive extruder can be parked which
// prevents hot-end ooze contaminating the print. It also reduces the weight of each x-carriage
// allowing faster printing speeds.
#define DUAL_X_CARRIAGE
#if ENABLED(DUAL_X_CARRIAGE)
  // Configuration for second X-carriage
  // Note: the first x-carriage is defined as the x-carriage which homes to the minimum endstop;
  // the second x-carriage always homes to the maximum endstop.
  #define X2_MIN_POS 0     // set minimum to ensure second x-carriage doesn't hit the parked first X-carriage
  #define X2_MAX_POS 257.4    // set maximum to the distance between toolheads when both heads are homed
  #define X2_HOME_DIR 1     // the second X-carriage always homes to the maximum endstop position
  #define X2_HOME_POS X2_MAX_POS // default home position is the maximum carriage position
      // However: In this mode the EXTRUDER_OFFSET_X value for the second extruder provides a software
      // override for X2_HOME_POS. This also allow recalibration of the distance between the two endstops
      // without modifying the firmware (through the "M218 T1 X???" command).
      // Remember: you should set the second extruder x-offset to 0 in your slicer.

  // Pins for second x-carriage stepper driver (defined here to avoid further complicating pins.h)
  #define X2_ENABLE_PIN 23
  #define X2_STEP_PIN 22
  #define X2_DIR_PIN 60

  // There are a few selectable movement modes for dual x-carriages using M605 S<mode>
  //    Mode 0: Full control. The slicer has full control over both x-carriages and can achieve optimal travel results
  //                           as long as it supports dual x-carriages. (M605 S0)
  //    Mode 1: Auto-park mode. The firmware will automatically park and unpark the x-carriages on tool changes so
  //                           that additional slicer support is not required. (M605 S1)
  //    Mode 2: Duplication mode. The firmware will transparently make the second x-carriage and extruder copy all
  //                           actions of the first x-carriage. This allows the printer to print 2 arbitrary items at
  //                           once. (2nd extruder x offset and temp offset are set using: M605 S2 [Xnnn] [Rmmm])

  // This is the default power-up mode which can be later using M605.
  #define DEFAULT_DUAL_X_CARRIAGE_MODE 1

  // Default settings in "Auto-park Mode"
  #define TOOLCHANGE_PARK_ZLIFT   0.42      // the distance to raise Z axis when parking an extruder
  #define TOOLCHANGE_UNPARK_ZLIFT 0.42        // the distance to raise Z axis when unparking an extruder

  // Default x offset in duplication mode (typically set to half print bed width)
  #define DEFAULT_DUPLICATION_X_OFFSET 100

#endif //DUAL_X_CARRIAGE

thinkyhead, 2016-05-17:
Based on your settings, and the error, one of the following problems is still possible:

- You have `EXTRUDERS` set to 1 (should be 2)
- You have `COREXY` enabled (not compatible)
- No X Max Endstop specified (required to home X2 carriage)

pbrgle1, 2016-05-18:
I forgot to enable X_Max endstop but now i get another error (in RCBugfix):

Arduino: 1.6.5 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from stepper.h:46:0,
                 from cardreader.cpp:26:
planner.h:169: error: storage class specified for 'extruder_duplication_enabled'
       extern bool extruder_duplication_enabled;
                   ^
storage class specified for 'extruder_duplication_enabled'

tiagom62, 2016-05-18:
Probably a good idea to add a sanity check for no x max endstop with X2.

thinkyhead, 2016-05-18:
> Probably a good idea to add a sanity check for no x max endstop with X2.

Well, that is the part of the sanity check that failed. But yes, it can be improved by splitting up these sanity checks so they all report separate errors.

thinkyhead, 2016-05-18:
This issue should be resolved with #3788 — merging it now.

closed 2016-05-24 by jbrazio

Ziyayildizz3, 2019-01-22:
Hello there. Is it possible to set z offset for dual x from the printer display? How about we make an addition?

Roxy-3D, 2020-07-25:
I think the github-actions[bot] did the right thing closing this issue.