MarlinFirmware/Marlin #3780: A question for the elders

jbrazio, 2016-05-17:
Does anyone knows what is the following script [Marlin/Marlin/scripts/g29_auto.py](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/scripts/g29_auto.py) ?

epatel, 2016-05-17:
See https://github.com/MarlinFirmware/Marlin/commits/RCBugFix/Marlin/scripts/g29_auto.py

and initial commit says 

> this script scans the first few lines from a gcode. If the line between 2 different z are greater than min_g1 this is our first layer. On this layer we search the min and max values of X and Y. With an offset we write that in a new file.

jbrazio, 2016-05-17:
@epatel..

I saw the git blame, but I still do not understand it's actual propose.. is this for MBL ?

epatel, 2016-05-17:
@jbrazio No, not for MBL. Looks like a script that will do some min/max search and then replace a tag for a new `G29` ABL g-code

jbrazio, 2016-05-17:
ABL.. you're right.

```python
new_command = 'G29 L{0} R{1} F{2} B{3} P{4}\n'.format(min_x,
                                                      max_x,
                                                      min_y,
                                                      max_y,
                                                      probing_points)
```

```cpp
   *  F  Set the Front limit of the probing grid
   *  B  Set the Back limit of the probing grid
   *  L  Set the Left limit of the probing grid
   *  R  Set the Right limit of the probing grid
```

I doubt someone is currently using this.

epatel, 2016-05-17:
Well, when I look at it right now I guess it's pretty neat. It looks at the print area (first layer) in the g-code file and makes a special `G29` command for ABL that _only_ will probe that part of the bed. So if you are printing a small thing it will probe a small area and get what I believe a pretty good tilt matrix for ABL, just for that area.

jbrazio, 2016-05-18:
So this is like a Slic3r script.

Roxy-3DPrintBoard, 2016-05-18:
The history of the Front, Back, Left, Right stuff is we thought there was benefit to only probing the area needed by the print.   It was an interesting collaboration.   But, I think people's experience is probing the whole bed works just as well and very possibly, even better.

jbrazio, 2016-05-18:
/me takes out the vacuum cleaner from the backpack

closed 2016-05-18 by jbrazio