MarlinFirmware/Marlin #3786: Idea: Endstop / Homeswitch / Probe

AnHardt, 2016-05-18:
Maybe we can make progress with sorting out the endstop/probe/homeswitch problems when we try to define them for us..

* Endstop (hardware): A security device to stop the axis movement before something gets demaged. 
 * A hit should never occur. If so ether the according software endstop is off and we tried to move to far, or we lost steps. An error should be thrown.
 * The action to stop the axis is a hard one - we lose steps. Homing the axis thereafter is required. 
 * Hitting an endstop is not homing - when homing we have a defined speed when the endstop is hit. 
 * Does not need no special procedure or position to work. 
 * An axis can have two of them.
* Endstop (software): A security device to prevent from hitting an hardware endstop, or a limit where we do not have a hardware endstop.
 * Clip a move before it is executed - so a smooth approach to the limit can be made.
 * Should prevent extrusion when at the limit.
 * printing a warning is adequate.
 * Need homing before they can work.
 * Should be always on after homing.
 * Need to be off during homing (probing?) if the homeswitch is also an endstop.
* Homeswitch: A measurement device to find a defined position of an axis. 
 * Needs a special procedure to succeed. 
 * Can be accomplished by the need for a special order to home the axis, or special positions of an other axis. 
 * Only one of them per axis makes sense. 
 * When not in the homing procedure a homeswitch should be ignored. 
 * Can measure only one point (homeposition).
 * Can share the physical switch with an hardware endstop. In that case the endstop functionality has to be switched off for the homing procedure. 
* Probe: A measurement device, similar to a homeswitch, but further accoplished by a deploy/stow procedure. 
 * Can not be used as an hardware endstop. 
 * Can measure multiple points. 
 * Should be ignored while stowed.
 * Can be used as a homeswitch if you can find a save way/sequence to deploy/stow the probe without knowing the exact position - or you take the risk. 
 * Can share the physical switch with an hardware endstop. In that case the endstop functionality has to be switched off for the probe procedure. 
 * Can share the physical switch with a homeswitch. In that case the homing functionality has to be switched off for the probe procedure. 
 * A machine could have may of them. In Marlin we concentrate on only one - a Z-Min-Probe.
* Pin: A device where one or more other devices are connected to the board.

Try to find loopholes, exceptions, errors, ... . Feel free to add more statements.

CONSULitAS, 2016-05-18:
@AnHardt 
GREAT!

But a few minor additions:

>* Endstop (hardware)
  * An axis can have two of them.

* An axis *should* have two of them. (Two nc switches in series on one pin would be enough for security.)

>* Hitting an endstop is not homing - when homing we have a defined speed when the endstop is hit.

* when homing we have a defined speed and direction when the Homeswitch is hit.

>* Homeswitch

* Shouldn't we call it *reference switch* instead?
* there can be one or two for an axis: with two of them the printer could measure its boundaries:
  * home to home direction and then home to opposite direction and measure distance gives length of axis

In general: with nc switches chained in series you could reduce the used pins if necessary. So a z-moin-probe and a z-reference-switch could share a common pin which makes sense if the deployed probe (e.g. servo) switches earlier than the other switch. Should be used with z safe homing position (besides the bed) in this example.

CONSULitAS, 2016-05-18:
@AnHardt 
BTW: You should change the topic from Idea to Concept

AnHardt, 2016-05-18:
@CONSULitAS 
Added some of your ideas in the original posting.

'reference' instead of 'home'  -switch makes perfect sense if you have more than one. But i'm not sure if i want to extend the Marlin concept that far. (Don't make the concept to complicated). Besides of that measuring the axis length is a great idea.
> So a z-min-probe and a z-reference-switch could share a common pin ...

In case of a triggered when stowed probe this won't work - also not with an endstop.
Am i right when is say: "A triggered when stowed probe needs it's own pin." ? Likely not - with a 'no'-switch at the probe. Then the logic has to be reversed when switching from endsotp/homing- to probe- mode for the pin.

>  (Two nc switches in series on one pin would be enough for security.)

Yes. But that makes distinguishing them impossible. If you touch one of them, it's impossible to say what is the right direction. (Same for _reference_ switches). You can only make assumptions with the history of moves. When booting with a triggered switch we have a problem. (Or not - if we simply trow an error and do nothing else.)

jbrazio, 2016-05-18:
I suggest to change the name from "software endstop" to "virtual endstop", as both of them require software to work.

My main question is why do we still speak of "software" and "hardware" endstops ? They are exactly the same thing, actually are things like this `A security device to prevent from hitting an hardware endstop` and `A hit should never occur` that make them different. If the plan is to consolidate and rebuild then we should think out of the box and not apply the same old concepts.

thinkyhead, 2016-05-18:
> why do we still speak of "software" and "hardware" endstops?

The "software endstops" are just a set of stored numbers. Simply the min and max X, Y, and Z positions. The `clamp_to_software_endstops` function prevents planner moves from going past these limits. They don't need to have any relation to a hardware switch. A machine with no endstop switches at all (e.g., the QuadRap) may still have "software endstops."

Currently, if you disable `ENDSTOPS_ONLY_FOR_HOMING` the machine is supposed to abort printing if an endstop switch gets triggered. I'm not sure, but it's possible that if that is the case, you must make sure to move the carriage and bed so no endstops are triggered before you start a print, or the print might abort right away.

AnHardt, 2016-05-18:
@jbrazio 
I'm open for new and working concepts. (I'm biased - i have seen real CNC machines before.)

> why do we still speak of "software" and "hardware" endstops?

Because the one needs hardware. The name of the other is just to have an other name. If 'virtual' or 'software' makes no real difference to me, as long we can distinguish them. when we talk about them.

@thinkyhead 
> the machine is supposed to abort printing if an endstop switch gets triggered.

Sadly only implemented in the 'dev' repository. But i updated the branch for RCBugFix in my repository today (Have a look at https://github.com/AnHardt/Marlin/tree/lock-after-h-endstop-hit and decide yourselves). So it's only a few clicks away. If that is a good idea for a RC is an other question. I don't think so.

jbrazio, 2016-05-18:
Can't we abstract ?

```cpp
struct endstop {
  int x;          // X position
  int y;          // Y position
  int z;          // Z position
  bool reversed;  // All endstops by default are [choose NC or NO here],
                  // this reverses the logic
  bool home;      // Endstop has dual function: endstop/home switch
  uint8_t pin;    // Digital IO pin for a endstop                
  bool triggered; // Is is currently trigered ?
};
```

> Because the one needs hardware

Yes Andreas, but you also build a list of things that separate and hardware endstop from a virtual one.

> Endstop (hardware): A security device to stop the axis movement before something gets demaged.
> Endstop (software): A security device to prevent the carriage from hitting an hardware endstop, or a limit where we do not have a hardware endstop.
> Endstop (hardware): A hit should never occur.

Maybe this is kinda of "lost into translation" situation we're having here.

thinkyhead, 2016-05-19:
> Have a look at https://github.com/AnHardt/Marlin/tree/lock-after-h-endstop-hit and decide yourselves

@AnHardt It looks pretty straightforward to me. As long as we can ensure no false-triggers when first starting the print, I don't think it would be a very disruptive change. Do you know why it was restricted to SD printing in the first place?

AnHardt, 2016-05-19:
@thinkyhead 
> As long as we can ensure no false-triggers when first starting the print,

You got the point.
That's one of the reasons for the 'hardware endstop selftest' witch also should ensure, the endstops are untriggered after `setup()`. Sadly the selftest completely relies on, you can distinguish between a min and a max endstop - so 'chaining' (see somewhere above) the endstops and this test don't work together.
The second required component is, to not sett the software endstops at exactly the same place as the hardware endstops, but a tiny bit further inwards (z-min is a special case with all that offsets).

AnHardt, 2016-05-19:
@jbrazio 
Your `struct` is exactly what i wanted to avoid. It's mixing the different functions of a pin/switch together.

More like
```
struct physical_switch {
  uint8_t pin;    // Digital IO pin for a hardware endstop                
  bool reverse;   // this reverses the logic // even this could be shifted to a higher level
  bool triggered; // Is is currently triggered ?
  bool active;    // test it or not or set triggered or not
};
```
the single physical switches could be collected in an array or list named  `physical_switches`.  Hardware endstops, home switches, pobes would be a level above.

boelle, 2017-04-26:
@thinkyhead 

since this has gone inactive maybe close it. discussions can still continue after it closed.

closed 2018-02-06 by thinkyhead