MarlinFirmware/Marlin #3791: encoder randomly misses steps

ruggb, 2016-05-19:
turning the encoder one click USUALLY causes the cursor to move one line up or down.
However, randomly it will take 2 clicks of the encoder to move one line.
It almost always occurs on the 1st to 2nd line after going to a new page and sometimes occurs on a line in the middle of a page.
You may also want to note for clarity that "enabling  //#define REVERSE_ENCODER_DIRECTION
also changes the action of  //#define REVERSE_MENU_DIRECTION".

Blue-Marlin, 2016-05-19:
@ruggb 
> You may also want to note for clarity that "enabling //#define REVERSE_ENCODER_DIRECTION
also changes the action of //#define REVERSE_MENU_DIRECTION".

Thank you for proving our current description of the setting is correct.

> randomly it will take 2 clicks of the encoder to move one line

We could check the encoder pin more frequent, but then we have less time for the important things like printing. That's not what we want.
One day i'll try to rewrite Marlin to use pin_change interrupts for the digital inputs (encoder, buttons, endstops). That may improve things like that.

ruggb, 2016-05-19:
Yes, it is technically correct, as are most all your verbiage. And as most, it is based on the assumption that a non-technical user can read your mind and interpret it as a programmer might. Just saying, someone with a user perspective might reword some of this stuff for clarity. I am technical, but not a programmer, and I didn't comprehend it till mine didn't work right and I had to figure out why.

at the time I observed the 2 clicks, the machine was idle.

Blue-Marlin, 2016-05-19:
@ruggb 
Ok. Fou you.
There is no fix in sight. Try to live with the problem.

jbrazio, 2016-05-19:
@ruggb you may submit a improvement to the comment section.

ruggb, 2016-05-20:
@jbrazio - it is included in quotes in the original issue statement.
**"enabling //#define REVERSE_ENCODER_DIRECTION
also changes the action of //#define REVERSE_MENU_DIRECTION"**

thinkyhead, 2016-05-20:
Maybe something like…
```cpp
// Everything is reversed? REVERSE_ENCODER_DIRECTION only
// Only Menu Navigation is reversed? REVERSE_MENU_DIRECTION only
// Only Value Editing reversed? Both REVERSE_ENCODER_DIRECTION and REVERSE_MENU_DIRECTION
```

jbrazio, 2016-05-20:
@thinkyhead seems straightforward.

ruggb, 2016-05-20:
maybe more like this
```
// enabling REVERSE_ENCODER_DIRECTION reverses the rotation of the encoder for scrolling and value setting
// enabling REVERSE_MENU_DIRECTION only reverses rotation for scrolling
// enabling both will reverse value setting and leave scrolling the same as both disabled.
```

thinkyhead, 2016-05-20:
@ruggb My version requires less thinking. ;-)

ruggb, 2016-05-20:
If you say so. But I would at least define EVERYTHING and REVERSED
EVERYTHING = Menu Navigation & Value Editing
REVERSED = turning the knob produces an action backwards of what you expect.
some ppl might expect a "reversed" operation.

```
// If encoder knob produces an action reversed from what you expect for:
// Menu Navigation & Value Editing > Enable REVERSE_ENCODER_DIRECTION.
// Only Menu Navigation > Enable REVERSE_MENU_DIRECTION.
// Only Value Editing > Enable BOTH options.
```

thinkyhead, 2016-05-21:
The user will know when something is "reversed" — and some of that depends on their preference. The verbiage I ended up going with should be more than clear enough for the average bear:
```cpp
/**
 * Encoder Direction Options
 *
 * Test your encoder's behavior first with both options disabled.
 *
 *  Everything is reversed?           Enable REVERSE_ENCODER_DIRECTION.
 *  Only Menu Navigation is reversed? Enable REVERSE_MENU_DIRECTION.
 *  Only Value Editing is reversed?   Enable BOTH options.
 */
```

ruggb, 2016-05-21:
yea, that is much better with the heading line.

thinkyhead, 2016-05-21:
Yeah, the heading helps. Also I took your suggestion to drop "Everything" and went with…
```cpp
/**
 * . . .
 *  Reversed Value Edit and Menu Nav? Enable REVERSE_ENCODER_DIRECTION.
 *  Reversed Menu Navigation only?    Enable REVERSE_MENU_DIRECTION.
 *  Reversed Value Editing only?      Enable BOTH options.
 */
```

WZ9V, 2016-05-21:
I can confirm the missing clicks on RC6, I have also noticed that sometimes when editing settings such as temperature it skips, so I can for example get 199C and 201C but can't stop at 200C.  I have not been able to nail down a particular set of steps to make it happen.

closed 2016-05-30 by jbrazio