MarlinFirmware/Marlin #3792: Z going below 0

MoonshineSG, 2016-05-19:
In my Configuration.h i have `#define Z_MIN_POS 0`. 
Home Z:
```
Send: G28 Z
Recv: X:105.00 Y:10.00 Z:3.60 E:0.00
```
I use a capacitive proximity sensor and M851 is
```
Send: M851
Recv: echo:Z Offset : -3.60
```

I can do a G1 Z0
```
Send: G1 Z0 F3000
Recv: ok
Send: M114
Recv: X:105.00 Y:10.00 Z:0.00 E:0.00 
```
The nozzle touching (or almost touching) the bed

All good so far.

But.... 

```
G1 Z-5
Recv: ok
Send: M114
Recv: X:105.00 Y:10.00 Z:-3.60 E:0.00 
```
Regardless what value I send for Z which is < -3.6 the head stops at -3.6, or whatever value I have set for M851. 
Since I have `#define Z_MIN_POS 0`, why is the head moving below 0 ? 

Is my understanding of Z_MIN_POS wrong ? Isn't that the min value for Z ?

Blue-Marlin, 2016-05-19:
That's a problem we hope to have fixed in `RCBugFix`.
If not please check if you have defined
```
#define min_software_endstops true
```

MoonshineSG, 2016-05-19:
I have `#define min_software_endstops true` in Configuration.h

Blue-Marlin, 2016-05-19:
Do you home with the probe?
Does it make a difference if you make the test before or after G29? If so - mesh, grid or 3-point?

MoonshineSG, 2016-05-19:
> Do you home with the probe?
i only have the probe as Z stop. 

I do grid, but let me test if it makes any dif before and after G29

Blue-Marlin, 2016-05-19:
Hmm ja. I see. The patch i mend only corrected to software endstops for G92.
Adding the probe-z-offset is no big thing - but finding the right conditions for that to not break something else.

MoonshineSG, 2016-05-19:
I can imagine... Let me know if you need some help testing (when you have smth to test :D ) ...
For now I guess I have to make sure nothing breaks...

jbrazio, 2016-05-24:
@Blue-Marlin is this a bug we must fix ?

jbrazio, 2016-05-30:
@Blue-Marlin will you care to submit a patch for this bug ?

Blue-Marlin, 2016-05-30:
Likely the error is about here (https://github.com/MarlinFirmware/Marlin/pull/3829/files#diff-1cb08de130a6ece2d1b5b9c37bcfef48R1295). @thinkyhead is already in that area.

jbrazio, 2016-05-30:
@thinkyhead I will assign this one to you then.

thinkyhead, 2016-05-31:
> Is my understanding of `Z_MIN_POS` wrong? Isn't that the min value for Z?

@MoonshineSG The software endstop for Z currently gets extended based on any negative **probe offset** and based on any negative `home_offset` (as set with `M206`). I'm pretty sure these adjustments to the software endstops are no longer required, so I'm removing them in #3829 (which is mainly to fix an issue where switching extruders could cause movement beyond the software endstops).

You could experiment with the same idea to see if it fixes the issue for you. Simply make this change to the `clamp_to_software_endstops` function in `Marlin_main.cpp`:
```diff
-    float negative_z_offset = 0;
-    #if ENABLED(AUTO_BED_LEVELING_FEATURE)
-      if (zprobe_zoffset < 0) negative_z_offset += zprobe_zoffset;
-      if (home_offset[Z_AXIS] < 0) {
-        #if ENABLED(DEBUG_LEVELING_FEATURE)
-          if (DEBUGGING(LEVELING)) {
-            SERIAL_ECHOPAIR("> clamp_to_software_endstops > Add home_offset[Z_AXIS]:", home_offset[Z_AXIS]);
-            SERIAL_EOL;
-          }
-        #endif
-        negative_z_offset += home_offset[Z_AXIS];
-      }
-    #endif
-    NOLESS(target[Z_AXIS], sw_endstop_min[Z_AXIS] + negative_z_offset);
+    NOLESS(target[Z_AXIS], sw_endstop_min[Z_AXIS]);
```

MoonshineSG, 2016-06-01:
yes, that works. Now my Z doesn't go below 0 even if i have a negative z offset set by M851

closed 2016-06-23 by jbrazio