MarlinFirmware/Marlin #3795: Cyclops HotEnd support

MoonshineSG, 2016-05-19:
Is there support for Cyclops hotend ? I found some reference to `SINGLENOZZLE` but it looks like it hasn't made it to the main repo. 
Any hints on how to get this working ?

jbrazio, 2016-05-19:
I don't believe it is working, if I do a search for `SINGLENOZZLE` it not found anywhere.

thinkyhead, 2016-05-20:
The MarlinKimbra fork implements `SINGLENOZZLE`. That isn't a current option in the main Marlin project yet. We are working on Mixing Extruder support, which requires similar hooks to do proportional mixing. If the Cyclops can do proportional mixing with both feeds coming in at the same time, then the Mixing Extruder stuff can be adapted for it. We need to have a look at the GCode for the Cyclops. If it uses `T0` and `T1` commands to switch the filament inputs (or mix) then it should work very well with this upcoming feature.

Sebastianv650, 2016-05-20:
E3d says mixing: no. So why not treat it as independent extruders sitting in the same location?

thinkyhead, 2016-05-20:
@Sebastianv650 If it can't do mixing, but has two E steppers, then wee have a choice. We could adapt the `SINGLENOZZLE` feature, which does work like that (so `EXTRUDERS == 2`). Or, we could try to be a little more more formal in how we define extruders, and actually have the concept of a single extruder (`EXTRUDERS == 1`) with multiple input feeds (e.g., `E_STEPPERS == 2`), which is more of the approach taken by `MIXING_EXTRUDER_FEATURE` (which uses `MIXING_STEPPERS >= 2`).

The `SINGLENOZZLE` approach is probably simpler to implement.

The advantage of the second approach is that it would allow `DUAL_X_CARRIAGE` with two Cyclops extruders, thus giving 4-color support.

thinkyhead, 2016-05-20:
Ah, what the heck. #3808

esenapaj, 2016-05-21:
It seems that PR #3808 isn't only support [Cyclops](http://e3d-online.com/Cyclops) but also support [Diamond Hotend](http://reprap.org/wiki/Diamond_Hotend)(but only with separated colour mode).
Is it right?

thinkyhead, 2016-05-22:
@esenapaj Yeah, it will support any hotend with multiple E steppers, but yes, only in separate-color mode. Proportional mixing is a little more complicated, but also ready to go.

MoonshineSG, 2016-05-23:
this is good. It also means that when you change from a dual nozzle to single you need to load new firmware... switching between single/dual with a special M code would be awsome... but maybe a bit complicated to implement, so firmware switch it is... :D 

Have ready build 2 firmwares... get Simplify 3D to insert a tag to identify cyclops...  then use an octoprint plugin to switch the firmware based on the tag... not too bad...

jbrazio, 2016-05-24:
Closing the issue: Feature PR was merged.

closed 2016-05-24 by jbrazio