MarlinFirmware/Marlin #3799: Upgrading Between Firmware

chevellebro1, 2016-05-19:
Is there an easy way to copy the changes between the Marlin firmware so I can upgrade to the newest without loosing all my settings?

jbrazio, 2016-05-19:
You have to manually copy all the settings from the previous `Configuration.h` to the new one.
Please pay attention that some settings may have changed their names.

adamfilip, 2016-05-20:
try a comparison software called Meld. it can help see differences between two files.or a folder of files. make it much easier to transition between builds :)

thinkyhead, 2016-05-21:
Using a "diff" tool like Meld, BBEdit, or Kaleidoscope is good. But you will need to compare your configs to the original unmodified `Configuration.h` and `Configuration_adv.h` files from the version of Marlin that you started with.

Roxy-3DPrintBoard, 2016-05-21:
> Please pay attention that some settings may have changed their names.

We have a whole bunch of names that need to change.   It wouldn't hurt to make those changes prior to going 'Golden'.

thinkyhead, 2016-05-21:
@Roxy-3DPrintBoard Do you have a list of the names needing a change? We should have a look and vote on the most important ones. I'd like to minimize the number of names we change at this point, but I'm open to discussion about that.

closed 2016-05-24 by jbrazio