MarlinFirmware/Marlin #3804: Print Drifting due to Step Pulse Width?

bmalcheski, 2016-05-20:
From the looks of it, this has been discussed a lot in the past but I cannot find a definitive answer as to how to increase the step pulse "1" and "0" time. I have tried two different standalone stepper drivers and I believe both have had issues related to the pulse timings. I'm currently (and would really like to continue using as I am building a large machine, plus the Gecko drives are very quiet ) using Gecko G213V drivers which require 1 microsecond on and 2 microseconds off. http://www.geckodrive.com/g213v.html

My current guess (which is likely way wrong as I have very little coding knowledge) is to add delays into stepper.cpp starting around line 719.

```
#define _COUNTER(axis) counter_## axis
      #define _APPLY_STEP(AXIS) AXIS ##_APPLY_STEP
      #define _INVERT_STEP_PIN(AXIS) INVERT_## AXIS ##_STEP_PIN

      #define STEP_ADD(axis, AXIS) \
        _COUNTER(axis) += current_block->steps[_AXIS(AXIS)]; \
        if (_COUNTER(axis) > 0) { _APPLY_STEP(AXIS)(!_INVERT_STEP_PIN(AXIS),0); }

      STEP_ADD(x,X);
      STEP_ADD(y,Y);
      STEP_ADD(z,Z);
      #if DISABLED(ADVANCE)
        STEP_ADD(e,E);
      #endif

      #define STEP_IF_COUNTER(axis, AXIS) \
        if (_COUNTER(axis) > 0) { \
          _COUNTER(axis) -= current_block->step_event_count; \
          count_position[_AXIS(AXIS)] += count_direction[_AXIS(AXIS)]; \
          _APPLY_STEP(AXIS)
        delayMicroseconds(2)
        (_INVERT_STEP_PIN(AXIS),0); \
        delayMicroseconds(2)

```

brainscan, 2016-05-20:
Not sure if this is related or helps #1394 @arno-millenaar  might have done what you need.

Sebastianv650, 2016-05-20:
According to #1394 you shouldn't have a problem with your required 1-2µs. If you could test the solution over there and it works, I think it's worth a pull request!

bmalcheski, 2016-05-20:
I did some oscilloscope testing with the machine today. Unless I was doing something wrong, the pulse widths were about 10µs. I'll have to do some more investigating to confirm. It looks like the code in stepper.cpp is different than it was when #1394 was discussed, but I would really like to try it if someone can direct me on how to properly implement it in the current code.

jbrazio, 2016-05-20:
If you have a noisy signal I would advise first to try to correct that, do you have ferrite beads on the stepper cables ?

bmalcheski, 2016-05-20:
I do not have any yet. I just ordered some. Thanks for the reminder!

jbrazio, 2016-05-30:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

bmalcheski, 2016-05-30:
I tried installing ferrite beads but they did not resolve the issue. While it appears they may have helped reduce the electrical noise it may still be too much and I need to try reducing the length of the power carrying cables to the motors.

As a note, the pulse widths that I have been measuring are about 10µs. This should be plenty long, but I am wondering if anyone can confirm what they are expected to be. In the past it was less than 2µs if I recall correctly.

bmalcheski, 2016-06-14:
Unfortunately I have not made any progress on this. I was making efforts to reduce the noise on the signal lines by shortening the motor cables and installing filters on the signal lines but one of my drives stopped working. In frustration I switched back to the N17s and SD8825 drivers. They just barely get by and I will use them for now. Hopefully I can rule out electrical noise in the future. For now my issue can be close though.

closed 2016-06-14 by bmalcheski