MarlinFirmware/Marlin #3810: Z axes wont move, only buzzes

Ciev, 2016-05-21:
Hi, 

Just updated the 1.1.0-RC6 firmware to my printer, the board is a GT2560 basically an arduino mega 2560 + Ultimaker + Ramps1.4 I live you this link if you wan to read deeper (http://www.geeetech.com/wiki/index.php/GT2560). 
Long story short I've ported the stock firmware configuration to RC6. All is working but the z axis that each time that I make it move buzzes, not sure if it's a bug of this version or I've done something wrong in the configuration. if you wanna check the configuration file, you can find it attached. 

Thanks for your help.

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/276048/Configuration.h.zip)

jbrazio, 2016-05-21:
If you are positive your `PINS`, hardware micro-stepping and current are correct.. you should validate if the following options in `Configuration.h` are correct:

```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,25600,93}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {400, 400, 2, 45}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {5000,5000,50,5000}    // X, Y, Z, E maximum start speed for 
```

In the wiki page link you shared I've downloaded the Marlin version there and the values are completely different from the ones above.

thinkyhead, 2016-05-21:
Usually when I hear a Z axis is buzzing, I think it's probably trying to move too fast. So I would assume the max feedrate for Z or the homing feedrate were too high. But your Z speeds seem pretty low (compared to mine). These speeds can also be changed by GCode and saved to EEPROM, and you can change them from an LCD controller also. So you might try messing with the Z rates just to see if lowering them helps.

I would also check the microstepping and voltage on the Z steppers, of course. There _have_ been some additions to Marlin for digital control of stepper motor voltage, but that should have no effect on a RAMPS.

Are you able to test with `RCBugFix` to see if it fares any better?

Ciev, 2016-05-22:
@jbrazio @thinkyhead  Thanks for your reply, it was actually a problem with the:
`#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,25600,93} `, the value 25600 was 10 times greater than supposed to, fixed the value to 2560, loaded it to the board and voila... Thanks for your help... I'd checked the code several times but never noticed the extra 0.

I'll let you know if I find something odd with this RC.

closed 2016-05-22 by Ciev