MarlinFirmware/Marlin #3814: dual z endstop Can not work

linjinc, 2016-05-22:
RC6 dual z endstop Can not work
RC5 is ok

linjinc, 2016-05-22:
Z  motor G28 fault
Z move normal

jbrazio, 2016-05-22:
Could you explain a bit better ? I'm not understanding your issue.

linjinc, 2016-05-22:
I use google translation
Z motor+z2 motor reset when turned only stopped a few laps
RC5 or RC4 do not have this problem

jbrazio, 2016-05-22:
Try to translate simple sentences.

1) is this related with Z endstop ?
2) is this real malted with your dual Z stepper motors ?

linjinc, 2016-05-22:
yes

thinkyhead, 2016-05-23:
How is `RCBugFix` in regard to this issue?

Also, be sure to double-check your settings for `DEFAULT_MAX_FEEDRATE` and `HOMING_FEEDRATE` in case they are allowing the Z axis to move too fast.

Which endstops are you using for your dual Z endstops? You will need to set `USE_ZMIN_PLUG` for one of them, then `USE_XMAX_PLUG` perhaps for the other one (it will depend on where you connected the second Z endstop).

You should post your configuration so we can check for any other possible missed settings.

linjinc, 2016-05-24:
[image]
[image]
The problem is that the phenomenon is G28 then Z1 motor and Z2 motor  stop

linjinc, 2016-05-24:
Dual Z-endstop not working on  RCBugFix

linjinc, 2016-05-24:
[image]
[image]

thinkyhead, 2016-05-24:
I see you have all your endstop plugs defined as in-use. Do you actually have all 6 endstop plugs attached to switches? From what I can see, your machine is set up like…
- **`XMIN`**: Nothing
- **`XMAX`**: X Endstop
- **`YMIN`**: Z2 Endstop
- **`YMAX`**: Y Endstop
- **`ZMIN`**: Z1 Endstop
- **`ZMAX`**: Nothing (Z probe?)

You should turn off any that aren't used. For example, if you're not using the X Min endstop at all, you can disable `USE_XMIN_PLUG`. I see you are using Y Min for your Z2 Min, so it's ok to leave `USE_YMIN_PLUG` enabled. Unless your Z Max is connected to a probe, you might want to disable `USE_ZMAX_PLUG` too.

linjinc, 2016-05-27:
Has been test,not working

boelle, 2016-05-27:
read what thinkyhead is asking you

how many endstop do you have?

boelle, 2016-05-27:
https://github.com/repetier/Repetier-Firmware/issues/528

it seems that you have made the feature work on repetier....

so if you know how to fix this issue then why not submit a PR ?

dp90pusika, 2016-06-05:
hey everyone,
I have a similar problem as @linjinc 
I have uncommented the #define Z_DUAL_STEPPER_DRIVERS and #define Z_DUAL_ENDSTOPS
set my _ZMAX_ as second z endstop

The problem is that when homing Z axis the bed moves away from the sensors for a few millimeters and stops. It does this regardless of the endstops physically triggered or not. It's like the homing bump but nothing after that. Or I'm guessing something's with the endstops because I get similar movement when an endstop is triggered and after a homing command the axis moves but not enough to disingage the endstop.

Endstops read correctly for M119.
[image]

Endstop settings:
`#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_PROBE_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
`
My board is a Ramps1.4 
I'm using an extra stepper on a breakout board, because the second extruder is in use.
`// The Z2 axis, if any, should be the next open extruder port
#if ENABLED(Z_DUAL_STEPPER_DRIVERS) && !defined(Z2_STEP_PIN)
  #define Z2_STEP_PIN   47
  #define Z2_DIR_PIN    45
  #define Z2_ENABLE_PIN 32
  #undef _Z2_PINS
  #define _Z2_PINS Z2_STEP_PIN, Z2_DIR_PIN, Z2_ENABLE_PIN,
#endif
`
If I only use the dual stepper drivers it works just fine, the problem occurs when the dual endstops are activated. 
Also I have dual x carriage setup, but it doesn't seems to affect the behavior

When can I start the search? 
My guess is that, the firmware thinks the endstops are triggered and tries to move away from them, but they never deactivate. If I send an M119 it clearly indicates the right state of the endstops

Roxy-3DPrintBoard, 2016-06-05:
My guess is you have one of the dual endstops set wrong with regard to inverted or not.   With your nozzle in the middle of the bed and 50mm above the bed, what does an M119 say?  You should have nothing triggered.  If something is triggered you probably need to invert that endstop.

(I don't know that this is happening...  But it kind of sounds like the firmware thinks an endstop is triggered even though it is not.  So it tries to back away from the endstop.)

dp90pusika, 2016-06-05:
nozzles in the middle of the bed, bed lowered 50mm and I get this:
[image]
the soldering and cables are in good shape, that's the part that I can handle, something is up with the programing part.

I don't really know what I'm looking at here, but it seems that whatever I chose it will invert the signal. Just guesswork, but is this part ok?

    Z_DUAL_ENDSTOPS endstop reassignment
   */
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define _XMIN_ 100
    #define _YMIN_ 200
    #define _ZMIN_ 300
    #define _XMAX_ 101
    #define _YMAX_ 201
    #define _ZMAX_ 301
    const bool Z2_MAX_ENDSTOP_INVERTING =
      #if Z2_USE_ENDSTOP == _XMAX_
        X_MAX_ENDSTOP_INVERTING
        #define Z2_MAX_PIN X_MAX_PIN
        #undef USE_XMAX_PLUG
      #elif Z2_USE_ENDSTOP == _YMAX_
        Y_MAX_ENDSTOP_INVERTING
        #define Z2_MAX_PIN Y_MAX_PIN
        #undef USE_YMAX_PLUG
      #elif Z2_USE_ENDSTOP == _ZMAX_
        Z_MAX_ENDSTOP_INVERTING
        #define Z2_MAX_PIN Z_MAX_PIN
        #undef USE_ZMAX_PLUG
      #elif Z2_USE_ENDSTOP == _XMIN_
        X_MIN_ENDSTOP_INVERTING
        #define Z2_MAX_PIN X_MIN_PIN
        #undef USE_XMIN_PLUG
      #elif Z2_USE_ENDSTOP == _YMIN_
        Y_MIN_ENDSTOP_INVERTING
        #define Z2_MAX_PIN Y_MIN_PIN
        #undef USE_YMIN_PLUG
      #elif Z2_USE_ENDSTOP == _ZMIN_
        Z_MIN_ENDSTOP_INVERTING
        #define Z2_MAX_PIN Z_MIN_PIN
        #undef USE_ZMIN_PLUG
      #else
        0
      #endif
    ;
  #endif

Roxy-3DPrintBoard, 2016-06-05:
Do you really have 6 end stop switches  plugged into your controller board?   That is what M119 is claiming.   Can you send us a picture of the wires plugged into your controller board?

dp90pusika, 2016-06-05:
Of course I can:
[image]
X_MIN is in normal use
X_MAX for X2Carriage homing
Y_MIN normal use
Y MAX endstop is not in use, but if I pul it is it really going to make a difference?
Z_MIN normal use
Z_MAX for second Z endstop in dual Z mode

Z2_MAX appeared in the M119 list after Z dual endstop was uncommented. If I trigger the second Z endstop, both Z_max and Z2_max gives back triggered for M119
[image]

thinkyhead, 2016-06-07:
@linjinc If `Z_DUAL_ENDSTOPS` works in `RC5` but not in `RC6` or `RCBugFix`, then we should have a closer look at the code that handles Dual Z Endstops and see what changed after `RC5`. I know there have been some patches in the area of homing, and it's possible something was changed without realizing that it was needed for `Z_DUAL_ENDSTOPS` to work properly.

I will compare `RC5` to `RC6` first and see if anything obvious stands out.

thinkyhead, 2016-06-07:
And this is your Z2 endstop setting in `Configuration_adv.h`…?
```cpp
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _ZMAX_
  #endif
```

dp90pusika, 2016-06-07:
Meanwhile, I was fiddling with RC5 and I found out that Dual Z Endstops work only if both z endstops are the same type. I mean boath are MAX or MIN. So Z_MAX and Z2_MAX or Z_MIN and Z2_MIN, first I mixed up them because I tought that the Z2 is just a secondary sensor and the min or max doesent matter. I will test this on the RCBugFix but havent got time just yet.

@thinkyhead yes, it's the same

closed 2016-07-23 by jbrazio