MarlinFirmware/Marlin #3815: Full graphic controller issues under Arduino >1.0.6

Gaffalover, 2016-05-22:
I have a Reprap discount full graphic smart controller connected to a K8200 (3DRag derivative).
U8glib is installed and #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER uncommented.

When compiling RC6 with Arduino 1.6.9 I get weird artefacts on the display:
http://fpv-community.de/showthread.php?69744-Hilfe-bei-Marlin-Mega-2560-Ramps-1-4-Full-Graphic-Display&p=885201&viewfull=1#post885201

Under 1.0.6 (version check in SanityCheck.h commented) it compiles with no errors and the display is working fine.

My configuration.h: http://pastebin.com/L0B7iak6

jbrazio, 2016-05-22:
@Gaffalover could you try to remove you current U8glib and do the following command:
```
# Install: Monochrome Graphics Library for LCDs and OLEDs
arduino --install-library "U8glib"
```

Compile with 1.6.9 and check if the error is still there ? Yesterday night I compiled my Marlin with 1.6.9 without issues.

Gaffalover, 2016-05-23:
Thanks for your reply.
Just gave it another shot:
Uninstalled 1.0.6 and deleted the %USERPROFILE%/Documents/Arduino folder.
Installed 1.6.9 and did your command -> same result as shown in the picture.
Just 4 fun I reinstalled 1.0.6 without touching the library folder or Program (except commenting the version check in SanityCheck.h) an it works fine.

For me it looks like an issue with the IDE at versions >1.0.6.
I even tried installing U8glib 18.0.0 instead of 19.x and lcddisplay 1.0 but without success.
I couldn't find anything noticeable in the changelogs of the Arduino IDE.

jbrazio, 2016-05-23:
Do you mind trying with 1.6.8 ?

Gaffalover, 2016-05-26:
I even tried it with 1.6.5 and still no improvement.
In another forum somebody had exactly the same issue: http://fpv-community.de/showthread.php?69744-Hilfe-bei-Marlin-Mega-2560-Ramps-1-4-Full-Graphic-Display/page3
He used RC2 and 1.5.x with the same artefacts.

jbrazio, 2016-05-26:
And clicking on the encoder to go into the menu doesn't clear the error ?

SwiftNick, 2016-06-01:
I have the same problem using **Arduino 1.6.9** compiling **Marlin 1.1.0-RC6**  
configured for the **REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER**

It seems to be caused by the compiler optimisation line in **'ultralcd_st7920_u8glib_rrd.h'**
Change line 31 
FROM:
**#pragma GCC optimise (3)**

TO:
**//#pragma GCC optimize (3)**

This will disable the optimisation and the LCD should display correctly.

**NOTE: I haven't run full tests on this to see if printing is adversely affected but I suspect the later optimisation has messed up the timing when sending data to the display.**

Roxy-3DPrintBoard, 2016-06-01:
@general-maintainers   Guys, this might be our chance to figure out what was going wrong with building Marlin with certain versions of Arduino.     I may be wrong, but can't we remove that one line of code:
~~~~
#pragma GCC optimise (3)
~~~~~
from the .h file and place it further and further down in the .cpp file until things work again?    When things start working, we will know the location of the optimization failure and be able to figure out what keeps killing us????      

This strategy is very crude, because if the LCD file has multiple failures, we may not get a good build until we have the #pragma line below them all.   So if this strategy did produce results, we might want to change the code and start the search again to see if there are any other failures with the LCD file.

With that said...    With this new information, we might be able to figure out what has been killing us.

@SwiftNick  Thank You for the help!   Do you have any comments on what I just suggested?

Blue-Marlin, 2016-06-01:
Dear @Roxy-3DPrintBoard 
We are not killed. With 5 ST7920 from different sources we do not having any problems here.

Using Arduino 1.6.9 we do not see any speed differences but compiling without `#pragma GCC optimise (3)` saves about 2k of progmem.
With earlier Arduino versions there used to be a good speed impovement. (Don't ask what versions - versions from about a year ago)

So for now we can comment the `#pragma GCC optimise (3)` because it does not find faster code (just biger). But as soon it will do that again we definitely want to use that!

Using the standard configuration with just added `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
and idelecounter patch (https://github.com/AnHardt/Marlin/commit/fd20acceb6e74f587287108d26a05503093f6a78)
System just booted and nothing to do.
```
//#pragma GCC optimize (3)

Der Sketch verwendet 83.886 Bytes (33%) des Programmspeicherplatzes. Das Maximum sind 253.952 Bytes.
Globale Variablen verwenden 3.842 Bytes (46%) des dynamischen Speichers, 4.350 Bytes für lokale Variablen verbleiben. Das Maximum sind 8.192 Bytes.

23:45:05.556 : echo:idles/second: 10499
23:45:06.559 : echo:idles/second: 10520
23:45:07.559 : echo:idles/second: 10509
23:45:08.558 : echo:idles/second: 10509
23:45:09.557 : echo:idles/second: 10509
23:45:10.556 : echo:idles/second: 10521
23:45:11.560 : echo:idles/second: 10509
23:45:12.559 : echo:idles/second: 10505
23:45:13.558 : echo:idles/second: 10521
23:45:14.557 : echo:idles/second: 10497

#pragma GCC optimize (3)

Der Sketch verwendet 85.922 Bytes (33%) des Programmspeicherplatzes. Das Maximum sind 253.952 Bytes.
Globale Variablen verwenden 3.842 Bytes (46%) des dynamischen Speichers, 4.350 Bytes für lokale Variablen verbleiben. Das Maximum sind 8.192 Bytes.

23:47:25.140 : echo:idles/second: 10517
23:47:26.139 : echo:idles/second: 10514
23:47:27.138 : echo:idles/second: 10515
23:47:28.138 : echo:idles/second: 10527
23:47:29.141 : echo:idles/second: 10515
23:47:30.140 : echo:idles/second: 10512
23:47:31.139 : echo:idles/second: 10527
23:47:32.138 : echo:idles/second: 10502
23:47:33.138 : echo:idles/second: 10527
```

Roxy-3DPrintBoard, 2016-06-01:
> So for now we can comment the `#pragma GCC optimise (3)` because it does not (create) faster code (just big(g)er). But as soon it will do that again we definitely want to use that!

@Blue-Marlin Yes, but we have seen build problems on many versions of Arduino.  And not all of those problem scenarios had graphics controllers.   Isn't this an opportunity to find out what code constructs are causing us problems?   Maybe we have something declared wrong.  Or maybe we are linking something wrong.  Or maybe we have something else strange going on.    

Even though we can get around this (for the moment) by commenting out the optimization, it bothers me we have to do that just because we have no idea why that is necessary.

SwiftNick, 2016-06-01:
It may be worth investigating the changes in the optimisation algorithm in newer versions of gcc that are shipped with Arduino. I have known bugs in optimisers that do stupid things like take items out of loops to make them run faster! Something has definitely changed as compiling with older tools seems to work. 03 turns on all optimisations so if size is the main reason for the optimisation it may be worth tweaking the flag.

Roxy-3DPrintBoard, 2016-06-01:
If we knew which function required the relaxed optimization, we can compile it both ways and see what is going on.  It may be a bug in the optimization.   But it is also possible we have something declared in a way that the optimizer makes a bad decision (which I guess is still a bug).   If we understood what was going wrong, it may be that we can harden the code (every where...  Not just in the Graphics Controller code) to keep this from happening in the future.

epatel, 2016-06-01:
How about doing some binary search in the file in question using  push/pop options? Easiest first test would be just to move the "optimize" to the middle of the file.

```
#pragma GCC push_options
#pragma GCC optimize (3)

#pragma GCC pop_options
```

Roxy-3DPrintBoard, 2016-06-01:
Yeah, but I didn't know about push & pop.  That is cool!    If I had a graphics controller, I would be trying to isolate it.    I would still need help to get it to generate and leave the .ii files around.  Every time I need to look at the generated assembly code, I can never get the files to stick around.  But I know jbrazio can do it because he was posting instruction cycle timing for various code constructs.

jbrazio, 2016-06-01:
> -O3 turns on all optimizations specified by -O2 and also turns on the -finline-functions, -funswitch-loops, -fpredictive-commoning, -fgcse-after-reload, -ftree-vectorize and -fipa-cp-clone options. 

` -funswitch-loops` and `-fpredictive-commoning` will produce bigger code and apply some optimization that could break loops().

The only person who is able to test which exact function/piece of code breaks with `-O3` is @Blue-Marlin which seems to have access to this specific LCD.

Roxy-3DPrintBoard, 2016-06-01:
And... given what you just communicated...   Once we identified the function with the optimization problem, we could even just turn on one more optimization at a time for the function until we find the one causing problems.

jbrazio, 2016-06-01:
Yes we can either keep behavior of optimizing all and exclude that specific part, or has @Blue-Marlin suggested disable all the optimization has it seems not to have a positive effect with the latest gcc toolset.

Concerning this being the reason for other IDE issue we had.. I don't believe so and this only activates optimization for this file and I did a quick grep a no other `#pragma` were found.

SwiftNick, 2016-06-02:
I've done the binary search and the optimisation fails for the first function in the file:
If you move the push just below and the pop at the end of the file, to exclude this function the display works just fine.

static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU == 20000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU == 20000000
      __asm__("nop\n\t""nop\n\t");
    #endif
  }
}

Roxy-3DPrintBoard, 2016-06-02:
Can you selectively turn on more of those flags for the first function and find out which one is causing the problem?   

And then using that information we may need to look at the generated assembly code.   But there is very little going on there.  It might be a timing problem.   Or maybe the GCC is optimizing away one of the writes to ST7920_CLK_PIN.  (Don't have time to look at those macros right now.  But will in the morning.)

Blue-Marlin, 2016-06-02:
@SwiftNick 
What board do you use?

SwiftNick, 2016-06-02:
The 3drag with the ATMega2560 (shipped with the Velleman k8200)
http://reprap.org/wiki/File:3Drag_controller_top.jpg

Blue-Marlin, 2016-06-02:
@SwiftNick 
Please make a test with O3 but activated nop's.

jbrazio, 2016-06-02:
I would bet on the `WRITE(ST7920_CLK_PIN,0);` and `WRITE(ST7920_CLK_PIN,1);` ending out of the cycle due to `-fpredictive-commoning`. But your suggestion is interesting.. you believe optimization is going "too fast" for bit banging so it requires the delay.

Blue-Marlin, 2016-06-02:
@jbrazio 
I suppose a bad display, to long cables, to much noise or a wrong crystal on the board - a hardware problem what requires a relaxed timing. 
If @Gaffalover and @SwiftNick have the same problem and both have a K8200, but "noone" else sees the problem, i suppose a systematic problem, like too long cables, with the K8200. Than a fix for K8200 is better then maybe slowing down all ST7920 displays.

In theory the impact of adding 3 nop's should be smaller than not unrolling the loop (about 5 nop's -  increase i in a register, jump back, compare i). I'm just interested in how far off some K8200's are.

SwiftNick, 2016-06-02:
I thought it was those too so I commented them out and the screen worked just fine.
The ATMega2560 has a 16Mhz clock so I guess they are disabled in my case anyway by the F_CPU definition.

SwiftNick, 2016-06-02:
Right final comment tonight then I'm going to bed!
The 3drag board runs at 16Mhz so the nop's are disabled.
if I enable the nop's for 16Mhz+ the display works just fine with the optimisations enabled.
so my function now looks like this: and the optimisation is enabled.

static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU >= 16000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU >= 16000000
      __asm__("nop\n\t""nop\n\t");
    #endif
  }
}

Roxy-3DPrintBoard, 2016-06-02:
> But there is very little going on there. It might be a timing problem. 

So does it look like a timing problem?  How do I get it to generate and leave the .ii files around so I can compile it both ways and see the difference?

SwiftNick, 2016-06-02:
You can find the object files in the Arduino temp build folder 
C:\Users\<your username>\AppData\Local\Temp\build<a-unique-number-for-your-build>.tmp
As this is a header file there is no individual object file created. From what I can figure out the changes end up in 'ultralcd.cpp.o'.
If its any use I have the 2 versions of the file here with optimisation enabled, one with the nop and one without. There are a lot of differences between the two files though so good luck!
[nop-or-no-nop.zip](https://github.com/MarlinFirmware/Marlin/files/295240/nop-or-no-nop.zip)

CONSULitAS, 2016-06-02:
@jbrazio @SwiftNick @Blue-Marlin 

I would think this is a different problem than a hardware problem with the cables.

Lets collect what we know now:
* For older Arduino IDEs it works well.
* With newer Arduino versions you need to deactivate optimizations or enable the ASM-NOP sequences for 16 MHz, too.
* For older IDEs you need the ASM-NOP sequences for 20 MHz boards, for 16MHz or slower they are omitted.
* NOPs are used to pause the program one cycle because it would be to fast. Mostly they are put in by trial and error.
* Optimization makes programs faster.
* The new GCC is sometimes significantly faster with activated optimization.

Conclusion(s):
* The new IDE speeds up the code.
* For fast execution the code needs NOPs.
* Now it needs them for 16 MHz, too.

@SwiftNick I have a K8200 too, but only with LCD2004 display module. 

Could you please test the following:

1)
  * first `#if F_CPU` to `== 20000000`
  * second `#if F_CPU` to `>= 16000000`:

```
static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU >= 20000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU >= 16000000
      __asm__("nop\n\t""nop\n\t");
    #endif
  }
}
```

2)
  * first `#if F_CPU` to `>= 16000000`
  * second `#if F_CPU` to `>= 20000000`:

```
static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU >= 16000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU >= 20000000
      __asm__("nop\n\t""nop\n\t");
    #endif
  }
}
```

3)
  * first `#if F_CPU` to `== 20000000`
  * second `#if F_CPU` to `== 16000000`: with only one NOP, two NOPs for 20MHz

```
static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU >= 20000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU >= 16000000
      __asm__("nop\n\t");
    #endif
    #if F_CPU >= 20000000
      __asm__("nop\n\t");
    #endif
  }
}
```

4)
  * first `#if F_CPU` to `>= 16000000`
  * second `#if F_CPU` to `== 16000000`: with only one NOP, two NOPs for 20MHz

```
static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #if F_CPU == 16000000
      __asm__("nop\n\t");
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #if F_CPU >= 16000000
      __asm__("nop\n\t");
    #endif
    #if F_CPU >= 20000000
      __asm__("nop\n\t");
    #endif
  }
}
```

We should then use the fastest running variant.

BTW: This should not work on the Arduino DUE, too. Definitely a point for the HAL.
BTW 2: The `== 20000000` is stupid: If you have 25 MHz, it won't work. But definitely need NOPs...

Gaffalover, 2016-06-02:
Wow, all you guys got my biggest respect!
I tried all 4 of CONSULitAS snippets:
1 trough 3 generated slightly different artefacts, but #4 did the trick!
Fully operational display! :)
Thank you so much.

Blue-Marlin, 2016-06-02:
@CONSULitAS 
I agree with your statements only if you are explicitly talking about not working K8200 with 3drag. For all other configurations we do not have any hint for misbehaviour. My displays run flawlessly even with a more aggressive timing.

jbrazio, 2016-06-02:
@Blue-Marlin if we can create a new baseline which works for all having one additional "nop" I would prefer it rather than spaghetti #ifdef..

Roxy-3DPrintBoard, 2016-06-02:
I'm willing to get beat up for this...  But now that it works, I would add an extra NOP or two at each of those locations.

Blue-Marlin, 2016-06-02:
Next week you will put delays in the stepper interrupt,for all, because a new stepper driver is too slow.

jbrazio, 2016-06-02:
That's not a logic reasoning from you..
 1. We're discussing **one** nop
 2. This is the LCD vs. the stepper which is critical code

Blue-Marlin, 2016-06-02:
We are discussing two from some 10000 displays.

Roxy-3DPrintBoard, 2016-06-02:
> Next week you will put delays in the stepper interrupt,for all, because a new stepper driver is too slow.
@Blue-Marlin  You not being fair and objective.  Remember...  You were the one that was totally comfortable with turning off all optimization to slow ALL OF THIS CODE down to keep the problem from happening.   

> We are discussing two from some 10000 displays.

Is it possible that it is not the displays, but instead we are seeing noise and/or ringing on the signals going to the display?   If it is only 2 out of 10000 that might be explain things better.

CONSULitAS, 2016-06-02:
@Blue-Marlin 

* How many MHz has your board? Perhaps 20MHz?
* Did you check your conclusion that it is noise on the lines with a scope?
* Did you check that you have the same hardware version and batch of the display like @SwiftNick and @Gaffalover ?

BTW:
* Marlin tries to work for **every hardware combination**.
* This is a verified bug (see labels).
* How do you know that it works on 10000 displays?
  * You have one (working).
  * I have one (working, but different model).
  * thinkyhead has one...

When i found the M117 bug which has broken SD file access, too, we had only a few complains. This has more than one reason:
* Only few users know about the Marlin Github repository and how to flash theirselves.
* Fewer dare to use a "non stable" version.
* Many of the rest say "Oh a bug, lets try an other version".
* Many don't know to make an issue and complain in other forums (ask foosel and others).
* ...

So
* We have found a (verified) bug.
* We have a solution.
* We know why my proposed code works (timing can be really critical).
* This will not affect overall speed of Marlin.

What the hell is the problem now?

Perhaps we go to
* http://opensoul.org/99ways/#50
* http://opensoul.org/99ways/#99

Sorry, but i feel that your feedback is not fair and objective, too.

@jbrazio @Roxy-3DPrintBoard 
I think we need a pull request right now. Shall i, will you?

Roxy-3DPrintBoard, 2016-06-02:
@CONSULitAS Please prepare a pull request that conforms to the fourth solution that seems much more reliable.   I don't have one of these boards.  But I would feel more comfortable if we could check to see if it is a signal / noise issue.   The problem is we need one that is failing to be at the same place as somebody with an Oscilloscope.  

@Gaffalover  Would you mind sending us some pictures of your cabling going to the display?   Is the wiring close to or running for some distance next to high current lines like the stepper motor wires?

Blue-Marlin, 2016-06-02:
Do what you want. I'm perfectly able to change it to my taste.

I'm very sorry for my obviously to bad writing. Most of you seem to read things in my messages i have never written, or do miss things, where i think, i have written about. Or is it your reading?

> With __5__ ST7920 from different sources we do __not having any problems__ here.

> Using the __standard configuration__ with just added REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER    

> So for now we can comment the #pragma GCC optimise (3) because it does not find faster code (just biger). __But as soon it will do that again we definitely want to use that!__

Roxy-3DPrintBoard, 2016-06-02:
Thanks, that helps make it easier to move to resolution.   Let's see what Jbrazio is thinking about the #ifdef's.

Gaffalover, 2016-06-02:
@Roxy-3DPrintBoard 

> Would you mind sending us some pictures of your cabling going to the display? Is the wiring close to or running for some distance next to high current lines like the stepper motor wires?

Sure.
The cable is a bit improvised but technically checked with a ohm meter.
The oscillogramms are taken as close to the connectors as possible:

The shots are taken with RC6 and the #4 modification mentioned by @CONSULitAS

[image]
pic_22_1.jpg seems to be some kind of enabling:
[image]
pic_22_5.jpg is the display data
[image]
and pic_22_6.jpg is a closeup of the first bytes from pic_22_5.jpg.
[image]

Roxy-3DPrintBoard, 2016-06-02:
Those signal pictures are taken at the graphics display?

Gaffalover, 2016-06-02:
I pierced the cables running to the connectors of the controller. :/
I personally won't believe its an noise error.

SwiftNick, 2016-06-02:
You can rule out any interference from other cabling, as the problem occurs when the machine is at idle and powering up. All stepper motors are disabled.
My cables are crimped at one end and soldered at the other and are very short.
[image]

I haven't had any display issues in any other version of Marlin I've used so it is something introduced in the later releases. It would seem that there is a timing issue for faster boards as the no-ops have been added already for CPUs with a 20Mhz clock. (they aren't in earlier releases). It seems with the improved optimisation, some 16Mhz board / display combinations are now on the edge timing wise.  

CONSULitAS
My test results:  
#1) Fails with display artefacts but slightly better than the original firmware
#2) Fails with display artefacts but slightly better than the original firmware but worse than #1)
#3) Fails with display artefacts similar to #2)
#4) Works just fine.

esenapaj, 2016-06-02:
@CONSULitAS
>BTW: This should not work on the Arduino DUE, too. Definitely a point for the HAL.

I don't see clearly that the following infomation about Due-based system is helpful to solve the problem in this thread or not, but I'd like to help as much as possible.

In case of [Marlin4Due](https://github.com/Wurstnase/Marlin4Due), it has been implementing by [_delay_us()](https://github.com/Wurstnase/Marlin4Due/blob/master/Marlin/HAL.h#L80-88) that it was written by inline assembler.
It seems that roots of this function is [HAL::delayMicroseconds()](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoDUE/Repetier/HAL.h#L413-422) in [Repetier-Firmware](https://github.com/repetier/Repetier-Firmware).
Waits are [2μs +  2μs](https://github.com/Wurstnase/Marlin4Due/blob/master/Marlin/ultralcd_st7920_u8glib_rrd.h#L30-34).
```cpp
static inline void _delay_us(uint32_t usec) {
  uint32_t n = usec * (F_CPU / 3000000);
  asm volatile(
      "L2_%=_delayMicroseconds:"       "\n\t"
      "subs   %0, #1"                 "\n\t"
      "bge    L2_%=_delayMicroseconds" "\n"
      : "+r" (n) :  
  );
}
```
```cpp
static void ST7920_SWSPI_SND_8BIT(uint8_t val)
{
  uint8_t i;
  for( i=0; i<8; i++ )
  {
    WRITE(ST7920_CLK_PIN,0);
    delayMicroseconds(2);
    WRITE(ST7920_DAT_PIN,val&0x80); 
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    delayMicroseconds(2);
  }
}
```

And in case of [Marlin Duo](https://github.com/esenapaj/Marlin/tree/Duo) that it's ported version from Marlin4Due and [MK4due](https://github.com/MagoKimbra/MarlinKimbra4due) to newest RCBugFix by me,
[Code in HAL](https://github.com/esenapaj/Marlin/blob/Duo/Marlin/ultralcd_st7920_u8glib_rrd.h#L46-68) is similar to Marlin4Due, but I decreased waits to [1μs + 1μs](https://github.com/esenapaj/Marlin/blob/Duo/Marlin/ultralcd_st7920_u8glib_rrd.h#L51-61). 
My RRD full graphic smart controller + long signal cable(about 800mm) with Due-based system(Arduino Due + RAMPS Duo(it's similar to [RAMPS4DUE](http://forums.reprap.org/read.php?219,479626)))
is working without problems with that setting.
Arduino IDE version is 1.6.10 Hourly Build 2016/05/31 11:13, U8glib version is 1.19.1, version of package for SAM is 1.6.8.
```cpp
  static FORCE_INLINE void HAL_delayMicroseconds(uint32_t usec) { //usec += 3;
    uint32_t n = usec * (F_CPU / 3000000);
    asm volatile(
      "L2_%=_delayMicroseconds:"       "\n\t"
      "subs   %0, #1"                 "\n\t"
      "bge    L2_%=_delayMicroseconds" "\n"
      : "+r" (n) :  
    );
  }
```
```cpp
static void ST7920_SWSPI_SND_8BIT(uint8_t val) {
  uint8_t i;
  for (i = 0; i < 8; i++) {
    WRITE(ST7920_CLK_PIN,0);
    #ifdef __SAM3X8E__
      HAL_delayMicroseconds(1);
    #else
      #if F_CPU == 20000000
        __asm__("nop\n\t");
      #endif
    #endif
    WRITE(ST7920_DAT_PIN,val&0x80);
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    #ifdef __SAM3X8E__
      HAL_delayMicroseconds(1);
    #else
      #if F_CPU == 20000000
        __asm__("nop\n\t""nop\n\t");
      #endif
    #endif
  }
}

```
[image]

Roxy-3DPrintBoard, 2016-06-03:
@Blue-Marlin 
> Do what you want. I'm perfectly able to change it to my taste.

Yes.  We know you can do that.   But if you have strong feelings, we want to understand them because we are concerned about the less technical users.  We don't want to go against good, solid technical advice.

> I'm very sorry for my obviously to bad writing. Most of you seem to read things in my messages i have never written, or do miss things, where i think, i have written about. Or is it your reading?

That is something that happens when people highly respect your opinion.  If you say something, even if it is poorly worded, they are going to listen!    They are going to try to make sense of it.  And it is possible (at least in my case) it is the reading of the message too.  With so many highly technical topics going by everyday, there is only time to skim a lot of them.   So everybody needs to be patient with everybody.

AnHardt, 2016-06-03:
May i suggest https://github.com/AnHardt/Marlin/pull/43 ?
Where all party's would have their profit.

It needs some further test by you to find the right delays 
All our `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
` clones do work completely without nops at our RAMPS boards - but your mileage may vary

An other micro second can be saved by replacing `u8g_10MicroDelay()` with `delayMicroseconds(5)` but here our displays differed more. (2x4µs, 2x5µs, 1x6µs)

jbrazio, 2016-06-03:
@SwiftNick @Blue-Marlin @CONSULitAS Can we all test Andreas solution ? It's a lot more verbose source code wise but has huge savings: 1k memory, the only issue is it will change code path for all.. working and non working LCDs so validation is more crucial.

Roxy-3DPrintBoard, 2016-06-03:
> All our REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER  clones do work completely without nops (on) our RAMPS boards - but your mileage may vary

@AnHardt   This is an example of where sending out a few loaner units helps everybody.   It is very difficult to test the firmware side without hardware but the people doing the firmware can't buy every piece of hardware.   Right now I'm trying to get consensus to make some changes to the Readme file that basically say to the hardware vendors we are going to support your hardware and your users no matter what.  But it would make our life a lot easier if we had real hardware on loan that we can code against.

AnHardt, 2016-06-03:
@SwiftNick @Gaffalover 
The timing of the ST7920 depends on the voltage it gets. Please measure GND - 5V at the display side of the cables.
https://github.com/AnHardt/Marlin/pull/43#issuecomment-223574911

thinkyhead, 2016-06-04:
@AnHardt Do you feel your PR https://github.com/AnHardt/Marlin/pull/43 is ready, or does it need more testing?

jbrazio, 2016-06-04:
More testing definitely, no one replied yet.

thinkyhead, 2016-06-04:
I can do some testing pretty soon. Maybe today, maybe Sunday.

SwiftNick, 2016-06-04:
Sorry away over the weekend, will test on Monday

SwiftNick, 2016-06-06:
I replaced my function with the new code below and I get very minor character corruption.
The top line of the ? on the info screen are corrupted until the second flash.
If I increase the first delay to 3uS there is no corruption.
 
Compiled using Arduino 1.6.9, UG8Lib v1.19.1, Marlin 1.1.0-RC6

```cpp
static void ST7920_SWSPI_SND_8BIT(uint8_t val)
{
  uint8_t i;
  for( i=0; i<8; i++ )
  {
    WRITE(ST7920_CLK_PIN,0);
    delayMicroseconds(2);
    WRITE(ST7920_DAT_PIN,val&0x80); 
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    delayMicroseconds(2);
  }
}
```
VCC at the edge connector of the LCD PCB is 4.98v

AnHardt, 2016-06-07:
@SwiftNick 
Thank you for your tests.
Would you please now be so kind to test how many delays you need for your machine in https://github.com/AnHardt/Marlin/pull/43.
That PR safes a bit of memory, speeds up copying one byte from memory to the display a bit in general, but allows to easily insert waits tor those who need them. Adjusting with _nop's_ can be much more exact than with delayMicroseconds(). 
The defines for
```cpp
#elif MOTHERBOARD == BOARD_3DRAG
   #ifndef ST7920_DELAY_1
     #define ST7920_DELAY_1 DELAY_0_NOP
   #endif
   #ifndef ST7920_DELAY_2
     #define ST7920_DELAY_2 DELAY_0_NOP
   #endif
   #ifndef ST7920_DELAY_3
     #define ST7920_DELAY_3 DELAY_1_NOP
   #endif
 #elif F_CPU == 16000000
```
The values are already adjusted for the theoretically needed pauses at 4.5V and above. 
We already know your display - controller combination needs more waits. Your job is to find out, _how many_.

SwiftNick, 2016-06-08:
The lowest number of nop's I had to insert to prevent display corruption was 3
``` cpp
#elif MOTHERBOARD == BOARD_3DRAG || (MOTHERBOARD == BOARD_K8200)
   #ifndef ST7920_DELAY_1
     #define ST7920_DELAY_1 DELAY_0_NOP
   #endif
   #ifndef ST7920_DELAY_2
     #define ST7920_DELAY_2 DELAY_3_NOP
   #endif
   #ifndef ST7920_DELAY_3
     #define ST7920_DELAY_3 DELAY_0_NOP
   #endif
```
The following nop configurations are all good and will stop the display corruption.
Taken in the order:
delay_1, delay_2, delay_3
0,3,0
1,1,2
1,2,1
2,1,1

anything smaller does reduce the corruption but does not eliminate it.
These combinations do not work.
0,0,3
3,0,0
0,1,2
0,2,0
1,1,1

AnHardt, 2016-06-08:
@SwiftNick 
Strange. At the place of DELAY_2 the datasheet requires 50ns - less than an instruction.
However. Thank you for testing. I'll make the PR ready tomorrow with 0,3,0.

SwiftNick, 2016-06-08:
I guess what we really need is to test a few more boards to see if the pattern is the same and the issue isn't down to a single rogue display.

roglio, 2016-06-13:
Disabling optimisation works for me too. My configuration: melzi ardentissimo atmega1284P and full graphic rrd smart controller.

[23 more replies]

closed 2018-08-21 by thinkyhead