MarlinFirmware/Marlin #3816: Nozzle position with autobed leveling

Drmaestro, 2016-05-22:
Hi,

I was wondering if there is a way to see the position of the nozzle in relationship to the bed when auto-leveling, without starting a print. I use BLTouch for auto leveling. After the auto-leveling procedure the probe goes to a certain Z height and doesn't go to 0.00 position, so it isn't possible to see the distance between the nozzle and the bed. This is a problem when you change a hotend or a nozzle, which can alter the vertical dimension of the hotend and requires changing the Z offset. Right now, I have to empirically set this height and watch it if it is too close or too high. It would be nice if the Z offset could be changed interactively.

boelle, 2016-05-22:
that would be a new feature. please make the feature and submit a PR

jbrazio, 2016-05-22:
> to a certain Z height and doesn't go to 0.00 position

Just to clarify, you speaking about ABL using `G29` ?

Drmaestro, 2016-05-22:
Yes, the G29 command, but it is also similar for G28. Because the probe is located lower than the nozzle, you can't be sure if the nozzle will be at the right position when printing will start. Of course, after the autoleveling command you can manually lower Z to 0.0 and see if it is level with the bed, and then calculate the difference and change the Z offset. Maybe, there isn't a faster way to do that, so we can close the subject if you want.

jbrazio, 2016-05-22:
We had some changes to `G92` that somehow "broke" a hack people where using to set the offset.

If you're correctly setting the offset, after a `G28` the Z height should be exactly the module of the probe offset.

Grogyan, 2016-05-22:
@Drmaestro I use the BLTouch and typically use the EEPROM to hold the offset.
Using a sheet of paper and setting the probe offset in EEPROM to for example 2mm, you can then drop the nozzle 0.1mm until the paper cannot be removed, then do some quick match to adjust the EEPROM offset value, and double check.
This process takes about 5 minutes for me.

closed 2016-05-24 by jbrazio