MarlinFirmware/Marlin #3817: Could we use more code people?

boelle, 2016-05-22:
Or would my thought risk breaking up the remaing 4 people?

jbrazio, 2016-05-22:
[The more the merrier](https://github.com/MarlinFirmware/Marlin/compare).

Roxy-3DPrintBoard, 2016-05-22:
In general...   The more the merrier...   But besides the obvious constraint of finding people that are good Team Players and not causing problems for no reason...  We need people that can stay out of each other's way while making valuable contributions.   Right now we are spread too thin and staying out of each other's way is not a problem.

It is my belief that when we finally feel the Release Candidate is ready to be promoted to 'Golden Status' it will attract some contributions.   And if I can keep up with all of ThinkHead's changes we are going to have a Development Branch based off of the Golden version at the same time.   The new features will encourage a lot of code changes.

boelle, 2016-05-22:
i just feel that we are drowning in issues and nothing gets done.... its very frustrating

am i wrong in what i said on hangout chat? in that people are so used to just file an issue and get a feature done in a few days?

boelle, 2016-05-22:
and also we agreed to get rid of the dev repro and pick issues from there...

we found an issue mover tool... but again nothing has happened.... i'm up for the task of moving issues, but i need the right access level

jbrazio, 2016-05-22:
Nothing gets done ?.. lol I suggest a `git log` reality check.

Btw, your vague comments telling people basically to go fuck themselves for sure will not help GTD. :-/

boelle, 2016-05-22:
well i could tell them to do that directly.... but then for sure i would piss off people

things get done i know that.... but the simple task of moving issues.... why have that not been done yet?

boelle, 2016-05-22:
and i dont want people to go fuck themselfes... but with 1-2 people coding we for sure cant take feature request also.... 

i mean dont you and scott have more than enough to do allready?

jbrazio, 2016-05-22:
Don't worry with new requests, we are able to decide if we can afford to implement them or not. Requests bring new ideas, new ways to improve old features.. user requests are the heart of the thing.

Of course we would love to receive PR instead of feature requests, but Marlin src code is not inviting to newcomers.

Going back to the dev repo, we decided to take an action after GA, so we haven't do it do lack of time or resources but by option.

boelle, 2016-05-22:
btw... you said " git log "

where is that?

jbrazio, 2016-05-22:
> Drowning in issues

We never had so many low issues as of today. The pending PR, 95% are concept proposals and that's the reason for not being merged.

jbrazio, 2016-05-22:
Command line git client, or you may select the `RCBugfix` branch and see the commit history in Github.

boelle, 2016-05-22:
oki... i dont use command line

WZ9V, 2016-05-22:
My opinion as an old timer in the C++ world (think OS/2 and Win 3.1 era) is be very careful expanding the team.  I manage programming teams now and it is very easy to upset a productive dynamic by introducing new elements.  I feel some of the management and structural issues need to be worked out first.  Without a solid plan for how to move forward and some way to divide up the work it will just get crazy.  Just my 2 cents.

boelle, 2016-05-22:
well our current situation is that we have 1-2 people that can do the code... this is just an estimate from who is most active

and one of them live in the us and have just had to move out from his current house/flat, so he is only able to help part time and need to steal internet access at various cafe's and mall's

boelle, 2016-05-22:
but without knowing 100% i think the plan is to sort the issues that are currently in the bugfix branch

and then do a release..... and then move over old isses from the dev branch and i assume start to fix those

Roxy-3DPrintBoard, 2016-05-22:
> well our current situation is that we have 1-2 people that can do the code... this is just an estimate from who is most active

There are way more than 1 or 2 people contributing to the code.   Obviously, ThinkyHead is very visible.  But lets just review the issue threads from the last couple of days.

Epatel grabbed the rudder and turned it to keep us from hitting the rocks here:  https://github.com/MarlinFirmware/Marlin/issues/3811   

Blue-Marlin/Anhardt jumps in on everything from End Stop issues to Thermal and Core-XY problems.   

A new poster has shown up at the end of this thread with an entirely new idea (it sounds like he actually has code working!) for Delta Bed Leveling:  https://github.com/MarlinFirmware/Marlin/issues/2159    

I've got a personal goal of having a Unified Bed Leveling branch of the Golden Release with other experimental Delta Configuration stuff ready within days of the announcement.  (My thinking is this would kick off the new development branch that pushes ahead of the Stable Release Branch.)   

And jbrazia obviously can code (doing instruction level cycle timing with the assembly instructions?????) in this thread https://github.com/MarlinFirmware/Marlin/issues/3680).  He seems to  prefer cleaning up some of the messes that we have outstanding and that is certainly welcome.

This is all in the past few days and certainly not inclusive.

Going back a few weeks and looking at the MarlinDev side we are also getting guidance from a number of experienced software/firmware types.   A new addition is gddeen promoting Object Orientation as a way to clean up some of the code.  We haven't discussed it enough to start moving on the topic but we very possibly are going to be borrowing heavily (stealing) the work that MagroKimbra did to reorganize the source files.   Both topics are discussed in this thread:   https://github.com/MarlinFirmware/MarlinDev/issues/395

Trying to plan a reasonable transition to higher powered microprocessors is on the table.  It is being discussed here:  https://github.com/MarlinFirmware/MarlinDev/issues/398  Reading that thread we obviously are getting good solid advice and help from hard core software types.   Some of these people have written code that we may choose to absorb (for example BobC's HAL for DUET).

We also have Laser enhancements to consider adding to the code base being talked about in this thread:  https://github.com/MarlinFirmware/MarlinDev/issues/418  There is another Laser thread with code being actively experimented with and worked on.   Obviously, we let those mature a little bit.

Mostly, my point is...    We have way more than 1 or 2 people that can do code...

jbrazio, 2016-05-22:
@Roxy-3DPrintBoard "jbrazi**o**" ! I'm a guy not a girl.. ;-P
(In Portuguese words ending with an "a" are feminine)

Roxy-3DPrintBoard, 2016-05-22:
That was a typo!   I didn't mean to do that!   You are now male again.

jbrazio, 2016-05-22:
I was just messing with you, altought it's a true fact about the language.

boelle, 2016-05-22:
well i guess i'm just scared from last time where things came to a grinding halt and people where more busy showing off that they where right

emartinez167, 2016-05-22:
I can also code, and have submitted a couple of new boards in the past with the help of others (RigidBot, Sainsmart 2-in-1). I just do not have enough depth of knowledge on Marlin (but read every contribution to try to learn). I have 30+ years of C/ASM coding experience, and 3 of Arduino doing other small jobs. Will try to contribute a bit more.

Regards/Saludos,

Ernesto

thinkyhead, 2016-05-22:
> need to steal internet access at various cafe's and mall's

Well, to be fair wifi access comes with the coffee.

Roxy-3DPrintBoard, 2016-05-23:
> Well, to be fair wifi access comes with the coffee.

The massively over priced coffee!

thawkins, 2016-05-23:
What's happened to the arduino 1.6.x only library based structure?  Has that been abandoned?

boelle, 2016-05-23:
"What's happened to the arduino 1.6.x only library based structure? Has
that been abandoned?"

who deleted that one?

Roxy-3DPrintBoard, 2016-05-23:
> What's happened to the arduino 1.6.x only library based structure? Has that been abandoned?

Much of the need for a library based organization was driven by the fact Arduino would not allow us to have the code spread across multiple folders.  The library based structure allowed a way around that.   As it turns out, enough people complained about that to the Arduino crew and now we can have separate folders for different portions of the code.   

We just haven't done that yet.    When a Release Candidate gets promoted to 'Golden' we will start a  development branch and one of the things will be to break the code apart into a better organization.   Right now, the 'Best Thinking' is we will copy a lot of what MarlinKimbra has done from an organizational perspective.

thinkyhead, 2016-05-23:
> The massively over priced coffee!

Thankfully, at my favorite cafe here it's only $1.25 a cup with $0.75 refills. They make very good food, also reasonably-priced.

boelle, 2016-05-23:
btw... we have some great stuff going on in the hangout chat

if you have time to come that is

boelle, 2016-05-23:
also dont move to denmark... coffee at caffee is 3x the price

jbrazio, 2016-05-27:
Now we have plenty, let's see how they code.

closed 2016-05-27 by jbrazio