MarlinFirmware/Marlin #3820: Delta: endstop Z_MIN, but only for probe.. possible?

otvald, 2016-05-22:
I use am using RC-branch, and have a delta printer with z-probe integrated with nozzle (Micromake).

I can do auto level, but when running, the Z_MIN can cause problems if triggered. This happens if I don't have a completely flat bed, and the nozzle hits the glass.

Is there a way to disable Z_MIN (and only use for probing)?

Blue-Marlin, 2016-05-22:
ENDSTOPS_ONLY_FOR_HOMING ?

otvald, 2016-05-22:
#define ENDSTOPS_ONLY_FOR_HOMING

Is defined in Configuration_adv.h

I assume it should be enabled (i.e. not commented out), right? But when triggering z probe (Z_MIN) during printing, this affects the planner. (and caused not just bad height, but also offsets horizontaly)

RicardoGA, 2016-05-22:
Maybe this is issue is related with endstop.cpp with the line 42-44 (the states are inverted) @otvald  could you try to change it from

Endstops::Endstops() {
  enable_globally(
    #if ENABLED(ENDSTOPS_ONLY_FOR_HOMING)
      true
    #else
      false
    #endif

To 

Endstops::Endstops() {
  enable_globally(
    #if ENABLED(ENDSTOPS_ONLY_FOR_HOMING)
      false
    #else
      true
    #endif

And check if this fix your issue

closed 2016-05-23 by jbrazio