MarlinFirmware/Marlin #3824: [Feature Request][Discussion] GCODE Multiline Buffer

jamiehofer, 2016-05-23:
I haven't seen many active discussions on the implementation of sending multiple lines of GCODE at once. I think that it would be nice to have it setup more like (I hate to use this analogy) UDP instead of TCP. We could send the printer:

`G1 [command 1];[command 2];[command 3]`

and reply

`ok Q1`

While I think that we are a long time away from ever changing the way GCODE is interpreted and sent, I want to know your opinions on the idea.

jbrazio, 2016-05-23:
This is out of all the G-Code specs I've seen so far, where did the idea came from ?

Roxy-3DPrintBoard, 2016-05-23:
> We could send the printer:
> G1 [command 1];[command 2];[command 3]
> and reply
> ok Q1

I suspect you mean something more like:

> We could send the printer:
> 124G1 [command 1];[command 2];[command 3]
> and reply
> ok Q124

What we need first is a very solid error detection and possibly an error correction for the transmitted lines.   Right now...  The checksum is just a lame 8 bit XOR.

jbrazio, 2016-05-23:
This breaks all G-Code standard on the planet.. lol

Blue-Marlin, 2016-05-23:
Especially because ';' starts a comment. I could imagine spaces or nothing - but not semicolons.
> G1 [command 1];[command 2];[command 3]

looks strange to me. Maybe 
```
G1 [point 1] [point 2] [point 3]
```
But even then it would be difficult to find out how to group coordinates together.

All in all i don't see an advantage. Text volume is currently not our problem. 

Has one of you a BNF or EBNF description of g-code? Is it worth the effort to make one?

jbrazio, 2016-05-30:
@Blue-Marlin do you indeed believe we should support this type of commands ?

Blue-Marlin, 2016-05-30:
@jbrazio 
No. How did you read any positive reaction in my posting?

jbrazio, 2016-05-30:
No but as you built a bit upon the initial proposal I was just clearing if we're all on the same page.

closed 2016-05-30 by jbrazio