MarlinFirmware/Marlin #3825: Software-Endstopy clipping

AnHardt, 2016-05-23:
I admit, normally this is not an issue, because the slicers know about our build-room.
But as soon we shift the part later, for example with a simple G92, we are in danger to hit the software endstops.
A first layer:
[image]
A first layer shifted to be 50% outside the bed:
[image]
For a good cliping algotythem i'd expect to not extrude the left part and omitting the moves, while the right part shold be exactly as drawn.

What we currently get is a bit different.
[image]

The short term solution is to stop printing when a software-endstop is hit.
The long term solution is a proper clipping algorithm.

jbrazio, 2016-05-23:
> For a good cliping algotythem i'd expect to not extrude the left part and omitting the moves, while the right part shold be exactly as drawn.

100% agreed.

Roxy-3DPrintBoard, 2016-05-23:
If part of the movement is clipped, I think it is hard to find a case where it makes sense to keep extruding.   It may or may not make sense to clip all portions of the movement if one axis goes out of bounds.   But extruding while you are not where you were told to be just feels wrong.

jbrazio, 2016-05-23:
I can think of a situation where this can happen even without using `G92`, brims and skirts.

thinkyhead, 2016-05-24:
Software clamping / clipping is mainly there to prevent the machine grinding, as a very dumb workaround. It was never designed to allow movements that go off the edge to be "recoverable" or with the expectation that running off the print bed should be a regular thing. If you're running off the bed, you're _supposed_ to fix the problem in the slicer.

Anyway, if you've got a print running off the edges, as pointed out simple clamping isn't going to be enough. To get "proper" clipping, the planner should instead chop lines off at the edge of the movement range, throw away any lines that lie entirely outside the movement range. The E axis will need to continue advancing logically (but not actually move the E motor) when the nozzle position is "out of range".

You might be able — with enough clever coding— to be able to have a print still work even though it's half off the printable area. But all this will require a lot of extra computing in the planner functions.

boelle, 2017-04-26:
@thinkyhead we are close to a year on this one

Q: is this still relavant and if so should it not be added to a milestone?

closed 2017-11-12 by thinkyhead

AnHardt, 2017-11-12:
Is this solved? Where? When?