MarlinFirmware/Marlin #3826: RC6 - "start print" does not zero print time after "stop print" command

Roo123, 2016-05-23:
RC6 firmware

Supposing you stop a print with "stop print" from LCD.

The current print time remains on the display. (Good.)

Then you start another print.

The time for this second print starts from the end time of the previous print. Without first zeroing.

Not good. ;-)

jbrazio, 2016-05-23:
We need more information.

How do you start printing is it from the LCD or from the host ?
Do you have `PRINTJOB_TIMER_AUTOSTART` set ?

Roo123, 2016-05-23:
Started from SD card / LCD.

In which file is the printjob_timer_autostart variable located?

jbrazio, 2016-05-23:
A few minutes ago a new PR fixing some issues with the print job timer was merged into RCBugfix, would you care to download the latest version and to a quick test to see if the behavior is still there ?

`PRINTJOB_TIMER_AUTOSTART` is in `Configuration.h`.

Roo123, 2016-05-23:
PRINTJOB_TIMER_AUTOSTART isn't showing up when I search in configuration.h of RC6 ?

Afraid I can't jump into the Bugfix just at the moment....

Blue-Marlin, 2016-05-23:
In that case you don't have a current RCBugFix. (https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L772)

jbrazio, 2016-05-23:
I've had a quick look and this is indeed a bug, we stop the SD print using a custom function and this function is not taking into account the print job timer.

```cpp
  static void lcd_sdcard_stop() {
    stepper.quick_stop();
    card.sdprinting = false;
    card.closefile();
    thermalManager.autotempShutdown();
    cancel_heatup = true;
    lcd_setstatus(MSG_PRINT_ABORTED, true);
  }
```

commit 3861f1e0 (2016-05-30): Merge pull request #3908 from jbrazio/bugfix/3826

Closes #3826: SD printing will now pause and stop the print counter
commit 40e64b36 (2016-05-30): Closes #3826: SD printing will now pause and stop the print counter

The LCD menu items for SD printing pause/stop have been updated to control
also the print job timer. This commit also fixes a small output bug with
M78, the failed print counter was displaying the wrong value.
closed 2016-05-30 by jbrazio