MarlinFirmware/Marlin #3827: Feature request: "go to Z max" + 10mm Zsteps menu options made possible by a Z home

Roo123, 2016-05-23:
One of the most tedious things is to crank the Z to the bottom using 1mm steps.

We could have a "go to Z max" menu option

and / or enable Z 10mm increments

IF both such commands launched a Z home command first to ensure Marlin orientated and prevent crashes

jbrazio, 2016-06-08:
```diff
   //
   // Auto Home
   //
   MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
+  MENU_ITEM(gcode, "Move Z to MAX", PSTR("G1 Z<YOU VALUE HERE> F<YOUR FEEDRATE HERE>"));
```

closed 2016-06-08 by jbrazio

thinkyhead, 2016-06-08:
The example above shows how you can add it yourself.

Also, the newest version of LCD-based manual movement should improve Z movement a bit. It delays movement until you pause the controller wheel. However it doesn't add an option to move Z in 10mm increments.

Roo123, 2016-06-08:
Thanks a lot.