MarlinFirmware/Marlin #3830: Random garbage displayed on LCD screen.

Drmaestro, 2016-05-24:
Hi,

On RC6, I occasionally get random garbage displayed on the LCD screen. It always happens on the info screen and pressing the encoder button always corrects the problem (by changing the screen). Is this a bug?

Thanks.

jbrazio, 2016-05-24:
This could be bad wiring, cold solder joint or noise on the cable going to the LCD.
Do the cables go close to the stepper cables ?

Drmaestro, 2016-05-24:
The left Z stepper cable is at 2 cm distance at a certain location. 

If wiring/soldering or the stepper cable were the culprits, wouldn't it be more frequent? I only get this once in a day maybe. 

I reported this issue to see if there are other people with a similar issue. If there are none, that would mean it's a local  problem with my own system.

jbrazio, 2016-05-24:
The other report that maybe similar is this #3815

Drmaestro, 2016-05-24:
I use the 4 line LCD, not the graphic one. Would it still apply to my case?

jbrazio, 2016-05-24:
Which IDE are you using right now ?

thinkyhead, 2016-05-25:
Random garbage on the screen mostly likely indicates a communication glitch which could be caused by loose wiring to the controller, a loose SD card, or electronic noise. The data on the LCD screen comes from the Arduino board, is sent to the controller by a serial line, and is not related to external USB connectivity. If only certain items on the screen were glitching out (for example just the XYZ coordinates) then I would start to think software bug. But if it's the whole screen, it's almost certainly hardware-related.

Drmaestro, 2016-05-25:
Ok. Thank you. I'll check my connections again.
@jbrazio My IDE is 1.6.8

closed 2016-05-30 by jbrazio