MarlinFirmware/Marlin #3834: feature request: 1st preheat option in list = "Bed + nozzle"

Roo123, 2016-05-24:
The preheat options are useful: they represent a shortcut to getting to a commonly desired temperature state.

In my experience, my most common pre-heat request is to get BOTH bed and nozzle up to temperature.

Currently I must enter the menu hierarchy twice to get both selected.

The list would be most helpful to my use case if it took the form:

1. Bed + Nozzle;
2. Bed (only);
3. Nozzle (only);

Thanks!

thinkyhead, 2016-05-25:
This would require some reworking of the existing code. All the lcd-based ABS and PLA heating functions heat the bed in addition to the indicated hot-end, and there's no function, currently, to heat only a nozzle without also heating the bed. On a setup with 4 nozzles, a menu with all the possible options might become a little unwieldy. And then the menu text starts to get longer too, and it gets a little tricky with some languages to keep these under 18 characters long.
- Preheat PLA (Bed+All Hotends)
- Preheat PLA (Bed Only)
- Preheat PLA (All Hotends)
- Preheat PLA (Bed+Hotend 1)
- Preheat PLA (Hotend 1 Only)
- Preheat PLA (Bed+Hotend 2)
- Preheat PLA (Hotend 2 Only)
- Preheat PLA (Bed+Hotend 3)
- Preheat PLA (Hotend 3 Only)
- Preheat PLA (Bed+Hotend 4)
- Preheat PLA (Hotend 4 Only)

Roo123, 2016-05-25:
Interesting.

Here's an alternative suggestion:

At present it's necessary to go through multiple menu selections to arrive at a choice. For example,

Prepare / Preheat PLA / Preheat Bed

After making a selection you are returned to Home screen.

How about if, after making a selection, the display remained on the "Preheat PLA" menu screen, READY FOR ANOTHER SELECTION?

Return to home screen would occur:

a) if after Time Out;

b) if after all possible selections had been made;

This system would be perceptibly nicer if, after a selection had been made, the menu item in question indicated that it had been selected (eg a tick).

thinkyhead, 2016-05-26:
> READY FOR ANOTHER SELECTION?

I would guess that the design choice to go back to the Info Screen immediately on setting the temperature is so you can start to monitor the temperature on-screen right away. I'm not sure there would be any other clear feedback on-screen to indicate that heating has started. Your controller board's heating LED will come on, and some screens can change color, but for the standard LCD controllers I think this "feedback" is preferable to simply going up a level in the menu.

> after a selection had been made, the menu item in question indicated that it had been selected (eg a tick)

If you press the controller button, there is a click. Currently there are no menu items that automatically activate by leaving the cursor on them, or anything like that.

jbrazio, 2016-05-27:
How about Ninjaflex ? And Nylon ? And HIPS ?.. Nowadays makes no sense to have a "pre-heat" for material type, the list will end up endless. Having said this I do agree a pre-heat menu which allows you to select pre-heat temp for nozzle and bed, or just nozzle, or just bed.

Roo123, 2016-05-27:
It's true that, whatever the menu display solution, it should be easy to edit the Material Types & corresponding temperatures in the config file.

Personally I never print in ABS and very rarely in PLA: the presets are nonetheless useful to get things going whilst I save my PETG print to SD !

jbrazio, 2016-05-27:
@clexpert is reworking this part of Marlin included into the manual filament change procedure.

Roo123, 2016-05-27:
Material saved to EEPROM. 

Clever!

thinkyhead, 2016-05-29:
Currently we have `M145` to change the presets for hotend temp, bed temp, and fan speed. For example:
```cpp
M145 S0 H185 B60 F100
M145 S1 H240 B100 F0
```
These can be saved to EEPROM also. There's no option to set the material's name, but that could be added. Allow up to 10 characters, or something.

boelle, 2016-11-05:
like @jbrazio commented on 27 May...

a list for each material can go to long...

but a  pre-heat menu which allows you to select pre-heat temp for nozzle and bed, or just nozzle, or just bed would be good. special for calibration where you need the nozzle hot as it expands with heat, not much but enough.

@thinkyhead @clexpert @jbrazio 

is there a status for this one?

thinkyhead, 2016-11-06:
First item in the menus is currently to heat both bed and nozzle for the first material (no longer assumed to be PLA). Last item within each nozzle group is to heat the bed-only. There's no nozzle-only heatup option at this time.

psavva, 2017-03-19:
Why not just use some preset gcode files saved on your SD card, and preheat using that.

Roxy-3D, 2017-03-23:
> Why not just use some preset gcode files saved on your SD card, and preheat using that.

Or...  

I have the Encoder Wheel setup such that if you Double Click it (while at the status menu) it drops into Z-BabyStepping.   That could be made to be user configurable so the user gets to pick
- What action happens.  And it doesn't need to be just one thing.   It can be a list of commands
- If the user wants two different 'Quick and Easy' response items, we take over the Kill button and let that be a second list of commands.
- For people that wanted a Pre-Heat of both PLA and ABS, one button could heat the bed and nozzle to 60 C and 205 C.   The other button could heat the bed and nozzle to 110 C and 235 C.

boelle, 2017-04-26:
i would argue that we dont need to have a materials list.

there is very little difference between temps that is hardly matters. 

the big elephant in the room is the bed, that one takes the longest to heat up. i would put that first in the list and and then nozzle and then both last.

if you  preheat nozzle and then bed without having the nozzle 0.05mm over the bed you will end up ozzing the nozzle empty and will have to prime at lot more prefore prints

closed 2018-02-26 by thinkyhead