MarlinFirmware/Marlin #3836: Menu freeze while Z is moving manual

Massa0815, 2016-05-25:
Hi
I set up a newSmartrap wirth Marlin 1.1.0 RC6.

When i move Z axis manualy per "Prepare" menu it is not possible to change the position with the rotary knob while z is moving. I have to wait until Z stops to increase or decrease the target position. 

I have a version 1.0.6 on my Prusa i3. With this Version it works fine.

thinkyhead, 2016-05-26:
@Massa0815 Sorry about this behavior. The change was made to prevent the controller from having the wrong value on-screen but dropping the last move. There's a patch in the queue that needs more testing, but it will fix this behavior, add a short delay to the movement control, and prevent the last move from being dropped. The PR is #3110, and the full branch can be downloaded at https://github.com/thinkyhead/Marlin/tree/rc_manual_moves

thinkyhead, 2016-06-05:
I've merged #3110 so that manual moves are no longer blocked, and so that there is a 1/2 second delay before a move starts. The upshot of this is that when trying to do very precise short moves, you must wait 1/2 second after each "tick" of the controller wheel. So perhaps this delay could be made shorter when doing very short moves such X or Y by <= 1mm.

Give the latest code a try and see if it behaves in a way that feels right.

Massa0815, 2016-06-12:
I testet it. It works better, but the delay is a little bit to high for my taste. Its a bit to annoying whne the movement comes so late.

thinkyhead, 2016-06-12:
@Massa0815 Try some lower delay values and see if there's one that feels better. Currently it's a 1/2 second, so maybe 1/4 second is better. Just alter the value here:
```cpp
inline void manual_move_to_current(AxisEnum axis) {
  manual_move_start_time = millis() + 500UL; // 1/2 second delay
  manual_move_axis = (int8_t)axis;
}
```

closed 2016-07-11 by jbrazio