MarlinFirmware/Marlin #3843: Ability to follow cubic Bézier splines

boelle, 2016-05-25:
_From @giomasce on May 5, 2016 16:33_

[this issue was already filed as MarlinFirmware/Marlin#3022, but it was closed and I was suggested to re-open it here]

I added support for drawing Bézier cubic splines to Mariln. The code is here: https://github.com/giomasce/Marlin/commits/Development. I use it for my Marlin-based laser cutter and it apparently works with no problems. A few comments about the implementation are both in the code and in MarlinFirmware/Marlin#3022.

Please note that the patches are based against a rather old version of Marlin. I do not have a recent enough building environment, so I cannot rebase them on the development branch. However the code is rather independent from the original codebase, so it should not be difficult to port it.

_Copied from original issue: MarlinFirmware/MarlinDev#417_

boelle, 2016-05-25:
_From @thinkyhead on May 9, 2016 3:14_

I'm finally getting a chance to examine the code. I will comment more when I have read it thoroughly!

boelle, 2016-05-25:
_From @thinkyhead on May 9, 2016 3:22_

Ok, looks like it should be easy to integrate with the latest code. I will package it up and post it for consideration.

boelle, 2016-05-25:
_From @thinkyhead on May 12, 2016 20:43_

@giomasce See MarlinFirmware/Marlin#3744 for the updated implementation.

jbrazio, 2016-05-27:
Closed by #3744 and followups.

closed 2016-05-27 by jbrazio