MarlinFirmware/Marlin #3847: Max Z acceleration not respected

Robertozzz, 2016-05-25:
Default accelertion at 3000, Max Z accel set to low value. It seems the max for Z is not respected and Z anyway accelerates at 3000 for at least non printing moves, which resulted in stepper skipping steps. 
(Because of dual Z axis/stepper, X axis tilted)

brainscan, 2016-05-25:
What are your steps per mm for z and feedrate? Sometimes it's not the acceleration but the amount of steps being sent. Mine works perfectly at 4mm/s and then grinds and skips at 5mm/s.

Robertozzz, 2016-05-25:
Z axis Steps per mm 395. Tried at different feedrates, same result. The X axis after some time gets slightly offset, for example after a print.  When I force the default acceleration to low level (900), no problems at all, even at feedrates of 100 mm/s !! on the Z, no problem at all, X axis stays perfectly level. My Z axis are 2 threaded rods. (Prusa I3 clone)..

Any Ideas?

Robertozzz, 2016-05-25:
Oops, by mistake closed.. Reopened..

closed 2016-06-11 by jbrazio