MarlinFirmware/Marlin #3848: Feature Request: Online Configurator

boelle, 2016-05-25:
_From @adamfilip on April 18, 2016 4:12_

Repetier has a very handy online configuration tool. that once filled out will output the necessary config.h files needed for a users machine. it will also input an existing .h file and let you make changes.
for novice users this is very handy to get things configured quickly.

Something like this would be great for each official release.

_Copied from original issue: MarlinFirmware/MarlinDev#404_

boelle, 2016-05-25:
_From @Pwuts on April 19, 2016 18:54_

There already was a configurator included in the Marlin v1.1.0-RC3 package but it has been removed due to problems with running it locally. There is an online version hosted by thinkyhead, but that too doesn't work anymore since the Github API was updated. It just has to be modified to suit the latest version of Marlin and Github. :)

boelle, 2016-05-25:
_From @thinkyhead on April 20, 2016 3:53_

https://github.com/thinkyhead/MarlinConfigurator

I started this configurator that parses Marlin's own `Configuration.h` and `Configuration_adv.h` files (so it can use them as the "database" for the configuration options). But it has fallen out of date, as @Pwuts points out. So I need to do some updates and then continue to work on it. It shows some promise as a starting point. After the first phase (making the "expert" configuration tool) I was planning to add the "easy step by step wizard" interface to help set the most important settings.

boelle, 2016-05-25:
_From @bobc on April 30, 2016 11:30_

Some time ago I created a Bounty asking someone to create a config tool. The Teacup ConfigTool is pretty much exactly what I had in mind http://forums.reprap.org/read.php?1,516185. It is in Python so cross-platform and always available locally, and Open Source of course.

It was written specific to Teacup, but hopefully some of the architecture can be re-used. At least it illustrates the principle.

boelle, 2016-05-25:
_From @thinkyhead on May 2, 2016 21:13_

@bobc My Javascript-based configurator runs everywhere too, with only a web browser needed. It could be reworked into other languages, but I find this is a good way to prototype and share with others more quickly. As mentioned, it currently needs some updates to work with the latest Github API, and it needs further work to parse the most recent configuration files. I will patch it up a bit tonight for the latest Github API, and get it into better working order with the most recent configurations.

https://rawgit.com/thinkyhead/MarlinConfigurator/master/src/index.html

The idea with this tool is to first get the "advanced mode" configuration mode working, and then build a simpler "wizard" to collect the core settings, and to config each add-on feature separately. Some of the core functions will also be useful for doing a "drag-and-drop" conversion tool to take 1.0.2-1 (and earlier) configuration files and move settings into a new configuration file. (That tool should be fairly straightforward, but I need to gather all the old/new settings and make a list of which need conversion and which can just move over as-is.)

jbrazio, 2016-05-27:
Duplicate

closed 2016-05-27 by jbrazio