MarlinFirmware/Marlin #3851: Recommendations for a good 32 bit microprocessor to run Marlin

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on March 30, 2016 19:59_

The Atmel ATSAM3x8e on the Due board is definitely viable.   It has the benefit it can plug into the RAMPS or RADDS base board and at that point it is easy to populate it with other needed items like Step Sticks to handle the stepper motors.

But my questions is "What other companies make viable platforms to run 3D-Printers?"   Ideally, these microprocessors have good compiler support and can have similar items to the Step Sticks plugged into them.

The best case would be if these other microprocessor boards were compatible with the RAMPS and RADDS boards and could just plug into them.

_Copied from original issue: MarlinFirmware/MarlinDev#398_

boelle, 2016-05-25:
_From @jbrazio on March 30, 2016 20:7_

This is an interesting topic. Following your discussion on the other thread really got me thinking to buy one of these boards (DUE/RADDS).

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on March 30, 2016 20:18_

@jbrazio    Well...   If you are not in a super big hurry...   It may be best to hold off for a couple of weeks.   The real purpose of this thread is to try to identify a 'Reference Platform' for 32-Bit Marlin's.  @BioPrinter and I have been exchanging emails.   If we can get some consensus on 2 or 3 viable platforms, he is going to start lobbying the microprocessor vendors to see if one of them will step up and provide the development hardware for our group.

The thinking is, if we have 6 or 7 contributors with the exact same hardware being used as a reference platform, we are going to be able to make the transition faster.   And then for our users, that exact reference platform is pretty much guaranteed to always work and to have the latest and greatest feature set at all times.   The hardware vendor would get a big magnet pulling new 32-Bit 3D-Printer sales in their direction.   And given the size of the Marlin user base, that is nothing to sneeze at.

boelle, 2016-05-25:
_From @Sniffle on March 30, 2016 20:19_

I would think if we build a hardware interface layer it could be just about
anything even a smoothie...

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on March 30, 2016 20:24_

Yeah...   That would be the theory!   Maybe I'm just being negative.  But having other contributors with the exact same hardware gives me more confidence.

boelle, 2016-05-25:
_From @jbrazio on March 30, 2016 20:29_

@Roxy-3DPrintBoard no worries, I believe your approach makes perfect sense and will for sure speed up the effort.

I just see one potential problem, if we want to keep somehow the same existing toolkit and implement some sort of HAL to allow us to have the same code base for 8bit and 32bit then we must stay inside something supported by the Arduino platform ?

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on March 30, 2016 20:40_

It is very possible that Atmel will be the most logical choice at which point, everything is simple.   But really...  The users don't use the the IDE environment of Arduino to do anything but compile and upload.   This is a valid point for discussion, but it is very possible that another environment for the 32-Bit stuff will make more sense.   It may be that some version of GCC with a makefile can do everything they need.

(All of these issues need to be put on the table and discussed!!!)

boelle, 2016-05-25:
_From @jbrazio on March 30, 2016 20:43_

Users do not use the IDE.. but we do use the Arduino core. :-/

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on March 30, 2016 20:45_

> Users do not use the IDE.. but we do use the Arduino core. :-/

At which point we need some type of HAL to move to another vendor's microprocessor!   :)

boelle, 2016-05-25:
_From @CONSULitAS on March 30, 2016 22:28_

@Roxy-3DPrintBoard @jbrazio
see my https://github.com/MarlinFirmware/MarlinDev/issues/317#issuecomment-203663421

@Wurstnase has a running (but rather old, based on Marlin of 01. Feb. 2015, and experimental) fork for Arduino Due: https://github.com/Wurstnase/Marlin4Due
He just stopped development regarding to his readme.

boelle, 2016-05-25:
_From @Wurstnase on April 1, 2016 12:51_

Yes, i've stopped.
Anyhow, there are a lot of 32bit Controller. Take a look at mbed.org for example.
You can use the arm-none-eabi-xxx for stm32, lpc, also the Arduino Due use it. The 'problem' is the HAL or whatever you want to use. But there is no standard lib like avr-lib.

boelle, 2016-05-25:
_From @Sniffle on April 7, 2016 5:2_

I already showed this to @Roxy-3DPrintBoard 

but what do you guys think?
main board - 2 exruders+ heated bed support + x, y, z drivers
http://www.filastruder.com/collections/electronics/products/duet-controller-v0-8-5

extension board to add 4 more extruders
http://www.filastruder.com/collections/electronics/products/duex4-expansion-board

touchscreen display
http://www.filastruder.com/collections/electronics/products/paneldue

and it's already supported by reprap firmware so we have somewhere to reference for pinouts, etc.

not to mention
http://reprap.org/wiki/Duet
https://github.com/T3P3/Duet/blob/master/Duet0.8.5/Duet0.8.5_Schematic.pdf

is it just me or does this board almost seem too good to be true because it covers pretty much every thing a "moddern" printer could ask for.  I'm sure there is a gotcha somewhere but damn talk about a complete package.

edit: think i found the one downside, It uses A4982 steppers so they are hard mounted 1/16th microstepping only.  If you can think of more I'm all ears.  as far as I can tell the 0.8.5 revision is pretty stable, but if anyone can break it and find faults we can :-P

boelle, 2016-05-25:
_From @jbrazio on April 7, 2016 10:59_

I like it uses the same uC as the Due but the drawback (without even deep diving into hardware specs) it's the price, the overall solution you propose is quite expensive ($285).

boelle, 2016-05-25:
_From @Sniffle on April 7, 2016 14:35_

Yeah that was my concern as well... It is just kinda the all in one
poackage aside from the 1/16th drivers.

boelle, 2016-05-25:
_From @Roxy-3DPrintBoard on April 7, 2016 15:24_

I wonder if they have an updated version of the board they are working on with 1/64 Micro-Stepping?   Other than that, and the cost...   This is a really good setup.

boelle, 2016-05-25:
_From @Wurstnase on April 7, 2016 15:53_

The firmware on this board is not fast enough to support higher
microsteppings. They can calculate Delta without segmentation. But only at
<20kHz.

um 17:24:

boelle, 2016-05-25:
_From @jbrazio on April 7, 2016 15:56_

> The firmware on this board

This is were Marlin comes in.. or which firmware are you talking about ?

boelle, 2016-05-25:
_From @Bio-Printer on April 7, 2016 16:5_

At:  https://docs.google.com/document/d/1whV779t_660hpgTxYxFomgqrOk1CpEQi-8QttTHaI-Q/pub

They say they pre-load the RepRap firmware.  They recommend:

> The DC42 version of the Firmware covers the widest range of printer configurations - cartesian, delta and core-XY and is recommended for general use. It is available for download at https://github.com/dc42/RepRapFirmware/tree/dev.

bobc, 2016-06-15:
If anyone is still tracking this topic, the new Duet board (now called DuetWifi) is available for pre-order. It has a 120MHz Cortex M4 with hardware FPU. Looks really nice but not cheap! More details at https://www.duet3d.com/

Roxy-3D, 2016-06-15:
Thanks for the Heads Up @bobc !   Yes, we are very interested!  We just aren't in a position to move forward on the 32-Bit code right now!

Grogyan, 2016-06-15:
I'm keen to follow any topics which bring forward a 32 bit microcontroller.  Especially where key features that i'm looking for are

- Onboard Micro SD socket
- SPI breakout
- A USB Type A/C port for flash drives
- WiFi and Bluetooth
- LAN socket
- Run a 3D print server as well, like OctoPrint
- The usual drivers and hot ends, thermocouple support preferred

jbrazio, 2016-06-15:
This is the issue: £118.80

Roxy-3D, 2016-06-15:
Yes, the cost is a little high.   But it is an Open Source design.  If there are any people buying this board, there will be clones of it showing up.   And the good thing is it has everything needed for a 3D-Printer on the board.

Sniffle, 2016-06-16:
I have a duet running my kossel mini, and im building a corexy and now that
this is being released it will be what runs it instead of the duet 0.8.5

paulusjacobus, 2016-06-16:
AZSMZ ARM 32bit controller board See
https://m.aliexpress.com/item/32421892934.html for $49 US

Roxy-3D, 2016-06-16:
> AZSMZ ARM 32bit controller board See
> https://m.aliexpress.com/item/32421892934.html for $49 US

Oh!  That is the RIGHT answer!

Grogyan, 2016-06-16:
As a side thought, what about 64bit SOC's?
That way both the Firmware and the Print Server (OctoPrint) can run on the same device.

jbrazio, 2016-06-16:
> That way both the Firmware and the Print Server (OctoPrint) can run on the same device.

This is a no-go.
https://github.com/MarlinFirmware/MarlinDev/issues/398#issuecomment-220853494

bobc, 2016-06-16:
Unfortunately the Venn diagram of all people's requirements results in no solutions. As I mentioned before, Marlin currently depends on an easily available cross-platform build tool in the form of Arduino IDE in order for the user to configure Marlin.
Other cross-platform IDEs are possible, e.g. PlatformIO, but we don't want anything as heavy as Eclipse.
In practice, that means choosing a board that has a reasonably well-maintained core package for Arduino IDE. That basically means SAM3X (Arduino Due), or maybe one of https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls
Duet/DuetWifi would take a minimal amount of work to make into a downloadable Arduino core package (even if they are not officially supported by Arduino). 
Anything based on LPC1769, or Linux capable SOCs is out of scope without a *lot* of work in Marlin, maintaining a build tool, or developing a new core package.

Roxy-3D, 2016-06-16:
Wow!   That is a valuable link!     

The only Cortex boards that are supported (in that list) are:

- RedBear: https://redbearlab.github.io/arduino/package_redbear_index.json
-- RedBear Duo (Cortex-M3, WiFi + BLE)
-- RedBear discussion forum
-Konekt Dash/DashPro (Official): http://downloads.konekt.io/arduino/package_konekt_index.json
-- Cortex M4-based global cellular dev kits, support for all board variants, support for USB and over-the-air programming straight from within Arduino IDE

WZ9V, 2016-07-07:
How about this one 
http://www.arduino.org/products/boards/arduino-star-otto

bobc on RepRap has been working on a update to the Due RAMPS card and he is keeping an eye on this one.  So there may be all the hardware needed before long.

esenapaj, 2016-07-07:
It's interesting...

Blue-Marlin, 2016-07-07:
Enough pins, fast, hardware floating point, Ardurino IDE. Very promising.

Drakelive, 2016-07-07:
http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f446ze.html

http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f767zi.html

All solutions already available

Drk

Wurstnase, 2016-07-07:
I have a Nucleo 411 with just 100MHz. But it works quit well!
https://www.youtube.com/watch?v=lXkOv9qoCp0#00m45s

esenapaj, 2016-07-08:
Well, it seems that Arduino Due was retired...
https://www.arduino.cc/en/Main/ArduinoBoardDue

Wurstnase, 2016-07-08:
Yes, the ArduinoDue from arduino.cc not from arduino.org. Little bit
confusing. For the STAR OTTO you will need then also the ArduinoIDE from
arduino.org, not that one most of the users uses (from arduino.cc).

11:11 Uhr:

jbrazio, 2016-07-11:
Otto is promising indeed.

bobc, 2016-07-13:
I haven't seen this before http://www.st.com/content/st_com/en/products/evaluation-tools/solution-evaluation-tools/computer-and-peripherals-solution-eval-boards/steval-3dp001v1.html - a 3D printer eval board from ST. They have a version of Marlin for it called Marlin4ST https://github.com/St3dPrinter/Marlin4ST

Might explain some recent support queries...

WZ9V, 2016-07-13:
Not bad, $120 from digikey for a six driver board.

Roxy-3D, 2016-07-13:
> Not bad, $120 from digikey for a six driver board.

I was thinking the same thing.   And they have done an initial port of Marlin to the board.  It is too bad they compile their version of Marlin with the Cube software.     I wonder if anybody knows how complicated it would be to shift that code base to Arduino?

esenapaj, 2016-07-16:
>to shift that code base to Arduino

Possibly ["Arduino STM32"](https://github.com/rogerclarkmelbourne/Arduino_STM32) + modification realize it?
But it's extremly difficult (probably impossible) for me.

And, it looks like that Marlin4ST is unfinished. For example,
[FAST_PWM_FAN](https://github.com/St3dPrinter/Marlin4ST/blob/master/stm32_cube/Middlewares/Third_Party/Marlin/Marlin_main.cpp#L4821), [HEATER_0_USES_MAX6675](https://github.com/St3dPrinter/Marlin4ST/blob/master/stm32_cube/Middlewares/Third_Party/Marlin/temperature.cpp#L1204), [ADVANCE](https://github.com/St3dPrinter/Marlin4ST/blob/master/stm32_cube/Middlewares/Third_Party/Marlin/stepper.cpp#L841), etc...

Roxy-3D, 2016-07-16:
@St3dPrinter   Can you please join this thread?    The board shown at the link that @bobc found is impressive.   If we could get 6 or 8 boards into the right developer's hands, we could get make it the reference platform for 32-Bit Marlin and bring up RC-7 on it.

jbrazio, 2016-07-16:
I ordered a Due for less than $12.. @bobc I only need your magic now ! :-P

esenapaj, 2016-07-17:
That reminds me..., new Due-based boards was released few month ago.
Ultratronics - ReprapWorld.com
https://reprapworld.com/?products/listing&cPath=1591__1750

Roxy-3D, 2016-07-17:
Are you saying it uses the same processor as the Due ?   Because it doesn't look like a Due plugs into that board.

esenapaj, 2016-07-17:
>Are you saying it uses the same processor as the Due ?

Yes it is. These are combined type like a [Duet](http://reprap.org/wiki/Duet). Maybe because my bad English? Sorry. 
These boards are already supported by Repetier-Firmware and MarlinKimbra4due (but experimental).

Roxy-3D, 2016-07-17:
> And, these are separated type (plugin type?).
> DUE3DOM and DUE3DOM MINI. http://fabrykator.pl/seasidecustoms/due3dom-elektronika-32bit-dla-drukarek-3d/  These are supported by Repetier-Firmware and my Marlin Duo.

Thank You.  That was helpful.   The web link https://reprapworld.com/?products/listing&cPath=1591__1750 didn't make everything clear to me.  It looks like the difference between the Full and the Lite board is the Lite is only setup for one extruder?   Did they just leave parts unpopulated that can be put in later by the user?   And one more question, do you know if Arduino can generate code directly for the Ultratronics boards?

esenapaj, 2016-07-17:
>Lite is only setup for one extruder?

Compare list says difference between Ultratronics FULL and Ultratronics LITE are 
Ethernet port(available/not available) and number of onboard thermocouples([MAX31855?](https://forum.repetier.com/discussion/1596/compilation-problems-with-thermocouples-max31855))(four/two).
number of extruders is same.
https://reprapworld.com/?elecinfo

>Did they just leave parts unpopulated that can be put in later by the user? 

Sorry, I'm not sure.

>Arduino (IDE) can generate code directly for the Ultratronics boards?

Yes it is.
Constituent of Ultratronics is Arduino Due + expansion circuit.
It's like a [Printboard](http://reprap.org/wiki/Printrboard)(Arduino MEGA2560 + expansion circuit) that it's supported by official Marlin. 
Also [datasheet](https://reprapworld.com/documentation/datasheet_ultratronics10_05.pdf) says:
>The design is based on Arduino Due and combines the best features of Megatronics to a new more powerful board. 
>The board is therefor compatible with Arduino software, but optimized for 3D printers.

Roxy-3D, 2016-07-17:
> and number of onboard thermocouples(MAX31855?)(four/two).
> number of extruders is same.

The reason I was thinking it only handled 1 extruder was because you need 1 temperature sensor for the bed and you need 1 for each extruder.   So if the Lite board only handles two temperature sensors, that made me think it was for 1 extruder.

Blue-Marlin, 2016-07-17:
There are another 4 thermistor (!= thermocoupler) inputs.

Roxy-3D, 2016-07-17:
Ok!   That makes more sense.    In that case, I think the Lite board would handle most people's needs.   With 4 thermistor inputs, you can have 3 nozzles, or you can have a backup sensor.  

It is encouraging that we are seeing multiple 32-bit boards that could work all of a sudden.

esenapaj, 2016-08-01:
I've found some STM32 boards.

・STduino and Marlin_STM32
STduino原型板如何使用(Arduino Compatible) - MakerLab创客实验室
http://www.makerlab.me/guides/2/articles
MakerLabMe/Marlin_STM32: 3d printer Marlin firmware for 32 bit arm cpu STM32F10X
https://github.com/MakerLabMe/Marlin_STM32

But I wasn't able to find where do they sell STduino.

・Rest32
rest32.tk
http://www.rest32.tk/
berryelectronics/ReST32: A modular 3D Printer Controller based on the STM32Duino project
https://github.com/berryelectronics/ReST32
Anyone tried to make a 3D Printer controller? - Page 19 - Arduino for STM32
http://www.stm32duino.com/viewtopic.php?f=19&t=572&start=180#p16470

alexxy, 2016-08-04:
There is arduino compatible STM32 board called nucleo. It has similar to arduino pin layout

JustAnother1, 2016-08-05:
There are several nucleo boards with different STm32Fxxx processors. But then only have the Arduino uno pins. Not the Arduino mega pins like the Otto.

St3dPrinter, 2016-08-08:
> @St3dPrinter  Can you please join this thread? The board shown at the link that @bobc found is impressive. ( http://www.st.com/content/st_com/en/products/evaluation-tools/solution-evaluation-tools/computer-and-peripherals-solution-eval-boards/steval-3dp001v1.html )
> If we could get 6 or 8 boards into the right developer's hands, we could make it the reference platform for 32-Bit Marlin and bring up Marlin Release Candidate RC-7 on it.

Sorry, for the late answer but I've  just come back from the summer break. 
For sure, it will be great to make this board a reference platform for the 32-bit Marlin.
The problem is that stm32 (and particularly the stm32f401 which is the board MCU) is not natively supported by the Arduino Ide.  
For now, we are preparing an adaptation of the RC7 for this board as the existing FW, the [Marlin4ST](https://github.com/St3dPrinter/Marlin4ST), is based on the "old" 1.0.2 version. 
Its project tree should be closest to the Marlin's one but it will require an OpenSTM32 IDE (based on Eclipse).

Regarding the possibility to provide some boards for evaluation to Marlin's firmware member, it  is something that can be discussed, particularly if you think it is possible to natively supported it.

esenapaj, 2016-08-08:
@St3dPrinter 
I'm not a Marlin's firmware member but I'd like to ask to you, 
Is connectivity of STEVAL-3DP001V1 compatible with STM32F4 Nucleo?
If it's true,  there is some possibility of supporting that STEVAL-3DP001V1 can be supported by Arduino IDE with [Arduino STM32 library](https://github.com/rogerclarkmelbourne/Arduino_STM32).

And, I make a request.
Would it be possible that you exclude STEVAL-3DP001V1 from EAR?
Because EAR interdict that STEVAL-3DP001V1 is exported to my country (Japan).

Roxy-3D, 2016-08-08:
@St3dPrinter 

> Sorry, for the late answer but I've just come back from the summer break. 

Not a problem!   :)

> For sure, it will be great to make this board a reference platform for the 32-bit Marlin.  The problem is that stm32 (and particularly the stm32f401 which is the board MCU) is not natively supported by the Arduino Ide.

I don't know that the reference 32-bit platform has to be natively supported by Arduino's IDE.   But it probably does need to be something Arduino's IDE can generate code for.  We are having problems with older versions of Arduino and trying to keep the code running on both the 8-bit and 32-bit platform is going to be a challenge.   Adding an extra compiler and development system into the mix might be too much.   

I'm not sure this is possible, but I think Arduino's IDE can be extended to support that microprocessor.  We do not have the expertise to do that.   But some of the Marlin fans like @Bobc might be able to help guide us on that topic.  Bobc, can you please offer your thoughts on this topic?

With that said, I'm not sure the Reference 32-bit platform for Marlin has to be natively supported by Arduino's IDE.   I think it would be acceptable to have a set of working board files that a person could use to supplement their Arudino IDE environment to handle the 32-bit platform.   We could package the additional board files with the 32-bit version of Marlin and also the documentation and support to use them.

> Regarding the possibility to provide some boards for evaluation to Marlin's firmware member, it is something that can be discussed, particularly if you think it is possible to natively supported it.

The entire Marlin Team would want to tie out on this decision.   But if we can get the Arduino IDE to compile code for the board, it will make the decision much easier!    May I suggest that you consider the possibility of  getting your board supported by Arduino's IDE as more important than porting the RC-7 code base to the board?   If we had the ability to compile and debug using the Arduino IDE, we can bring up RC-7 and start the process of putting a HAL (Hardware Abstraction Layer) in place to insulate the code from differences in the hardware.   In that scenario, the users would configure the Arduino IDE environment for the type of board they have (8-bit or your 32-bit) and we would make the code adapt to what is selected.

tcurdt, 2016-08-08:
...but there are already ways to compile STM32 with the Arduino IDE. It's just not out-of-the box.
[image]

Roxy-3D, 2016-08-08:
> ...but there are already ways to compile STM32 with the Arduino IDE. It's just not out-of-the box.

That is what I was hoping was the case.  Can you tell us what else has to be changed to configure the Board Menu to support the STEVAL-3DP001V1 board?  I presume the additional Board files would have to specify things like where the interrupts are and memory configuration.   And most likely they would also have to set up the various options like clock speed and how the GPIO registers work.   

That is why I suggested:

> May I suggest that you consider the possibility of getting your board supported by Arduino's IDE as more important than porting the RC-7 code base to the board? If we had the ability to compile and [...]

If we could get Arduino to generate code for the board, I'm pretty sure we can bring up RC-7 on it and start the process of formally supporting the board!

bobc, 2016-08-08:
@Roxy-3D I've created 3 CPU packages for Arduino, so I have a pretty good idea what is involved. I did the original Arduino_STM32 package, it was quite easy as Leaflabs already wrote the code for Maple, I wrapped it up for the latest IDE. The hard part is bringing the codebase up to date, fixing bugs, and supporting many variants, which Roger Clark and others have been doing very well over at STM32Duino.

But the question is still, does Marlin team want to use an "unofficial" CPU package, or support an in-project package? An Eclipse based IDE like SWSTM32 is fine for developers like me (in fact I much prefer it to Arduino IDE), but Marlin still requires Arduino IDE for users to configure their firmware.

Personally, I still think the best board for an "official" Marlin32 is one based on SAM3X8E (ie. Due compatible), or close relative such as Duet.

It seems I can't order STEVAL-3DP due to "export restrictions", so it's a non-starter for me. I am also very unimpressed with ST's aggressive approach to IP (i.e. patent trolling), so I would not consider them as a good Open Source partner.

[428 more replies]

closed 2018-05-11 by boelle