MarlinFirmware/Marlin #3856: FORK 5 Axis....

boelle, 2016-05-25:
_From @D9ve on September 10, 2015 16:18_

Hello to all, 

I know this really isn't a issue, but... I am currently looking into the marlin firmware to see it it is possible to customise this for a 5 dof/axis 3D printer, I will also develop a simple slicer as well an integrated 3D scanner.

Primarily the software I will be using is MACH3, RHINO, and atmel studio 6.2 for development

So really I am asking do you think customising this code for 5 axis is possible? 

I am using ramps 1.4, with only a single extruder, this leaves me with 2 stepper drivers available, mind one is on the AUX pins 23 25 29!

What do you think? should I be placing my ideas and comments on another forum? Any info would be greatly appreciated.

_Copied from original issue: MarlinFirmware/MarlinDev#158_

boelle, 2016-05-25:
_From @Dark-Guan on February 29, 2016 2:36_

A 5 dof/axis 3D printer mean there is 6 motors . Marlin can aford it

jbrazio, 2016-05-27:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

Haraldschmidt, 2019-06-13:
> From @D9ve on September 10, 2015 16:18
> Hello to all,
> I know this really isn't a issue, but... I am currently looking into the marlin firmware to see it it is possible to customise this for a 5 [...]

-------------------------------------------------------------------------------------------------------------------

I am currently searching for a slicer which can handle 5 axis 3D printing on an optimized prusa i3 MK3. Are you still working on that? 
Thanks a lot for your response in advance!
Best regards,
Harald

closed 2020-11-03 by shitcreek