MarlinFirmware/Marlin #3860: PlatformIO fail when settings use U8glib

boelle, 2016-05-25:
_From @Nandox7 on September 25, 2015 22:6_

Hey,

When trying to run platformio with settings that require the usage of the U8glib, for example the Full Graphics LCD controller, it fails to compile missing to find the U8glib.h file.

```
Marlin.ino:44:101: fatal error: U8glib.h: No such file or directory
compilation terminated.
scons: *** [.pioenvs/mega2560/src/tmp_ino_to.o] Error 1
```

I did tried to manually install the lib for platformio using: platformio lib install 9
(as seen here: https://libraries.io/platformio/U8glib)

It install correctly but still doesn't work.

_Copied from original issue: MarlinFirmware/MarlinDev#216_

boelle, 2016-05-25:
_From @Wackerbarth on September 25, 2015 22:18_

The start of PlatformIO support was entered at the request of one of our users who had promised to "flesh it out". However, he never provided any updates. Unless someone fixes it, we will soon remove it entirely.

boelle, 2016-05-25:
_From @Nandox7 on September 25, 2015 22:24_

I can look into this then as I'd really like to avoid using the junky Arduino UI.

boelle, 2016-05-25:
_From @clefranc on September 25, 2015 22:29_

@thawkins https://github.com/MarlinFirmware/Marlin/issues/2472

boelle, 2016-05-25:
_From @thawkins on September 25, 2015 23:0_

Platformio lib install 7 

Works for me for a ramps card. 

Im not sure what extra is needed to "flesh it out", a number of folks over on printrbottalk are already using it to build marlin for printrboards. It works. Does it need any more. 

If its docs that are needed then i will try and write something  this week end.

boelle, 2016-05-25:
_From @Nandox7 on September 25, 2015 23:15_

Got it.
With lib install 7 it works.

Maybe just add that to the docs/readme. As more like me may just find first
the lib install 9 and it will fail.

-Nando

boelle, 2016-05-25:
_From @thawkins on September 26, 2015 0:11_

I will start a wiki page on the release repo to document using platformio
with marlin.

boelle, 2016-05-25:
_From @Nandox7 on September 26, 2015 13:47_

Perfect! If i can help in any way please let me know.

PS: I've been searching but haven't found answer for thi yet. Is it
possible to just compile one environment?
I just need the atmega2560 but ifi  do platformio run it will compile all
the ones defined. Solution so far isto edit the ini file and remove the
entries for the ones i dont want.

boelle, 2016-05-25:
_From @thawkins on September 27, 2015 0:59_

Platformio run -e mega2560

boelle, 2016-05-25:
_From @thawkins on September 27, 2015 1:2_

http://docs.platformio.org/en/latest/userguide/cmd_run.html

boelle, 2016-05-25:
_From @thinkyhead on April 18, 2016 3:34_

Sorry this issue is lingering for so long. As soon as 1.1.0 is released we will be migrating code into the new file layout (as in the MarlinKimbra fork), and we'll be paying much more attention to issues in MarlinDev to see what still applies with the latest code.

jbrazio, 2016-05-27:
PlatformIO is no longer supported.

closed 2016-05-27 by jbrazio

thawkins, 2016-05-27:
So what is supported as an alternative to using the arduino ide, which is
garbage.

thinkyhead, 2016-05-27:
Need to get more feedback on PlatformIO and who uses it.

boelle, 2016-05-27:
most users will use arduino ide regardless if its garbage or not, but dev people can use anything they want

jbrazio, 2016-05-27:
Actually.. the IDE is used with "one button": upload, how can it suck ?