MarlinFirmware/Marlin #3871: Filament Width - Dual Extruder

boelle, 2016-05-25:
[image] **Issue by [nio40k](https://github.com/nio40k)**
_Wednesday Feb 24, 2016 at 09:50 GMT_
_Originally opened as https://github.com/MarlinFirmware/MarlinDev/issues/372_

----

Please add support for the Filament Width Sensor in combination with Dual Extruder.

Currently the filament width sensor can only used with one extruder. I would like to use two sensors with two extruders.

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Tuesday May 03, 2016 at 03:20 GMT_

----

What pins do you connect your sensors to?

nio40k, 2016-06-06:
I use a Ramps 1.4 and will use what ever pin you recommend ...

thinkyhead, 2016-06-07:
Basically, we need to specify a second pin for each board that currently has a `FILWIDTH_PIN`. It can be called…
```cpp
#define FILWIDTH_2_PIN      ?  // ANALOG NUMBERING
```
Currently `FILWIDTH_PIN` is ANALOG pin 3 on those boards that specify it. We only need to pick a next pin to use, define `FILWIDTH_2_PIN`, and then add code alongside the current `FILWIDTH_PIN`-dependent code, which handles the second pin. And we need to add a new configuration option, such as `FILAMENT_WIDTH_SENSOR_E1`, to enable the second sensor.

There's a lot of code related to `FILAMENT_WIDTH_SENSOR` so it's a few hours of work, at least…

boelle, 2017-04-26:
maybe add this one to the 1.1.1 milestone ? if its still relavant?

closed 2018-03-16 by thinkyhead