MarlinFirmware/Marlin #3873: FeatureRequest - Automatic Mesh Leveling

boelle, 2016-05-25:
[image] **Issue by [mirage335](https://github.com/mirage335)**
_Monday Feb 15, 2016 at 22:28 GMT_
_Originally opened as https://github.com/MarlinFirmware/MarlinDev/issues/362_

----

Please consider implementing automatic mesh leveling. Current probe-based automatic leveling support does not correct well for bumps, Y-axis twisting, X-axis sagging, or other nonlinear deformities.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Feb 15, 2016 at 23:11 GMT_

----

I'm working on it for Delta's.   If I'm careful, I might be able to make it work for Cartesian and Delta machines.

boelle, 2016-05-25:
[image] **Comment by [mirage335](https://github.com/mirage335)**
_Tuesday Feb 16, 2016 at 02:12 GMT_

----

Much thanks. My request is for Cartesian machines, like the LulzBot Taz, Mini, etc.

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Sunday Feb 28, 2016 at 03:34 GMT_

----

Yes please. im willing to donate some money if it will help get it added sooner :)

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Wednesday Mar 02, 2016 at 04:11 GMT_

----

Donating test time would be more helpful.   I'll let you know when there is something 'feature rich' enough and stable enough to test.

boelle, 2016-05-25:
[image] **Comment by [Sniffle](https://github.com/Sniffle)**
_Wednesday Mar 02, 2016 at 04:15 GMT_

----

I would roxy but my delta is running smoothie :-D

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Wednesday Mar 02, 2016 at 14:05 GMT_

----

Im ready and willing to do some testing.

boelle, 2016-05-25:
[image] **Comment by [mirage335](https://github.com/mirage335)**
_Wednesday Mar 02, 2016 at 18:03 GMT_

----

Also ready and willing to test automatic mesh leveling.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Thursday Mar 03, 2016 at 19:37 GMT_

----

Here is the header block in front of the code I'm writing.   You should be able to understand a lot of what I'm thinking from the comment block.    Probably, you can understand why it is going to take 5 or 6 weeks to get the system fully baked and ready to hand out for people to try.

Are there any other feature requests or omissions that need to be handled?  One thing I'm wondering about is if it makes sense to store the F-Fade parameter on a per Mesh basis.  I think
usually people will want to fade at the same rate independent of which piece of glass they are using.   But I'm not sure that is true.  I also have not figured out the best way to handle the fine tuning of the Mesh without an LCD Panel.  But it can be done with some special parameters on the G29 P4 command from PronterFace.  I just don't know how I'm going to do that yet.

 (Please ignore the messed up alignment...  GitHub isn't doing the right thing even with code tags around the block)

 ~~~~
/**
   * G29: Delta Mesh Based Compensation System
   *
   * Parameters understood by this leveling system:
   *
   *  A		Activate  Activate the specified Mesh.   All loads and stores of Mesh Compensation data
   *                  will occur at the Activated location in EEPROM.  A separate Mesh definition is 
   *  	              needed for each piece of glass you will printing on.  These pieces of glass need to
   *  	              be positioned the same during printing as during the Mesh Calibration (and
   *  	              storage) process.  Activation of a specified set of Mesh Data happens before
   *  	              any subsequent processing of the command line.   Activation of a specified set
   *  	              of Mesh Data does not implicitly Store or Load any Mesh data to the EEPROM.  
   *  	              It also does not affect the default Mesh Data specified by the EEPROM.  It only
   *  	              affects the current session and an M500 is required to update the contents of
   *  	              the EEPROM.   The GCode command:  G29 A 3 S   performs the same as  G29 S A 3 
   *
   *  F		Fade      Fade the amount of Mesh Based Compensation over a specified height.  At the
   *  		          specified height, no correction is applied and natural printer kenimatics
   *  		          take over.
   *
   *  L		Load     Load the currently Active Mesh from Active EEPROM location.
   *
   *  M		Map	     Display the Mesh Map.
   *
   *  P0   Phase 0   Zero Mesh Data and turn off the Mesh Compensation System.  This reverts the
   *                 Delta Printer to the same state it was in before the Delta Mesh Based Compensation
   *                 System was turned on.
   *
   *  P1  Phase 1  Perform Phase 0 and continue with automatic generation of the Mesh data using
   *                   the Z-Probe.   Depending upon the values of DELTA_PROBEABLE_RADIUS and 
   *                   DELTA_PRINTABLE_RADIUS some area of the bed will not have Mesh Data automatically 
   *                   generated.  This will be handled in Phase 2.  The Delta Mesh Compensation System is 
   *                    turned on at the conclusion of Phase 1.
   *
   *  P2	Phase 2   Probe areas of the Mesh that can not be automatically handled.
   *
   *  P3	Phase 3   Manually fill in the unprobed areas of the Mesh. 
   *
   *  P4	Phase 4   Fine tune the Mesh.  The Delta Mesh Compensation System assume the existance of
   *                  an LCD Panel.  It is possible to fine tune the mesh without the use of an LCD Panel.
   *                  The System will search for the closest Mesh Point to the nozzle.  It will move the
   *                  nozzle to this location.  The user can use the LCD Panel to carefully adjust the nozzle
   *                  so it is just barely touching the bed.  When the user clicks the control, the System
   *                  will lock in that height for that point in the Mesh Compensation System.
   *
   *                  Phase 4 has several additional paramters that the user may find useful.  Phase 4
   *                  can be started at a specific location by specifying an X and Y parameter.  Phase 4
   *                  can be requested to continue the adjustment of Mesh Points by using the R(epeat) 
   *                  parameter.  If the Repetition count is not specified, it is assumed the user wishes
   *                  to adjust the entire matrix.  The nozzle is raised after a Mesh Point is calibrated
   *                  and before the nozzle is moved to the next location.  This height can be controled
   *                  with the H parameter.  The default value is 5mm.
   *
   *                  Phase 4's general form is G29 P4 [R # of points] [X position] [Y position] [H height]
   *
   *  P5	Phase 5	  Undefined right now.
   *
   *  S		Store	  Store the current Mesh in the Activated area of the EEPROM.
   *
   *  T		Test      Load specified Test Pattern to assist in checking correct operation of system.
   *
   *  Z		Zero      Probes to set the Z Height of the nozzle.  With no specified paramter 
   *  			      Z_PROBE_OFFSET_FROM_EXTRUDER is used without any extra correction.  If a parameter
   *  			     is specified, it is added to Z_PROBE_OFFSET_FROM_EXTRUDER for the final zero position.
   *
   */
   ~~~~

boelle, 2016-05-25:
[image] **Comment by [mirage335](https://github.com/mirage335)**
_Thursday Mar 03, 2016 at 20:51 GMT_

----

Thanks for working on this. Does indeed seem like quite an extensive codebase to develop.

Is the idea to store multiple meshes in EEPROM, to be recalled for use with different print surfaces? Personally, I still might want to regenerate the mesh before every job. In particular, it is becoming common practice to laminate glass with PEI or other materials, and the adhesive layer may change thickness between jobs.

Fade rate might need to change depending on the job or tools used. Exceptionally precise jobs might benefit from little or no fading.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Thursday Mar 03, 2016 at 21:04 GMT_

----

> Is the idea to store multiple meshes in EEPROM, to be recalled for use with different print surfaces? 

Yes.  There are some constraints however.  I want to be able to do up to a 15 x 15 Mesh.  So, that is a lot of data.   Depending upon the resolution (and how well I can compact the data), higher resolution Mesh's might limit how many of them can be stored in EEPROM. 

In reality, I mostly use one piece of glass.  But some times while one is in the freezer to help me break the parts free, I'll use another one to get a 2nd print started.   So for me, storing two Mesh's would be fine.

> Personally, I still might want to regenerate the mesh before every job. In particular, it is becoming common practice to laminate glass with PEI or other materials, and the adhesive layer may change thickness between jobs.

Well...  More power to you!  :)  But even with a fancy system that helps you dial in the Mesh as fast as possible (like I'm trying to do), I suspect it will take 20 or 30 minutes to get the Mesh perfect.  I kind of doubt you will want to do that for every print.

I'm also thinking of doing some other items as a follow up.  For example, suppose you get a Mesh perfectly defined.  But when you put the piece of glass down on the heat bed there is a small piece of dirt that raises an edge or corner.  It should be possible to do a 3-Point leveling on the Mesh to correct for that also.   I'm thinking we get the High Resolution Mesh Leveling out there and see what kind of things it has problems handling.  And then we will go after those next.

> Fade rate might need to change depending on the job or tools used. Exceptionally precise jobs might benefit from little or no fading.

Yes.  I can see how that could be the case.   Probably the thing to do is get it working and get some experience with the concept.  And once people get comfortable (and knowledgeable)  about it's use, we will have lots of ideas to deal with the limitations of the first implementation.

boelle, 2016-05-25:
[image] **Comment by [mirage335](https://github.com/mirage335)**
_Thursday Mar 03, 2016 at 21:30 GMT_

----

Well, 30 minutes is a long time. Probing a 5x5 grid takes tens of seconds, and happens before the hotend finishes heating up. Is manual intervention expected to be necessary for routine use?

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Thursday Mar 03, 2016 at 21:51 GMT_

----

> Well, 30 minutes is a long time. Probing a 5x5 grid takes tens of seconds, and happens before the hotend finishes heating up. Is manual intervention expected to be necessary for routine use?

Maybe fine tuning won't be necessary.   I don't know yet because the code isn't done and working.  It is very possible that if you are content printing within the area the probe can measure things, no fine tuning will be necessary.   And if your bed is fairly flat, a 5 x 5 grid just might work fine.

I'm mostly doing this because it is an interesting exercise and I want it to handle a very poorly calibrated Delta.    Do you remember the original Auto Bed Leveling videos that showed up when the feature first came out?   Some people were printing on a bed raised a couple of inches on one side.   That was enough to get me to print some parts and order a Servo.

I want the same thing to happen except with a poorly calibrated Delta Printer.   If DELTA_RADIUS and DELTA_DIAGONAL_ROD are off by .5mm the movement becomes horribly un-flat.  I want the system to handle a huge error like that.   (That is kind of like saying I want to handle as big of an error as the original Auto Bed Leveling videos showed.)   Even though I'm doing this for Delta's, there is no reason it can't be made to work for Cartesian Printers too.

So...   I'm way over designing the system just to make sure I can get there.  If your printer is pretty normal and well calibrated, it is very possible you don't need anything but the Phase 1 stuff if you don't want to print outside of the probed area.   If you want to print outside of the probed area, you have no choice but to manually fill in the rest of the Mesh.  But that is the whole purpose of Phase 2.  It's purpose is to let you very quickly fill in the rest of the grid manually.  It will move the nozzle to each location of the grid it needs data for and let you quickly dial the nozzle down to the bed and then it will continue to the next unfilled location of the Mesh.

boelle, 2016-05-25:
[image] **Comment by [mirage335](https://github.com/mirage335)**
_Thursday Mar 03, 2016 at 22:14 GMT_

----

Makes sense. Leveling a delta machine can require some interesting math to begin with, handing extreme errors like that could take a lot more precise measurements.

Filling in areas outside the mesh... would expect some squares outside of the mesh could be estimated based on nearby squares, at least for cartesian machines. Likely, the two or three points needed to define an adjacent square could be estimated rather well simply by drawing a stright line through two other points that have been probed. At least for cartesian machines with flat glass surfaces.

When this is operational, it is also possible cartesian machines will again become more competitive with delta machines, as the demand for mechancial performance is lessened.

boelle, 2016-05-25:
[image] **Comment by [samwyze](https://github.com/samwyze)**
_Friday Mar 04, 2016 at 07:01 GMT_

----

I'm also very interested in this feature, though for Cartesian kinematics with a servo mounted z-min endstop probe. I have a heap of messy glass plates that are far from flat, so am willing to help with testing when the time comes.

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Monday Apr 18, 2016 at 04:09 GMT_

----

how are things progressing?

boelle, 2016-05-25:
[image] **Comment by [psyvision](https://github.com/psyvision)**
_Monday Apr 18, 2016 at 09:12 GMT_

----

Willing to donate some testing time for this on a Cartesian printer with IR z-probe.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Tuesday Apr 19, 2016 at 12:49 GMT_

----

> how are things progressing?

It's coming along.   And in fact, it may be possible to incorporate (unify) the 3-point and grid leveling into it.   But I keep getting sucked into my brother's software projects and that is a big time drain.

boelle, 2016-05-25:
[image] **Comment by [Roo123](https://github.com/Roo123)**
_Tuesday Apr 19, 2016 at 17:03 GMT_

----

assign your bro a lower interrupt priority ! ;-)

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Wednesday Apr 20, 2016 at 14:55 GMT_

----

Yeah, that would be nice.   The problem is he seems to be a NMI.

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Thursday Apr 28, 2016 at 03:40 GMT_

----

Hope to see Auto Mesh Leveling in RC7 :)

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Friday Apr 29, 2016 at 00:03 GMT_

----

> Hope to see Auto Mesh Leveling in RC7 :)

No you don't!   You want RC7 to be stable and well shook out!   But it will be available as a fork of the last Release Candidate when it goes Golden.  So if that happens, it won't be a big deal to bring up if you have already brought up the RC family.

boelle, 2016-05-25:
[image] **Comment by [gddeen](https://github.com/gddeen)**
_Friday Apr 29, 2016 at 14:04 GMT_

----

My Wishes:

I've always wanted to simply turn a dial to raise the printhead up or down VERY small increments at the beginning of a print. (Cartesian printer). The idea being to adjust the first adhesion layer.  Different
materials need different heights to make them stick to the bed. Also at different temperatures they have different viscosities which changes the bead on the first layer.

MID-PRINT is would be nice to say up, up, up, up, there...  Even if the print had to pause, raise 10mm, recalculate everything, lower then resume.

As faster boards get used, it would be nice to have a PC and do any type of translation to the mesh bed level (While it is printing...)

boelle, 2016-05-25:
[image] **Comment by [gddeen](https://github.com/gddeen)**
_Friday Apr 29, 2016 at 14:14 GMT_

----

Are there any specific test platforms I should build to work out MABL? I might be able to construct a bunch of tests that I can run and rerun to validate the features. (A System test.)  Are you using something
like a mega2560 or a new 32bit board?

I can also unit test cartesian if I get much more info...

boelle, 2016-05-25:
[image] **Comment by [Blue-Marlin](https://github.com/Blue-Marlin)**
_Friday Apr 29, 2016 at 14:59 GMT_

----

@gddeen 
> MID-PRINT is would be nice to say up, up, up, up, there... 

Did you try `BABYSTEPPING`? ~~~Ah - sorry - not for DELTA.~~~ OK for Z for DELTA

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Wednesday May 18, 2016 at 20:47 GMT_

----

Is there some code I can test out, insert in RC6?
to get Auto Mesh levelling working?

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Thursday May 19, 2016 at 01:54 GMT_

----

@adamfilip Word on the street is that @Roxy-3DPrintBoard is working on that — making it so that we can have Automatic Mesh Leveling, and also Manual Grid Leveling (and Manual Three-Point Leveling?). But I don't know how far along she is.

boelle, 2016-05-25:
[image] **Comment by [adamfilip](https://github.com/adamfilip)**
_Thursday May 19, 2016 at 16:52 GMT_

----

I noticed that the new Reprap i3 Mk2 has automatic Mesh leveling listed as a feature
http://shop.prusa3d.com/en/3d-printers/53-original-prusa-i3-mk2-3d-printer.html

I wonder what code they are using to accomplish it.

boelle, 2016-05-25:
[image] **Comment by [Blue-Marlin](https://github.com/Blue-Marlin)**
_Thursday May 19, 2016 at 18:15 GMT_

----

At least in their public repository there is nor the slightest hint to MBL (https://github.com/prusa3d/Prusa-Firmware)

GreatGrizzly, 2016-05-26:
I talked to their customer service, and they confirmed that it is closed source. I made sure to confirm that it was to compensate for bed imperfections, not just bed tilt.

adamfilip, 2016-05-26:
reprap is closed source!!?? now im really confused.

Roxy-3DPrintBoard, 2016-05-26:
> reprap is closed source!!?? now im really confused. 

I have not dug into what has been said here.   My guess is they have pre-compiled a module and offer it to anybody that wants to link Marlin against it.   And they probably provide the source to a version of Marlin that calls into their module.    But that is just my guess.    

If you post the .HEX file or what ever they do provide, somebody will know how to disassemble it and we will get a much better idea what they are doing.

adamfilip, 2016-05-26:
Here is the firmware Hex
[prusa3d_fw_3_0_0.zip](https://github.com/MarlinFirmware/Marlin/files/285272/prusa3d_fw_3_0_0.zip)

Roxy-3DPrintBoard, 2016-05-26:
This does not look legitimate.  They are appear to be distributing Marlin without including the source code.   The approach I mentioned above would be legitimate.  They do not appear to have done that.

I think a few pointed questions to them are appropriate.   When requested for the source, they are supposed to provide it. Are there any volunteers to send an email requesting access to the source that created these .HEX files?   If possible, it might make sense to CC: their legal department besides Customer Support.

GreatGrizzly, 2016-05-26:
There is a link to the source code where the hex files can be downloaded... However, the code is a few months old. I don't think that is the right source code. I sent a message to their customer service to get an official response as to where their source code is.

jbrazio, 2016-05-27:
Is it this ?
https://github.com/prusa3d/Prusa-Firmware

Eriobis, 2016-05-31:
Hi, I have tried the RC6 auto leveling ... it does not seem to be working correctly. Maybe i'm doing something wrong. I'm doing g28 and then G29 and then print ...

I did some tests with verbose ans Topology
`ecv: 
Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  -0.29714 -0.14046 -0.00119 -0.08326 -0.29714
Recv:  -0.04347 +0.13808 +0.15051 -0.00617 -0.06337
Recv:  +0.05352 +0.30719 +0.31465 +0.11321 -0.02358
Recv:  -0.05591 +0.04606 +0.21268 +0.05849 -0.15538
Recv:  -0.16533 +0.07590 +0.17787 -0.07580 -0.23994
Recv: 
Recv: 
Recv: Corrected Bed Height vs. Bed Topology:
Recv:  +0.00000 +0.17593 +0.33444 +0.27162 +0.07700
Recv:  +0.23139 +0.43218 +0.46386 +0.32644 +0.28849
Recv:  +0.30609 +0.57901 +0.60572 +0.42353 +0.30599
Recv:  +0.17438 +0.29560 +0.48147 +0.34653 +0.15190
Recv:  +0.04268 +0.30316 +0.42437 +0.18995 +0.04506
Recv: 
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 -0.000535
Recv: -0.000000 +1.000000 -0.000637
Recv: +0.000535 +0.000637 +1.000000
Recv: echo:busy: processing
Recv: echo:busy: processing`

To make the correction apply I have to run G29 without the "T" right ??

Roxy-3DPrintBoard, 2016-05-31:
The Mesh Bed leveling does not have a Bed Level Correction Matrix.  I don't think you have the Mesh Bed Leveling turned on.

GreatGrizzly, 2016-05-31:
@jbrazio I got an email back from them:

Hello

We have a repository on GitHub with all our parts and firmware source code. At this time it's not updated with the MK2 parts, but it will happen over the next few weeks I'd expect.

Regards,

_________

Greyson Haswell
Prusa Research
Prague, Europe
info@prusa3d.com

Eriobis, 2016-05-31:
Oh I'm so sorry, I use auto bed leveling.
Should I use mesh leveling instead ?

Blue-Marlin, 2016-05-31:
@Eriobis 
Your date tells a story of a bed where the mid is about 0.6mm higher than the corners. That can't be succesfully handled with grid bed levelling. Mesh could work better for your bed.

Eriobis, 2016-05-31:
Indeed. And how can I get the mesh leveling ?

Blue-Marlin, 2016-05-31:
@Eriobis 
If you can't find it in your Configuration.h the first step is to updae your Marlin.

Eriobis, 2016-05-31:
Oh ... once again I'm sorry, I was looking around the auto bed leveling part in the code. Did a couple ctrl + shift + f and I found the mesh leveling. ( I have RC6 btw ).

Thanks for the infos. I don't like ppl on forum who doesn't search a minimum before asking question. But for this mesh thing I coudn't find much info.

Blue-Marlin, 2016-05-31:
https://github.com/MarlinFirmware/Marlin/wiki/Mesh-Bed-Leveling

Eriobis, 2016-05-31:
Yup, did implement it and i'm currently printing with it. Works perfectly thanks !

WZ9V, 2016-05-31:
If this has been implemented I'll help out with testing.  I'd prefer to use mesh but I'm lazy and manual probing is not my thing.

Eriobis, 2016-06-01:
@Blue-Marlin 
I did the sequence on the wiki, when I try to send G29 the answer is : Mesh Bed Not Active
So I am not able to save the points and restore them ... any clue ?

Blue-Marlin, 2016-06-01:
```
 /**
   * G29: Mesh-based Z probe, probes a grid and produces a
   *      mesh to compensate for variable bed height
   *
   * Parameters With MESH_BED_LEVELING:
   *
   *  S0              Produce a mesh report
   *  S1              Start probing mesh points
   *  S2              Probe the next mesh point
   *  S3 Xn Yn Zn.nn  Manually modify a single point
   *  S4 Zn.nn        Set z offset. Positive away from bed, negative closer to bed.
   *
   * The S0 report the points as below
   *
   *  +----> X-axis  1-n
   *  |
   *  |
   *  v Y-axis  1-n
   *
   */
```

thinkyhead, 2016-06-01:
> I did the sequence on the wiki

You should manually do:
```gcode
G29 S1 ; home and probe 1
G29 S2 ; next probe... 2
G29 S2 ; next probe... 3
G29 S2 ; next probe... 4
G29 S2 ; next probe... 5
G29 S2 ; next probe... 6
G29 S2 ; next probe... 7
G29 S2 ; next probe... 8
G29 S2 ; next probe... 9
```
And then you should be all set to start printing.

GreatGrizzly, 2016-06-02:
Is it possible to home the z axis without setting the z axis origin?

epatel, 2016-06-03:
@Eriobis MBL should do a homing as the last part of the probing sequence. MBL is not enabled/turned on until after the homing has stopped. Maybe you are trying to save it to EEPROM before the end of the homing? Try to wait to after the homing is done. This waiting might be necessary after this PR https://github.com/MarlinFirmware/Marlin/pull/3835 because the `has_mesh()` is set before the final `G28` is made.

Eriobis, 2016-06-03:
It is stored into the EEPROM, but after a print if I send a G29 i have the answer Mesh leveling not active
I have to to a M501 every time to load the mesh map, then launch the print and the print does a G28 and it starts perfectly.
Is there any way to fix this loss of "mesh map" so I don't have to do M501 every time ?

epatel, 2016-06-03:
@Eriobis Sounds like you maybe have a homing single axis in your "end g-code". The PR I reference above is addressing that. See https://github.com/MarlinFirmware/Marlin/issues/3750

thinkyhead, 2016-06-04:
#3956 (and #3835) merged. Should be better.

petrzjunior, 2016-06-05:
> I got an email back from them, they dont have all the code updated as I suspected.

@jbrazio @Roxy-3DPrintBoard If you want the source, try to contact some owner of the MK2, they are allowed to request source, by the license. :laughing:

Eriobis, 2016-06-05:
The code has been on prusa's github for 3 month now.
[https://github.com/prusa3d/Orignal-Prusa-i3/tree/MK2](url)

WZ9V, 2016-06-05:
That link didn't work is this the right one

https://github.com/prusa3d/Prusa-Firmware/tree/9c2dc0bb42de19c0e9bda81d48beeddf9b166443

Roxy-3DPrintBoard, 2016-06-05:
> If you want the source, try to contact some owner of the MK2, they are allowed to request source, by the license.

I don't want the source.   I want the source to be available to the people that need it.   And it doesn't have to be a customer of the company that requests it.   It can be anybody that just wants to look and see how they did things.

WZ9V, 2016-06-05:
I believe the source is there on github.

https://github.com/prusa3d/Prusa-Firmware/commits/MK2

GreatGrizzly, 2016-06-06:
@WZ9V 

Hmm, it looks like it is all there. Under the MK2 branch.

[3 more replies]

closed 2016-06-16 by jbrazio