MarlinFirmware/Marlin #3878: The feed rate or flow rate percentage mark isn't changing on the LCD panel.

boelle, 2016-05-25:
[image] **Issue by [a4jp-com](https://github.com/a4jp-com)**
_Monday Jan 04, 2016 at 21:41 GMT_
_Originally opened as https://github.com/MarlinFirmware/MarlinDev/issues/327_

----

The feed rate or flow rate percentage mark on the status screen isn't changing when I go into tune during a print and adjust the flow rate / feed fade. Does everyone else have this problem or is it just my machine?

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Thursday Jan 07, 2016 at 14:38 GMT_

----

How do we change the value on the LCD screen?

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Sunday Jan 10, 2016 at 12:39 GMT_

----

Any ideas on this?

boelle, 2016-05-25:
[image] **Comment by [KiteLab](https://github.com/KiteLab)**
_Sunday Jan 10, 2016 at 13:06 GMT_

----

How about changing Feedrate and or Flow in the Tune-Menu?
Changing Feedrate on the Status screen may be broken, for you, because of your missing encoder-wheel.

https://github.com/MarlinFirmware/MarlinDev/blob/dev/ultralcd.cpp#L346

```    
#if ENABLED(ULTIPANEL_FEEDMULTIPLY)
      // Dead zone at 100% feedrate
      if ((feedrate_multiplier < 100 && (feedrate_multiplier + int(encoderPosition)) > 100) ||
          (feedrate_multiplier > 100 && (feedrate_multiplier + int(encoderPosition)) < 100)) {
        encoderPosition = 0;
        feedrate_multiplier = 100;
      }
      if (feedrate_multiplier == 100) {
        if (int(encoderPosition) > ENCODER_FEEDRATE_DEADZONE) {
          feedrate_multiplier += int(encoderPosition) - ENCODER_FEEDRATE_DEADZONE;
          encoderPosition = 0;
        }
        else if (int(encoderPosition) < -ENCODER_FEEDRATE_DEADZONE) {
          feedrate_multiplier += int(encoderPosition) + ENCODER_FEEDRATE_DEADZONE;
          encoderPosition = 0;
        }
      }
      else {
        feedrate_multiplier += int(encoderPosition);
        encoderPosition = 0;
      }
    #endif // ULTIPANEL_FEEDMULTIPLY

    feedrate_multiplier = constrain(feedrate_multiplier, 10, 999);
```

Test if ULTIPANEL_FEEDMULTIPLY is defined.
Test if a smaller ENCODER_FEEDRATE_DEADZONE can help you.

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Sunday Jan 10, 2016 at 19:42 GMT_

----

I can change the feed rate number while printing with the up, down, left and right buttons but the status screen values don't change.

How can I get this code working with my buttons? I tried #define ULTIPANEL_FEEDMULTIPLY in my Customization.h file but nothing changed.

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Monday Jan 11, 2016 at 21:24 GMT_

----

@KiteLab when you say "How about changing Feedrate and or Flow in the Tune-Menu?" do you mean in the code?

boelle, 2016-05-25:
[image] **Comment by [KiteLab](https://github.com/KiteLab)**
_Tuesday Jan 12, 2016 at 00:04 GMT_

----

I meant, during a print you should have a 'Tune' menu in the main menu, where you can tune Feedrate and Flow. The code i mentioned above is the place where the Feedrate is tuned on the Status-screen.
But all that seems to be irrelevant since you clarified that you can change the values, only the __display__ for the Feedrate on the Status-screen is messy.
This evening i investigated if FR could be written over from an overflow of an previous line. But since everything is written top down on the display that is unlikely. Sorry i have no further ideas. Even if you messed up #311 it should work. But without having a 16x4 display i'm not absolutely sure..

Did i mention, there is a death zone around 100%, and you may have to push the buttons several times to edit the Feedrate on the status screen?

Feedrate is about to adjust the print speed.
Flow is about to adjust the amount of extruder material.

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Friday Apr 22, 2016 at 01:32 GMT_

----

@a4jp-com Any luck? Still need this issue open?

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Friday Apr 22, 2016 at 07:17 GMT_

----

It still doesn't change.

boelle, 2016-05-25:
[image] **Comment by [Blue-Marlin](https://github.com/Blue-Marlin)**
_Friday Apr 22, 2016 at 11:03 GMT_

----

Who wonders. It would need about the same changes for your up/down buttons as in the menu-code, but at a different place (checked in the status screen).

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Tuesday May 03, 2016 at 03:23 GMT_

----

> It still doesn't change.

Using `RC6` or `RCBugFix`? I will take a look at the code and try it on my machine. I haven't used this option lately, so who knows, perhaps it's totally broken.

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Tuesday May 03, 2016 at 03:38 GMT_

----

Thank you for checking Thinkyhead. I'm not sure what version of the software I'm on but @Wackerbarth helped me set up everything and the button code for the printer was working after some super programming on his part. I'll see if I can find out from him. Hopefully all the updates from the version I'm on now got moved into the latest version here.

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Tuesday May 03, 2016 at 03:48 GMT_

----

> got moved into the latest version here

The latest version is actually here: https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

If you haven't tested that, you should. If you need help applying button changes, let me know. I'll be around this week.

boelle, 2016-05-25:
[image] **Comment by [Blue-Marlin](https://github.com/Blue-Marlin)**
_Tuesday May 03, 2016 at 11:02 GMT_

----

@thinkyhead 
> and try it on my machine

@a4jp-com has no encoder-wheel but a bunch of buttons. Wackerbarth made a kind off custom version for him. It's very unlikely that RCBugFix will work for him. Search for "Scoovo" in Dev.
[image]

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Tuesday May 03, 2016 at 19:39 GMT_

----

> It's very unlikely that RCBugFix will work for him.

So… Looks like some fairly simple changes are needed. Currently Marlin only acknowledges the `BTN_UP` (etc.) pins for the `RIGIDBOARD` controller. If we relax that condition and simply initialize these pins in `lcd_init` when they exist, and then check them in `lcd_buttons_update` (when `MB(AJ4P)`?) then this controller should _just work_.

Try MarlinFirmware/Marlin#3670 at your earliest convenience @a4jp-com

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Wednesday May 04, 2016 at 07:10 GMT_

----

Hi. Sorry for the late reply. The version that works so far on my machine is here: [https://github.com/a4jp-com/MarlinDev-a4jp/](https://github.com/a4jp-com/MarlinDev-a4jp/)

boelle, 2016-05-25:
[image] **Comment by [thinkyhead](https://github.com/thinkyhead)**
_Thursday May 05, 2016 at 03:52 GMT_

----

@a4jp-com Yep. I looked at your branch first, then I created this PR which makes the necessary changes to add support to the current Marlin codebase: MarlinFirmware/Marlin#3670 Please test at your earliest convenience.

jbrazio, 2016-05-27:
Merged.

closed 2016-05-27 by jbrazio