MarlinFirmware/Marlin #3882: Printrbot planning new fork

boelle, 2016-05-25:
[image] **Issue by [thawkins](https://github.com/thawkins)**
_Sunday Dec 27, 2015 at 02:45 GMT_
_Originally opened as https://github.com/MarlinFirmware/MarlinDev/issues/316_

----

The printrbot team is planning a fork of Marlin,  Brooke Drumm has announced this on thier google++ feed. 

The Marlindev community should probaly talk to them to determine what thier beef is. And why they feel that they need to fork the firmware. 

https://plus.google.com/u/0/+BrookDrummpb/posts/gWupi2KvLiC?cfem=1

Jnesselr, 2016-05-25:
I'm going to go ahead and close this issue. Brook is notorious for wanting his own little printrbot ecosystem, and it's not something I think the maintainers of marlin should worry about.

closed 2016-05-25 by Jnesselr

thinkyhead, 2016-05-27:
I would recommend to them that after 1.1.0 is released, they should think about basing their next forked version off of that.