MarlinFirmware/Marlin #3883: How to configure no XY endstops

bond4u, 2016-05-25:
Comment in Configuration.h says "leave undefined", but when I comment them out then Marlin doesn't compile.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
No X and Y endstop, only Z min is present.

jbrazio, 2016-05-27:
@thinkyhead can you review this one ?

AnHardt, 2016-05-27:
A machine without at least one endstop/homeswich per axis can not be homed - it's unusable. (Ok, triks with G92 are possible but unsafe because even the software endstops can not work.)

thinkyhead, 2016-05-27:
The "QuadRap" has no endstops. Perhaps it should be easier to do that.

AnHardt, 2016-05-27:
The workaround is to define one endstop per axis and playing with `?_???_ENDSTOP_INVERTING` until this endstop is always untriggered.

What is the supposed workflow with a  "QuadRap"? Homing is manually moving to [0,0] and reset?

Dippyskoodlez, 2016-05-29:
> What is the supposed workflow with a "QuadRap"? Homing is manually moving to [0,0] and reset?

I'm setting up a new printer and have some endstop troubles.  Allowing this functionality would be pretty nice to have for some niche users. My workflow is indeed a manual zero and then only zero when the controller+arduino have reset.

thinkyhead, 2016-05-30:
> What is the supposed workflow with a "QuadRap"? Homing is manually moving to [0,0] and reset?

Setting up a new printer, as @Dippyskoodlez is doing, is one use-case for disabling all endstops, and it's the reason cited for making the QuadRap (initially) with no endstops, to reduce the cost (a tiny bit) and to make it easier to do the initial build. I actually edited the article about QuadRap in RepRap Magazine Issue #3, You can [read about the QuadRap starting on Page 61](https://issuu.com/garyhodgson/docs/reprapmagazine_issue_3?e=2800691/7156572).

bond4u, 2016-06-21:
I had it working with 1.1.0-rc3 - move the head where You want 0,0 to be and print away (from sd card).
1.1.0-rc6 seems to "remember" it's position and during 2nd print attempts to go out of build area.

thinkyhead, 2016-06-22:
@bond4u Once the machine "knows" where 0,0,0 is, it should always be able to re-align for the next print, even if you don't re-home. But if your GCode includes a `G92 X0 Y0` or something similar, that could explain why a second print could start at a different point. I can't think of any other explanations at the moment….

jbrazio, 2016-07-11:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-07-11 by jbrazio

Ianmcmill, 2016-09-18:
I just stumbled into this trap aswell. 
Haven't used endstops (hardware or software) for three years. Always disliked the config logic. Too much hassle. More cables. More mess. I have never ran into my bed or off-limit areas. G92 X0Y0 respectively reseting the board was my fellow companion for years. 1.1.0-RC5. 
Recently I got an inductive sensor probe to do the auto bed-leveling and now I am forced to deal with endstops. My goal is to use the z probe for bed leveling without any kind of endstops. Be it hardware or software. How would I accomplish this?

I tried RC7 Bugfix but got the same (error, I mean it is not an error as it was intended to work exactly as the authors want it to work) as with RC7 and bond4u explained above.

`Almost all printers will be using one per axis.`
Almost all printers are not configured as mine. However, Marlin used to be individually configurable so I had never had any issues with this .

With the new version I am not able to disable endstops at all (??). I need some sort of endstops to be able to move my printhead freely around the place as I like.

Am I getting this wrong? Did I overlooked something? Code structure in the recent RC7 (Bugfix) release has been modified in several places so I might just forgot one thing. I am most likely sure that I overlook something or get something wrong.

Configuration.h from RCBugfix (from RC7)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/478744/Configuration.h.txt)

Ianmcmill, 2016-09-18:
I had the same issue as bond4u.
Eventually I got around this by inverting the min_endstops.

`#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.`

At least I think this what this issue was about....