MarlinFirmware/Marlin #3886: Hardware Endstops

boelle, 2016-05-25:
[image] **Issue by [AnHardt](https://github.com/AnHardt)**
_Sunday Nov 22, 2015 at 18:23 GMT_
_Originally opened as https://github.com/MarlinFirmware/MarlinDev/issues/285_

----

Hardware Endstops

A small collection of problems with hardware endstops. I'm not sure i'v got all of them.
Sorry. I was to lazy to search for all the original issues. Some of the below questions are constructed.

* 1 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-max endstop. __Before__ homing i can't go in -z direction. What to do?
Workaround: Home the machine.

* 2 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-max endstop. __After__ homing i can't go in -z direction. What to do?
Workaround: Set `#define ENDSTOPS_ONLY_FOR_HOMING`

* 3 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-min endstop. I can't go in -z direction and I can't home. What to do?
Workaround: Don't use this configuration.

Reason: The z-min probe acts as a z-min endstop - always when the endstops are checked.
Solution: Add a new variable `bool z_probe_is_ready = false`. Don't test the z-min-probe-switch when false. Set and reset the variable during deploing/undeploing the probe. Make all this configurable with `//#define Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE`. Connect your probe to a probe-pin, not the z-min-endstop-pin.

* 4 Question: I set hardware endstops to on with M120. After G28, G29 or any other g-code using the probe this is forgotten.
Workaround: Use M120 after using the probe.
Reason: The endstop-state is not remembered. After using the probe the state is set depending on `ENDSTOPS_ONLY_FOR_HOMING` only.
Solution: Remember the previous state when switching to "on" for a wanted endstop move, like it is done for the feedrate in `setup_for_endstop_move()`. Restore the previous state in `clean_up_after_endstop_move()`. Disable M120 before homing and give a warning - that's dangerous.

* 5 Question: On my delta the coordinate system is messed up when the probe triggers somewhere between starting G29 and the first probe.
Workaround: Set up your machine so that the probe is not triggered before.
Reason: When on a delta a endstop / probe is triggered unexpected no correction can be made.
Solution: Make software endstops work and narrow down the time the hardware endstops are active to the absolute minimum (`run_z_probe()`, `homeaxis()`). The right reaction on an unexpected hardware endstop hit is to stop the current and all further moves.

* 6 Question: On my delta machine the coordinate system is messed up randomly.
Workaround: Set `ENDSTOPS_ONLY_FOR_HOMING`. 
Reason: Noise on the endstop/probe inputs.
Solution: Fix the endstops, their wiring, ... . Fix software endstops. The right reaction would be to stop the print, instead of making random looking moves.

* 7 Question: When i print the layers shift in some direction. Somewhen later a hardware endstop is hit. Thereafter the layers are shifted a lot in an other direction. The part is a mess but print continues.
Workaround: You are loosing steps in one direction. If you see the layer shifting, stop the print and try to work against loosing steps.
Solution: If a hardware endstop was hit after the machine was homed, something catastrophic has happened before. The software endstops should have prevented that the hardware endstops could have been hit. Even without software endstops the next move will never end where it is expected to. Stop the machine when a hardware endstop hit was detected after homing.

* 8 Question: I homed my machine, send a `G1 X10`, `G1 X0`. I get a message that the endstop was triggered.
Workaround: Set `#define ENDSTOPS_ONLY_FOR_HOMING`
Reason: The x-min-position is set while the switch is triggered. Returning to there will trigger the switch again.
Solution: Back up from the switch before the min-position is set. For example by `X_HOME_BUMP_MM`.

* 9 Question: When i move around i can hit my hardware endstops and get messages. What to do?
Answer: Home your machine before moving around.
Question: If that does not help?
Answer: Activate software endstops.
Question: Even that has not helped. The next move after a endstop hit is strange.
Answer: You have a delta machine. Software endstops are more or less unimplemented there.
Workaround: Don't move to a impossible place.
Solution: Implement proper software endstops for deltas.

* 10 Question: What is the action to do if a hardware-endstop-hit is detected before homing?
Workaround: Homing should always be the first move.
Solution: I don't know. The minimum is to stop the actual move. I guess the strongest is to clean the planner buffer, to prevent further already buffered moves from hitting the other side, where potentially no endstop is.

* 11 Question: I can't move an axis in any direction. What to do?
Workaround: Test your endstops with M119. Maybe both endstops of that axis are triggered at the same time. Configure your endstos correct.
Solution: Send a warning during boot/selftest. Triggered endstops at both ends, at the same time, indicate an error.

* 12 Question: I can't move an axis towards an endstop.
Workaround: Test your endstop and configure it correct.
Solution: During boot/selftest, if an endstop is triggered but not both, back up slowly `?_HOME_BUMP_MM`. If the endstop then is not free, throw an error.

* 13 Question: When a hardware endstop is triggered and the hardware endstops are tested i can move away from them. The move is not stopped. I expected they'd stick on them.
Answer: Thats a feature, not a bug. The endstops are not tested when the move is away from them.
Question: But isn't testing the direction slower than the actual test of the endstop?
Answer: Maybe we should test the endstop first and make the decision what to do with that information later.
Question: But wouldn't sticking be a simple method to stop any pending moves until we actively free the endstop?
Solution: To be discussed.

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Sunday Nov 22, 2015 at 18:55 GMT_

----

* 14 Question: When i move down, the nozzle stops to go down. I can't go below that height. The height is about `Z_PROBE_OFFSET_FROM_EXTRUDER`.

Answer: Are the hardware endstops on? Test with M119 if your z-min-endstop ore zmin-probe is triggered. If yes, you could have a inductive probe. 
Workaround: In that case you have to switch off the hardware endstops. define `ENDSTOPS_ONLY_FOR_HOMING`
Solution: Connect your probe as a probe. Use the new `#define Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE` feature. This way x and y stops are still checked but not z.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Sunday Nov 22, 2015 at 20:25 GMT_

----

"Connect your probe to a probe-pin, not the z-min-endstop-pin." --  Shouldn't this be the proper solution for all cases? If the signal is a probe, then call it that and if you don't actually have an end-stop, don't claim that you do.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Sunday Nov 22, 2015 at 20:36 GMT_

----

5 - "On my delta the coordinate system is messed up when the probe triggers somewhere between starting G29 and the first probe." -- How is this an issue different that Q6? An end stop SHOULD never trigger unless we are doing an "align" (homing) operation. If that is true, then it is either "electrical noise" (false triggers) or you are truly trying to exceed the working range of the actuators.
In the former case, you really need to fix the underlying cause. In the latter case, there is some error in the configuration of the mechanical boundaries.

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Sunday Nov 22, 2015 at 21:13 GMT_

----

In 5 you don't have the choice to set `ENDSTOPS_ONLY_FOR_HOMING` because G29 took control over the endstops. In that special case the the user was deep enough to trigger the z-min-probe, but did not crash and messed up the stepper coordinate system.
And yes, the propper reaction is to stop this and all further moves until rehomed. 

For a while i thought about how to reanimate a sticked to an endstop, or halted for a endstop hit printer. A special g-code, a button, a menu point? A good answer could be G28 - homing - that's needed anyway.

__EDITED__
There is no reason for not respecting `ENDSTOPS_ONLY_FOR_HOMING` during the not actualy probing moves in G29 and the other probing g-codes. 

For the actual progaming this means we must rename `setup_for_endstop_move()` to `save_feedrate()`  take out the endstop part, put that in a new 'prepare_endstop_move()' and looking for places where to apply. Same for `clean_up_after_endstop_move()`.

The correctness of the current use of  `setup_for_endstop_move()` is questionable anyway. Sometimes i have very slow feedrates after a G29 if not explicit set.

__EDIT 2__
Bah. I'm mad. We have to activate the probe by `deploy_z_probe()`, deactivate by `stow_z_probe()` and leave the endstops as they are. Or are there cases, where we have to deactivate a z-min-endstop to guarantee a successful probing? Isn't that already handled, somewhere?

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Sunday Nov 22, 2015 at 21:44 GMT_

----

If we'd say, a probe is like a endstop but needs to be activated/deactivated we have a reasonable and logic difference between them. Until now that was a bit blurry in my brain (It was more a second z-min-endstop for me). If all kinds of probes are not tested when not activated we do not need `Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE`.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Sunday Nov 22, 2015 at 21:57 GMT_

----

From my perspective, a probe is a sensor that needs to be tested in only two circumstances:
1) The obvious, probing -- that time during which we are actively moving toward the bed while looking for the proximity detection event.
2) Possibly as a part of the probe deploy/stow procedure -- For example my "Allen Key" probe registers "in proximity" when it is properly stowed.

Both of these circumstances are related to short durations and specific places in to motion code.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Sunday Nov 22, 2015 at 22:59 GMT_

----

Just as an FYI:    In the RC-2 I'm bringing up on my Delta, I warmed over Stepper.cpp to do this.   The Z-Probe switch is only checked if the probe is not stowed.  The z_probe_is_stowed variable gets set and cleared in stow_z_probe() and deploy_z_probe() respectively. 

~~~~
           if (z_test && current_block->steps[Z_AXIS] > 0) { // z_test = Z_MIN || Z2_MIN
              endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
              endstop_hit_bits |= BIT(Z_MIN);
              if (!performing_homing || (z_test == 0x3))  //if not performing home or if both endstops were trigged during homing...
                step_events_completed = current_block->step_event_count;
            }
          #else // !Z_DUAL_ENDSTOPS
	    if (z_probe_is_stowed==false)
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
          {
            endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
~~~~

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Sunday Nov 22, 2015 at 23:07 GMT_

----

The test, if a Allen Key was deployed and or stowed  successfully can be done and is done outside the interrupt, asking the pin directly.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Sunday Nov 22, 2015 at 23:34 GMT_

----

@AnHardt  I'm not following what you are saying.   update_endstops()  is called at interrupt time.  Can you point me to the code you are talking about?

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Sunday Nov 22, 2015 at 23:37 GMT_

----

"The test, if an Allen Key was deployed ... done outside the interrupt"
I agree. I've even wondered if we might even take all of the end stop / probing tests out of the stepper interrupt and handle them at a lower priority.

There are only two cases where these sensors are important.
When they are used as a "safety", their sensing of proximity is going to abort the print (hopefully before the mechanism is damaged).  In this case, I suspect that there is some "slop". As long as the actuator doesn't take "too many" additional steps, we should be OK. Once the condition is triggered, we no longer really care about the true position of the mechanism.

The other case is during "alignment" (AKA homing) or "probing".  Under these conditions, accuracy is important. However, done properly, we should be moving very slowly while looking for that accurate edge.  Additionally, we really need only worry about one sensor at a time.

Does anyone have an idea just how quickly we MUST react to an end stop? (The time from when an end stop triggers until the steppers no longer receive pulses)

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Nov 23, 2015 at 00:19 GMT_

----

> Does anyone have an idea just how quickly we MUST react to an end stop? (The time from when an end stop triggers until the steppers no longer receive pulses)

This seems like it should be straight forward to calculate.    Maybe in the morning I'll feel like getting out the calculator!   :sunglasses:

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 01:10 GMT_

----

@Roxy-3DPrintBoard 
~~~~
           if (z_test && current_block->steps[Z_AXIS] > 0) { // z_test = Z_MIN || Z2_MIN
              endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
              endstop_hit_bits |= BIT(Z_MIN);
              if (!performing_homing || (z_test == 0x3))  //if not performing home or if both endstops were trigged during homing...
                step_events_completed = current_block->step_event_count;
            }
          #else // !Z_DUAL_ENDSTOPS
/*
	    if (z_probe_is_stowed==false)
I'm quite sure the z-min-endstop should not depend on the probe.
If you connect your probe to a z-min-pin because you have no z-min-switch connected, you should #define Z_MIN_PROBE_PIN former_Z_MIN_PIN 
*/
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
//            ^^^^^^^^^^^^^^^^^^^^^^^^^ why testing this twice? how about a '{' behind the if 3 lines above.
          {
/*
          endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
         This is already done in the UPDATE_ENDSTOP(Z, MIN_PROBE) macro.
*/
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
~~~~

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 01:12 GMT_

----

@Roxy-3DPrintBoard 
MarlinMain.cpp
```
      #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        bool z_probe_endstop = (READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING);
        if (!z_probe_endstop)
      #else
        bool z_min_endstop = (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING);
        if (!z_min_endstop)
      #endif
        {
          if (IsRunning()) {
            SERIAL_ERROR_START;
            SERIAL_ERRORLNPGM("Z-Probe failed to retract!");
            LCD_ALERTMESSAGEPGM("Err: ZPROBE");
          }
          Stop();
        }
    #endif // Z_PROBE_ALLEN_KEY
```

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 01:23 GMT_

----

During probing we need the tests to be in the interrupt, because they are in sync with the steppers there and we recalculate the position. During homing we only want to stop - but as fast as possible. The other time we only need the information if one of the switches was hit.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Monday Nov 23, 2015 at 01:38 GMT_

----

"in sync" only implies that we know the step position to within one step. That is not REQUIRED to be at the interrupt level IF the sampling rate is greater than the step rate.  If, when we are probing, we are moving slowly enough, the sampling can also be relatively slow and can be in a completely independent loop.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Nov 23, 2015 at 03:50 GMT_

----

>  That is not REQUIRED to be at the interrupt level IF the sampling rate is greater than the step rate. If, when we are probing, we are moving slowly enough, the sampling can also be relatively slow 

Cool!   I just realized something.   This is important if I get out my calculator tomorrow.    I certainly could be wrong.   But this sounds a lot like 'Nyquist Theorem' stuff.   If that is true, our sampling rate has to be twice (at a minimum) of what ever frequency we want to measure.   

I'll be reading this tomorrow:  https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem

PS.  I didn't do well on the test where this was.   Any time those i's show up as an exponent in an integral I just move on to the next problem.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Monday Nov 23, 2015 at 04:39 GMT_

----

@Roxy-3DPrintBoard  -- No, Nyquist has to do with determining the frequency components of signals.

In our case, we need only have a sampling rate no lower than the stepping rate.  That way, we are ALWAYS assured that for each step taken, we have the ability to sample the sensor at least once (just as would be the case if the sampling is done in the ISR)

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Monday Nov 23, 2015 at 05:10 GMT_

----

I read in another thread the hardware endstops were disabled but in RC-2 have they been enabled again?

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 12:55 GMT_

----

@a4jp-com 
Please reread. The hardware endstops are still deactivated after homing everywhere by default. 'But it's your choise. You can activate or deactivate `ENDSTOPS_ONLY_FOR_HOMING` by your will.
`ENDSTOPS_ONLY_FOR_HOMING` is activated by default because otherwise it's very likely you'll experience some of the problems described above. Please have an eye on the workarounds.
In this thread we try to find out what is necessary to change, to get less trouble and more profit from hardware endstops.
You are the one who asked for this thread.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Nov 23, 2015 at 15:33 GMT_

----

@AnHardt 
~~~~
          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
//            ^^^^^^^^^^^^^^^^^^^^^^^^^ why testing this twice? how about a '{' behind the if 3 lines above.
          {
~~~~
I usually catch those kind of easy optimizations!   But I missed that one!   I wish I could get Arduino to generate .ii files so I could look at the generated code.   It would be interesting to see if GCC caught that and optimized the extra check out.

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Monday Nov 23, 2015 at 15:39 GMT_

----

@AnHardt I reread the top bit but couldn't see the answer to my question but lots of different answers. Your post here clears up my question though ^^ 

Thank you so much for taking the time to write such a detailed answer and thank you also for creating this thread. 

Hopefully I should be getting my printer back in a day or two. Everything is looking good here. I even found a way to get the RAMBo board listed in Arduino so I just need to fill out the details in the Marlin files and do a few test prints after flashing the firmware. 

I physically turn off my printer so I don't think I'll have any fires but it still good to hear about things like that.

@Roxy-3DPrintBoard nice catch.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Monday Nov 23, 2015 at 15:46 GMT_

----

@a4jp-com -- AFAICT, the RAMBo is nothing more than a Mega2560. I suggest that you look at the MarlinDev wiki and obtain the build package for Marlin rather than attempting to use some other version.

Our standard builds make use of modified "cores" and you are likely to have issues unless you carefully mix your components.

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 16:13 GMT_

----

Hardware endstop update outside the stepper IRQ?

The IRQ frequency varies withe the speed of the motion.
Outside the IRQ we can not guarantee any timing.
To get the same precision of the timing, we have to check the endstops at least as often as inside the IRQ.
What is the possible win? 

During probing we want that precision, we are measuring a distance.
During homing we want that precision, we want to find a point.
During normal operation we do not need that precision but want to stop the move before the endstop is smashed. When moving fast, because of the mechanical inertia, this means, as fast as possible. 
Especially when moving fast, we are slow outside the IRQs.

Can the work to be done inside the IRQ be reduced, if we only want to stop?

When we take away all the stuff for CORE?? and DUAL_X_CARRIAGE the remaining structure is much more obvious.
```
if (TEST(out_bits, X_AXIS))   // stepping along -X axis (regular Cartesian bot and DELTAS!)
  {
            #if HAS_X_MIN
              UPDATE_ENDSTOP(X, MIN);
            #endif
  }
  else { // +direction
            #if HAS_X_MAX
              UPDATE_ENDSTOP(X, MAX);
            #endif
  }
for X, Y and Z axis. Only for z- we additionally ask for the z-probe
```
I don't think that can be speed up anyhow. In the best case, moving away from all endstops we check no endstop, in the worst case all. If we'd drop the direction test we'd have to check always all endstops. The direction test at least looks much cheaper than the UPDATE_ENDSTOP()_macro.

Let's have a look into `UPDATE_ENDSTOP()`
```
  #define UPDATE_ENDSTOP(AXIS,MINMAX) \
    SET_ENDSTOP_BIT(AXIS, MINMAX); \
    if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX))  && (current_block->steps[_AXIS(AXIS)] > 0)) { \
      endstops_trigsteps[_AXIS(AXIS)] = count_position[_AXIS(AXIS)]; \
      _ENDSTOP_HIT(AXIS); \
      step_events_completed = current_block->step_event_count; \
    }
```
Hard to say what is going on. Let's expand some macros for AXIS = X and MINMAX = MIN.
Where X_MIN is now a bitmask like 00001000.
```
    if(READ(X_MIN_PIN) ^ X_MIN_ENDSTOP_INVERTING) current_endstop_bits |=  X_MIN:
    if (current_endstop_bits&X_MIN ==  old_endstop_bits&X_MIN)  && (current_block->steps[X] > 0)) { 
      endstops_trigsteps[X] = count_position[X]; 
      endstop_hit_bits |= X_MIN; 
      step_events_completed = current_block->step_event_count; 
    }
...
old_endstop_bits = current_endstop_bits;
```
endstop_hit_bits is the communication to the rest of the world outside the IRQ.
current_endstop_bits and old_endstop_bits build a minimal debouncer.
if(current_block->steps[X] > 0)) = are we still moving?
(step_events_completed = current_block->step_event_count; ) = stop the move!

And hura! (endstops_trigsteps[X] = count_position[X]; ) = the part where we store the position and what could be omitted.

That's disappointing. There is about nothing to improve.  :-) 

So why do we get complaints about hardware endstops are CPU intensive?
I think this is not a general problem, but it is for CORE?? printers.
There we have additional tests like
```
if ((current_block->steps[A_AXIS] != current_block->steps[B_AXIS]) || (TEST(out_bits, A_AXIS) == TEST(out_bits, B_AXIS))) {
```
before every direction test. That is at least 2-3 times more work than the direction test alone. That means the best case scenario is lasting at least 2-3 times longer.

boelle, 2016-05-25:
[image] **Comment by [a4jp-com](https://github.com/a4jp-com)**
_Monday Nov 23, 2015 at 16:55 GMT_

----

RAMBo setup is a bit different to the Mega2560. Even though it is kind of the same, if I set the board to Mega2560 it works, but the LCD doesn't. I have to use the RAMBo setup to get the LCD working.

boelle, 2016-05-25:
> if (TEST(out_bits, X_AXIS))   // stepping along -X axis if (TEST(out_bits, X_AXIS))   // stepping along -X axis (regular Cartesian bot and DELTAS!)
> That's disappointing. There is about noting to improve. :-)
> So why do we get complains about hardware endstops are CPU intensive?

[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Monday Nov 23, 2015 at 17:03 GMT_

----

A few comments, “homing” is actually a series of two operations. The first is a rapid movement to (or near) the end stop. The second is to back off and SLOWLY re-approach. This second operation is just like “probing” and, I do agree, does need accuracy.

In reference to  “crash detection”, I agree that you have stated the reason for rapid response. However, I do not see anything more than your assertion that that rapidity MUST be "as fast as possible”.

I’ve worked on massive machinery that would never even consider having its “crash detection” as a part of the controller. When those limit switches are hit, they throw the breaker on the mains.

Every limit switch has SOME overshoot, even the poorly designed ones where the mechanism runs directly into the switch rather than just deflecting a control arm.

The question becomes one of relative scales. How much extra time/steps can you wait and still stop in time. Obviously that depends on mass, speed, and stopping distance.

But, just because you could, doesn’t mean that you need to apply maximum braking.
So, I renew my query concerning just how fast we really NEED to react.

boelle, 2016-05-25:
[image] **Comment by [Wackerbarth](https://github.com/Wackerbarth)**
_Monday Nov 23, 2015 at 17:09 GMT_

----

@a4jp-com -- Have you tried the MARLIN Mega2560 setup (not the Arduino one)?
I suspect that your issue might be with the libraries rather than the "pins".  In any case, if our setup doesn't work, I need to know EXACTLY where it differs from yours so that I can fix ours to work.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Nov 23, 2015 at 18:49 GMT_

----

> But, just because you could, doesn’t mean that you need to apply maximum braking.
So, I renew my query concerning just how fast we really NEED to react.

This answer doesn't really help, but it is probably accurate:  We need to stop before the switch bottoms out and things crash.   The stepper pulses are relatively slow compared to how quickly we can get into the ISR and start doing things.   

~~~~
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80, 80, 80, 93}
#define DEFAULT_MAX_FEEDRATE          {1500, 1500, 1500, 45}    // (mm/sec)              
~~~~

Does this imply a max step rate of 80*1500 steps / sec ?   Because that is 8 micro-seconds of time before the next step happens.    My micro-switches have at least 1/32" of travel.  So that is going to be 20 or 30 steps.    In my case, if we could stop the motors within 20 or 30 steps I wouldn't have a crash.

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 21:59 GMT_

----

@Wackerbarth 
>A few comments, “homing” is actually a series of two operations. The first is a rapid movement to (or near) the end stop. The second is to back off and SLOWLY re-approach. This second operation is just like “probing” and, I do agree, does need accuracy.

You are right. It should be that way, it is this way for Cartesian printer, but not for delta printers - what is not perfect, but works - for now.

>In reference to  “crash detection”, I agree that you have stated the reason for rapid response. However, I do not see anything more than your assertion that that rapidity MUST be "as fast as possible”.

A typical Mini Kossel Kit Endstop Mount:
[image]
Maximum braking distance about a mm from trigger to bang. Typically 80steps/mm, 500mm/s. (Values from our KosselMini/Configuration.h)
s = v^2/(2*a) >= a = v^2/(2*s).
500^2/(2*1) = 250,000/2 = 125,000mm/s^2. Assuming we can brake with the same deceleration than during accelerate that is more than factor 100 to much (DEFAULT_MAX_ACCELERATION 9000). As fast as possible is not fast enough. Hopefully we can decelerate much better than accelerate. But our A is adjusted to a value where the steppers just do not skip. Full-stop braking means just skipping. So likely we talk about decelerating as much as possible before the bang.

>I’ve worked on massive machinery that would never even consider having its “crash detection” as a part of the controller. When those limit switches are hit, they throw the breaker on the mains.

Good idea. But here we talk about RepRap (crap).

__EDITED__: Not a good idea. If we have no current, we have no, or low, holding torque with steppers.

boelle, 2016-05-25:
[image] **Comment by [AnHardt](https://github.com/AnHardt)**
_Monday Nov 23, 2015 at 22:10 GMT_

----

@Roxy-3DPrintBoard 
You are calculating the distance until the braking begins, not until the stop.

boelle, 2016-05-25:
[image] **Comment by [Roxy-3DPrintBoard](https://github.com/Roxy-3DPrintBoard)**
_Monday Nov 23, 2015 at 22:12 GMT_

----

> Assuming we can break with the same deceleration than during accelerate that is more than factor 100 to much (DEFAULT_MAX_ACCELERATION 9000). As fast as possible is not fast enough. Hopefully we can decelerate much better than accelerate. But our A is adjusted to a value where the steppers just do not skip. Full stop breaking means just skipping. So likely we talk about decelerating as much as possible before the bang.

This is good thinking...    It would be nice to not lose position but the most important thing is to slow down the movement as much as possible before it crashes.

Is the force the stepper motors putting out depend much about where the motor is?  For example, my guess is the micro-steps are not as powerful as running the motors in the normal stepping modes.   And what happens if we lose a step?   Does the next step start pulling the motor forward instead of trying to slow down the movement?

jbrazio, 2016-05-27:
This is a duplicate.

closed 2016-05-27 by jbrazio