MarlinFirmware/Marlin #3890: Hannover (Germany) Maker Faire this weekend

CONSULitAS, 2016-05-26:
Hi folks,

are there any Marlin enthusiast on the Hannover Maker Faire this weekend?

There is somebody talking, we know rather well: http://maker-faire.de/hannover/vortrag/mit-der-machete-in-den-lizenzdschungel/

I am there (Saturday and Sunday 28.05./29.05.), too.

Perhaps we can meet and drink a coffee together...

Greetings

Jochen

AnHardt, 2016-05-26:
I'll likely be there on Sunday.

Andreas (Dr. Marlin)

clexpert, 2016-05-28:
@CONSULitAS @AnHardt Hi guys, we (with my son @petrzjunior) are going to be on Hannover Faire this sunday too. How will we meet up?

AnHardt, 2016-05-28:
13:00 near the fish. Dr. Marlin

clexpert, 2016-05-28:
OK. I do not know if we will find any fish. :-) But we will have blue t-shirts with logo our blog Zahradnik se bavi (see http://www.zahradniksebavi.cz/).

Roxy-3DPrintBoard, 2016-05-28:
Please take and post some pictures!!!!

CONSULitAS, 2016-05-28:
@clexpert 
You talk about the flower? OK.
We (me with my son) have been there today. But we have too look for the fish, too. 😄 
And if you see a guy with this nameplate and the face on my profile...
[image]

@Roxy-3DPrintBoard 
Of the Makerfaire, the projects or the makers, or all of them? 😄 
At your service!

clexpert, 2016-05-28:
@CONSULitAS Yes, flower too :smile: And title "Zahradnik se bavi" too :smile: It means Zahradnik (my name) has fun. We will find laughing guy with moustache and nameplate with marlin fish around fish :smile:

CONSULitAS, 2016-05-28:
@clexpert Sometimes i am stupid:
http://maker-faire.de/hannover/maker/m2016/der-fisch/
😁 
You will find it outside.

CONSULitAS, 2016-05-28:
@clexpert 
Ah: You are our czech care taker! 👍

clexpert, 2016-05-28:
@CONSULitAS Yes :rocket:

AnHardt, 2016-05-30:
[image]

jbrazio, 2016-05-30:
Andreas wasn't kidding.. Dr. indeed !
Nice foto, congrats everyone, I hope it was a good day.

thinkyhead, 2016-05-30:
Some of those faces look quite familiar!

clexpert, 2016-05-30:
Hi everybody! Thank you very much for the meeting at the Maker Faire Hannover 2016. My 16-years son was excited from both Marlin developers @CONSULitAS and @AnHardt (white coated Dr. Marlin). I am finishing video editing from the fair. Andreas, thank you for the photo, I use it in the video too. I can post video link if you want (@Roxy-3DPrintBoard ?).

clexpert, 2016-05-30:
The video from Maker Faire Hannover 2016 is here: https://youtu.be/Xqx3cuDeUn0
It is vlog in Czech language. But do not worry! The greater part of the video is video footage without speaking. You can see a lot of interesting things from the faire.

Roxy-3DPrintBoard, 2016-06-01:
I don't understand the device at 2:43 on the video.   It appears to be a Tesla coil.  But it has a big ball (with no sharp edges on it) at the top like a Van de Graaff generator.   Can the two be mixed together?  The Van de Graaff generator is static electricity and that is why it needs the big sphere at the top to accumulate the charge.  But the lower part of the unit appears to be a high frequency step up transformer.

Either way...   I'm surprised this was allowed into the fair.   I certainly would not want either of those kind of devices any where near my electronics!

jbrazio, 2016-06-01:
btw, @clexpert who was taking the photo, was it @Blue-Marlin ? :-P

AnHardt, 2016-06-02:
No. It was my son.

clexpert, 2016-06-02:
@jbrazio The photo was taking Dr. Marlin's son or wife :smile:

closed 2016-06-15 by jbrazio